Koenkk/zigbee2mqtt #11268: [New device support]:  TuYa 5in1 LED ZigBee-RF-RGBW-CCT-GL-HUE

biern, 2022-02-06:
### Link

https://www.houseiq.pl/pl/p/Kontroler-LED-5w1-ZigBee-RF-RGBW-CCT-GL-HUE-TUYA/1541

### Database entry

{   "id": 25,   "type": "Router",   "ieeeAddr": "0x2c1165fffe9b863d",   "nwkAddr": 24624,   "manufId": 4098,   "manufName": "_TZE200_aa9awrng",   "powerSource": "Mains (single phase)",   "modelId": "TS0601",   "epList": [     1   ],   "endpoints": {     "1": {       "profId": 260,       "epId": 1,       "devId": 81,       "inClusterList": [         0,         4,         5,         61184       ],       "outClusterList": [         25,         10       ],       "clusters": {         "genBasic": {           "attributes": {             "modelId": "TS0601",             "manufacturerName": "_TZE200_aa9awrng",             "stackVersion": 0,             "dateCode": ""           }         },         "genOnOff": {           "attributes": {}         }       },       "binds": [         {           "cluster": 0,           "type": "endpoint",           "deviceIeeeAddress": "0x00124b001949b3de",           "endpointID": 1         }       ],       "configuredReportings": [],       "meta": {}     }   },   "appVersion": 66,   "stackVersion": 0,   "hwVersion": 1,   "dateCode": "",   "zclVersion": 3,   "interviewCompleted": true,   "meta": {     "configured": 821693351   },   "lastSeen": 1644185607734,   "defaultSendRequestWhen": "immediate" }

### Comments

This device looks very similar to https://www.gledopto.com/h-col-388.html which has an open issue here: https://github.com/Koenkk/zigbee-herdsman-converters/issues/3016 but I couldn't make it work in any way.

So far I was trying to make this device work only in RGB+CCT mode. Each mode of operation seems to yield a different manufacturerName when pairing.
The device pairs fine. I tried using one of existing converters to control it with no success. Than I tried the TuYa debug datapoints approach and it didn't log anything at all.
The only thing I was able to get in zigbee2mqtt logs is, after turning on the device:
```
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-02-06 23:13:07: Received Zigbee message from '0x2c1165fffe9b863d', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":768}' from endpoint 1 with groupID 0
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-02-06 23:13:07: No converter available for 'WIP LED Controller' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":768}'
zigbee2mqtt_1    | Zigbee2MQTT:debug 2022-02-06 23:13:13: Received Zigbee message from '0x2c1165fffe9b863d', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
```
I also tried reading max/min color temperature range, however I got an error:
```
error Publish 'set' 'read' to '0x2c1165fffe9b863d' failed: 'Error: Read 0x2c1165fffe9b863d/1 lightingColorCtrl(["colorTempPhysicalMin","colorTempPhysicalMax"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

The device works fine with tuya gateway.
One more thing I've checked was reading the datapoints using Tuya IoT Platform. This is what I was able to retrieve for basic commands:
1 - on/off
3 - brightness
4 - color temp
5 - color

However, that didn't get me anywhere. 
I'd appreciate any help to make this device work.

### External converter

```shell
{
    model: 'TuYa 5in1 LED controller',
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aa9awrng'}],
    vendor: 'TuYa',
    description: 'RGB+CCT light controller',
    extend: extend.light_onoff_brightness_colortemp_color(),
  },
// for tuya data points debug:
{
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aa9awrng'}],
    model: 'TuYa 5in1 LED controller',
    vendor: 'Tuya',
    description: '',
    supports: '',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
}
```

### Supported color modes

color temperature, color (rgb)

### Color temperature range

?

kirovilya, 2022-02-07:
It already works as Skydance WZ5 https://github.com/Koenkk/zigbee-herdsman-converters/blame/master/devices/skydance.js
You need to update Z2M.

biern, 2022-02-07:
Thanks @kirovilya it works perfectly! 
I even was using newest Z2M version, however I still used my debug converters that were prioritized, I didn't anticipate it was recently added. It's manufacturerName hasn't been picked up anywhere by google indexer yet.

I'm still puzzled why the `fz.tuya_data_point_dump` didn't report anything for this device though.

closed 2022-02-07 by biern

allmoney, 2022-10-17:
I have Zigbee2mqtt Current version: 1.18.1-1

But in logs:

Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_aa9awrng'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

How fix?

[image]

allmoney, 2022-10-19:
All is OK. Was old repo. Updated to new version z2m.