Koenkk/zigbee2mqtt #11271: Devices go offline shortly after Zigbee2MQTT startup

mweinelt, 2022-02-07:
### What happened?

For roughly two weeks I see timeouts on all devices in my zigbee2mqtt instance shortly after starting up zigbee2mqtt. Things work for a 10-20 minutes, then only timeouts - until I restart zigbee2mqtt.

- The first zigbee2mqtt version I experienced this with was 1.22.2.
- I'm currently using zigbee2mqtt 1.23.0 on NixOS 21.11
- Communication between Home Assistant and zigbee2mqtt works fine
- Communication between Zigbee2Mqtt and ZigBee devices times out
- Communication can be restored for a while (<30 minutes) by restarting zigbee2mqtt again
- Bindings are still working, so this is likely an issue between the coordinator and zigbee2mqtt

My network consists of:
- Coordinator: CC2652RB (Slaesh)
- Routers: 
  - Philipps Hue Bulbs (6x 920012573A))
  - Ikea Tradfri (1x LED1924G9)
- Enddevices:
  - Xiaomi Aqara Sensors (2x WSDCGQ11LM)
  - Danfoss Ally (1x 014G2461)
  - Tradfri Remote (1x E1524/E1810)

- I already ruled out the Tradfri Router, by disconnecting it from power. I could still reproduce the problem.
- The last two devices I added were the Ally and the Aqara sensors, that was around 2021/12/06 and everything worked fine until mid-january

I can rule out power issues, the adapter is connected to a proper x86_64 server board with plenty of power budget and no other USB devices attached.

I can reproduce the problem with the following coordinator firmwares and zigbee2mqtt 1.23.0:
 - CC2652RB_coordinator_20210708.hex
 - CC2652RB_coordinator_20211217.hex
 - CC2652RB_coordinator_20220103.hex

### What did you expect to happen?

Devices should stay reachable

### How to reproduce it (minimal and precise)

Unclear. I can just wait and it will happen less than an hour into restarting zigbee2mqtt.

The earliest zigbee2mqtt version I saw this happening with was 1.22.2.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20220103

### Adapter

Slaesh's CC2652RB

### Debug log

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/8011624/z2m.log)

Looks like zigbee2mqtt is seeing my Valetudo robot (l10pro) via home assistant. I got that one on 2022/01/27, could that be a problem?

lux73, 2022-02-07:
Try using USB Extension Cable - Coordinator directly connected to USB Port ist not recommended

mweinelt, 2022-02-07:
The coordinator is connected via an 1.5m USB3 extension card.

mweinelt, 2022-02-07:
I enabled the availability feature just yesterday to check if I could further pinpoint the crash. I had it disabled before that, when the issue was already happening.

Koenkk, 2022-02-07:
Given that + that your setup doesn't have many devices, I don't it is a coordinator crash but rather a connection problem. It could be that another process disconnects the stick after which z2m cannot access it anymore. Things to check:

- https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#modemmanager-is-installed
- https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#hciuart-is-running
- disable HA usb discovery: https://www.home-assistant.io/integrations/usb/

mweinelt, 2022-02-07:
I have neither modemmanager (or NetworkManager) nor hciuart running on that machine.

I also do not have the usb component enabled, in fact I verified that my PYTHONPATH does not hold its dependencies `pyserial` and `pyudev`.

mweinelt, 2022-02-07:
There is not. The `dmesg` is particularly quiet.

I've used the linux audit framework to monitor access to the serial device, and noone but z2m accesses it.

mweinelt, 2022-02-07:
Still reproducible, and still z2m is the only user of the serial device.

```
# ausearch -k zigbee 
time->Mon Feb  7 20:02:05 2022
type=CONFIG_CHANGE msg=audit(1644260525.852:179): auid=4294967295 ses=4294967295 subj=kernel op=add_rule key="zigbee" list=4 res=1
----
time->Mon Feb  7 20:02:08 2022
type=PROCTITLE msg=audit(1644260528.037:306): proctitle=2F6E69782F73746F72652F367636386D7878327166697931697338643166646E326B32626776376661646C2D6E6F64656A732D31342E31382E332F62696E2F6E6F6465002F6E69782F73746F72652F367A787237317333686C6978726D6E787731326E3666613478686A38687137612D7A6967626565326D7174742D312E3233
type=PATH msg=audit(1644260528.037:306): item=0 name="/dev/ttyUSB0" inode=432 dev=00:05 mode=020660 ouid=0 ogid=27 rdev=bc:00 obj=unlabeled nametype=NORMAL cap_fp=0 cap_fi=0 cap_fe=0 cap_fver=0 cap_frootid=0
type=CWD msg=audit(1644260528.037:306): cwd="/var/lib/zigbee2mqtt"
type=SYSCALL msg=audit(1644260528.037:306): arch=c000003e syscall=257 success=yes exit=20 a0=ffffff9c a1=305b058 a2=181902 a3=0 items=1 ppid=1 pid=2820 auid=4294967295 uid=317 gid=317 euid=317 suid=317 fsuid=317 egid=317 sgid=317 fsgid=317 tty=(none) ses=4294967295 comm="zigbee2mqtt" exe="/nix/store/6v68mxx2qfiy1is8d1fdn2k2bgv7fadl-nodejs-14.18.3/bin/node" subj=kernel key="zigbee"
----
time->Mon Feb  7 20:02:26 2022
type=PROCTITLE msg=audit(1644260546.076:1019): proctitle=2F6E69782F73746F72652F367636386D7878327166697931697338643166646E326B32626776376661646C2D6E6F64656A732D31342E31382E332F62696E2F6E6F6465002F6E69782F73746F72652F367A787237317333686C6978726D6E787731326E3666613478686A38687137612D7A6967626565326D7174742D312E3233
type=PATH msg=audit(1644260546.076:1019): item=0 name="/dev/ttyUSB0" inode=432 dev=00:05 mode=020660 ouid=0 ogid=27 rdev=bc:00 obj=unlabeled nametype=NORMAL cap_fp=0 cap_fi=0 cap_fe=0 cap_fver=0 cap_frootid=0
type=CWD msg=audit(1644260546.076:1019): cwd="/var/lib/zigbee2mqtt"
type=SYSCALL msg=audit(1644260546.076:1019): arch=c000003e syscall=257 success=yes exit=20 a0=ffffff9c a1=408da98 a2=181902 a3=0 items=1 ppid=1 pid=3642 auid=4294967295 uid=317 gid=317 euid=317 suid=317 fsuid=317 egid=317 sgid=317 fsgid=317 tty=(none) ses=4294967295 comm="zigbee2mqtt" exe="/nix/store/6v68mxx2qfiy1is8d1fdn2k2bgv7fadl-nodejs-14.18.3/bin/node" subj=kernel key="zigbee"
```

Koenkk, 2022-02-08:
Could you check if you can also reproduce the problem when running on another system? E.g. a raspberry pi with just the dongle connected through a usb extension cable.

mweinelt, 2022-02-08:
Migrated the deployment to an APU2C4 that usually function as my router. It has two USB3 ports, and no other USB devices connected.

```
[root@io:/var/lib]# lsusb
Bus 001 Device 002: ID 0438:7900 Advanced Micro Devices, Inc. Root Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Koenkk, 2022-02-08:
@mweinelt strange, does it happen even when just 1 device is paired to it? (a router)

mweinelt, 2022-02-09:
Sadly also reproducible with only a single hue bulb paired.
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/8029226/z2m.log)

Koenkk, 2022-02-09:
`Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}` the extended pan ID id fishy here. Is this a restored network from a CC2531 backup? Maybe something is corrupt in the backup which causes the coordinator to crash. 

I suggest to do the following:
- Stop z2m
- Reflash the coordinator
- Remove the `database.db`, `state.json` and `coordinator_backup.json` from your data directory
- Remove `extended_pan_id` from your `configuration.yaml` (in case it is present)
- start z2m

mweinelt, 2022-02-09:
No, I only ever had this one ZigBee dongle.

```
[root@meduna:/var/lib/zigbee2mqtt]# systemctl stop zigbee2mqtt
[root@meduna:/var/lib/zigbee2mqtt]# ls
configuration.yaml  coordinator_backup.json  database.db  devices.yaml  log  state.json
[root@meduna:/var/lib/zigbee2mqtt]# rm database.db 
[root@meduna:/var/lib/zigbee2mqtt]# rm state.json 
[root@meduna:/var/lib/zigbee2mqtt]# rm coordinator_backup.json 
[root@meduna:/var/lib/zigbee2mqtt]# cd ~/cc2538-bsl
[root@meduna:/cc2538-bsl]# ./cc2538-bsl.py -p /dev/ttyUSB0 -e -w -v ../CC2652RB_coordinator_20220103.hex 
Opening port /dev/ttyUSB0, baud 500000
Reading data from ../CC2652RB_coordinator_20220103.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:21:CB:FD:AC
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F980
    Write done                                
Verifying by comparing CRC32 calculations.
    Verified (match: 0x4373e2e7)
[root@meduna:/cc2538-bsl]# systemctl restart zigbee2mqtt
```

Then paired the Tradfri bulb and a ~20 minutes later the same issue comes up again.

[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/8035493/z2m.log)

sorentorp, 2022-02-09:
following this thread. I have the same problem.

Koenkk, 2022-02-10:
Then it might be that somehow the firmware is not flashed correctly. I've seen this earlier with the CC2652RB, flashing on a different system might fix the issue:

- https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel (second point)
- https://github.com/Koenkk/zigbee2mqtt/issues/6302

Koenkk, 2022-02-11:
@mweinelt could you provide the logging when you run z2m with env variable `DEBUG=*`? Provide the logging starting from the point where you start z2m until it fails.

mweinelt, 2022-02-11:
New log is here!

https://gist.github.com/mweinelt/c8301d2753966b1362adb66e53600c74

Koenkk, 2022-02-12:
@mweinelt I've checked the logs but nothing suspicious there. TBH I've never seen such issue before, you will find many issues in the z2m issue tracker of the coordinator crashing but these are always due to the usual suspects (modemanager, bad usb cables, bad power supply, other software interfering). What is also different is that is crashes so fast and regular (10-20 min after uptime). I'm really starting to wonder if something is wrong with your USB stick, if I understand you correctly you never had a stable z2m setup with the CC2652RB right?

mweinelt, 2022-02-12:
I bought that stick in 2020/08 and it  had been working fine until now.

Koenkk, 2022-02-12:
@mweinelt ah, since it has been working well, what changed since then? Does downgrading to 1.21.0 fix the issue?

mweinelt, 2022-02-12:
Nice pointer regarding the serial driver, I have been on recent kernel verisons so this seems unrelated, luckily.

```
reboot   system boot  5.10.95          Thu Feb 10 00:11   still running
reboot   system boot  5.10.95          Mon Feb  7 20:01 - 00:10 (2+04:08)
reboot   system boot  5.10.95          Wed Feb  2 18:03 - 20:00 (5+01:57)
reboot   system boot  5.10.93          Thu Jan 27 01:28 - 18:02 (6+16:34)
reboot   system boot  5.10.93          Thu Jan 27 01:22 - 01:27  (00:04)
reboot   system boot  5.10.93          Wed Jan 26 01:43 - 01:21  (23:37)
reboot   system boot  5.10.91          Sun Jan 16 05:11 - 01:42 (9+20:30)
reboot   system boot  5.10.84          Fri Dec 17 23:51 - 05:10 (29+05:18)

mweinelt, 2022-02-12:
Failing with 1.22.1 as well. https://gist.github.com/mweinelt/6a95de1d2d0761f56ba007cd538b76ca

Checking if I can downgrade to 1.21.0 easily.

mweinelt, 2022-02-12:
Tested 1.21.0 and after ~24 minutes I ran into timeouts again. I have no debug log this time, but should you need one, please do tell.

Koenkk, 2022-02-13:
@mweinelt I expect the logs are the same, so no need to. Try downgrading things until it starts working again. (however since you downgraded to 1.21.0 already I don't expect z2m is the problem)

mweinelt, 2022-02-13:
Right. So my working theory is, that zigbee2mqtt or zigbee-herdsman does some intiation dance with the stick on startup and that it becomes unresponsive after ~20 minute, likely due to a crash or something similar. Does that sound plausible?

Either way, I'm going to try out another stick as soon as possible, because we basically ruled out zigbee2mqtt and the only other thing I took to my APU2 was the stick, so logically it must be that.

mweinelt, 2022-02-16:
Exchanged the stick with a colleague, who also has one of the 2652RB. It runs firmware 20210708 and after switching it to channel 20 and repairing my Tradfri bulb it kept working for over an hour.

Now waiting to hear back if my stick works for them, which would be surprising.

mweinelt, 2022-02-17:
Apparently it does, so it might be entirely related to the channel used, that channel 11 for some reason degraded greatly where I live. When I get my stick back I will try to confirm that theory.

mweinelt, 2022-02-19:
Got my stick back, left it running over night (~9h up) and everything still works after 9h. Will start repairing most devices after the weekend.

mweinelt, 2022-02-20:
Touchlink is still somewhat painful with 6 Hue bulbs. Often running into timeouts after fresh z2m startup.

```
Zigbee2MQTT:debug 2022-02-20 22:39:44: Received MQTT message on 'zigbee2mqtt/bridge/request/touchlink/scan' with data '{"transaction":"yfpuz-9","value":true}'
Zigbee2MQTT:info  2022-02-20 22:39:44: Start Touchlink scan
Zigbee2MQTT:error 2022-02-20 22:40:01: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'SRSP - AF - interPanCtl after 6000ms'
Zigbee2MQTT:debug 2022-02-20 22:40:01: Error: SRSP - AF - interPanCtl after 6000ms
    at Timeout._onTimeout (/nix/store/k444hgzggirrazqwcsz0jzwws9l83z5d-zigbee2mqtt-1.23.0/lib/node_modules/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
Zigbee2MQTT:info  2022-02-20 22:40:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"SRSP - AF - interPanCtl after 6000ms","status":"error","transaction":"yfpuz-9"}'
```

Currently at 3/6.

Koenkk, 2022-02-21:
@mweinelt touchlink crashing the coordinator was a known issue, it should be solved in the latest firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

mweinelt, 2022-02-21:
Indeed touchlink stability is much better, thank you!

mweinelt, 2022-02-21:
In general I think this issue was solved by changing the channel.

Now I have one problem left. One of the six hue bulbs I can touchlink reset, it will blink, it will join the network again, it works shortly, and then it stops working ("No network route"). I've retried this multiple times. Z2M flags the device as online after pairing, but after the next check it is offline. This has been going on since I paired it yesterday.

> Publish 'set' 'color' to '0x00178801088a3e7f' failed: 'Error: Command 0x00178801088a3e7f/11 lightingColorCtrl.moveToColor({"transtime":0,"colorx":41337,"colory":21173}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

 It is also not joined into a group definition, even though I set it up in `configuration.yaml` and that used to work before.

```yaml
groups:
  '1':
    devices:
    - '0x001788010889875c'
    - '0x00178801088a3360'
    - '0x00178801088a3e7f'
    - '0x00178801088a3478'
    - '0x0017880108898892'
    - '0x00178801088a3e7e'
    friendly_name: Stehlampe
    optimistic: true
    retain: false
    transition: 1
```

Guess I should provide another debug log next.

mweinelt, 2022-02-21:
I also noticed that while I have connection to that one bulb after pairing, I cannot control any of my zigbee devices. It feels like the one bulb is pulling my coordinator over to channel 11 when pairing.

mweinelt, 2022-02-21:
Log from when I touchlink reset the bulb on channel 25. How could I tell that it switches back to channel 20 before pairing?

https://gist.github.com/mweinelt/e1b6926dda4bfc31d217acbd8f07585a

closed 2022-02-22 by mweinelt

RubenKelevra, 2022-04-07:
Well, I guess I'm seeing the same issue here. On the same update to 1.22.x I saw some issues, several devices go offline, I have sometimes multiple seconds to tens of seconds delays on commands etc. A restart of z2m always fixed this for a day and up to 3 days.

So it's not a super heavy issue right now.

I didn't want to change a "working setup" so I did not update my Slaesh stick after receiving it like 2 years ago. But I did that today. The whole network is MUCH more responsive and haven't had a short dropout on any device yet.

So this issue was fixed for me by upgrading from 2020xxxx firmware to 20220219 on the stick.

Maybe this is helpful for you @leelooauto?