Koenkk/zigbee2mqtt #11272: [New device support]: Mercator Ikuu Double Power Point

palitu, 2022-02-07:
### Link

https://www.ikuu.com.au/product/double-power-point/

### Database entry

{"id":44,"type":"Router","ieeeAddr":"0x588e81fffed4940f","nwkAddr":46926,"manufId":4098,"manufName":"_TZ3210_raqjcxo5","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820,61184,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3210_raqjcxo5","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6,61184,3],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1644206933454,"defaultSendRequestWhen":"immediate"}

### Comments

Another Ikuu device, which could be quite popular in Australia.  This is a dual GPO (general purpose outlet), with power monitoring.

I have played almost entirely unsuccessfully with the base exportor.js for new Tuya devices.

Currently running the latest dev (1.23.0-dev commit: 5dd8873)

Any help would be much appreciated!

cheers

closed 2022-04-23 by github-actions[bot]

maxim75, 2022-07-31:
Any updates on this issue? Still shows "Unsupported"

[image]

TheBlackMini, 2022-12-01:
This device appears as unsupported like above, the USB variant seems like it got the support but this one got missed?