Koenkk/zigbee2mqtt #11273: Sonoff SNZB-02 no data

Agasfer, 2022-02-07:
### What happened?

Sonoff SNZB-02 successfully paired but there is no temperature, humidity, battery or any other data. All values are 'N/A'
With ZHA integration this sensor works well.
Tried to install z2m edge but with the same result.
Other devices (SNZB-01, Aqara Cube) works normally.

### What did you expect to happen?

Temperature, humidity and other data received properly.

### How to reproduce it (minimal and precise)

Pair Sonoff SNZB-02 

### Zigbee2MQTT version

1.23.0 

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

2022-02-07T07:33:38.733Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 5 - 255 - [0,0,0,4,128,0,88,0,0,0,0,0,16,1,4,1,0,4,0,2,0,0,25,0,2,0,5,2,5] - 112
2022-02-07T07:33:38.733Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - msgCbIncoming - {"srcaddr":0,"wasbroadcast":0,"clusterid":32772,"securityuse":0,"seqnum":88,"macdstaddr":0,"msgdata":{"type":"Buffer","data":[]}}
2022-02-07T07:33:38.734Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-02-07T07:33:38.738Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,101,80,0,0,9,98,26,0,0,198,16,121,37,0,75,18,0,0,0,0,0,0,0,0,0,11,132]
2022-02-07T07:33:38.738Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,101,80,0,0,9,98,26,0,0,198,16,121,37,0,75,18,0,0,0,0,0,0,0,0,0,11,132]
2022-02-07T07:33:38.738Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 3 - 5 - 80 - [0,0,9,98,26,0,0,198,16,121,37,0,75,18,0,0,0,0,0,0,0,0,0,11] - 132
2022-02-07T07:33:38.740Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extNwkInfo - {"shortaddr":0,"devstate":9,"panid":6754,"parentaddr":0,"extendedpanid":"0x00124b00257910c6","parentextaddr":"0x0000000000000000","channel":11}
2022-02-07T07:33:38.740Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-02-07T07:33:38.746Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - stackTune - {"operation":0,"value":20}
2022-02-07T07:33:38.747Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,15,0,20,56]
2022-02-07T07:33:38.753Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,15,0,111]
2022-02-07T07:33:38.754Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,15,0,111]
2022-02-07T07:33:38.754Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 15 - [0] - 111
2022-02-07T07:33:38.754Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - stackTune - {"value":0}
2022-02-07T07:33:38.755Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2022-02-07 11:33:38: Set transmit power to '20'
Zigbee2MQTT:info  2022-02-07 11:33:38: Currently 6 devices are joined:
Zigbee2MQTT:info  2022-02-07 11:33:38: 0x00124b0024cc4547 (0x00124b0024cc4547): SNZB-02 - SONOFF Temperature and humidity sensor (EndDevice)
Zigbee2MQTT:info  2022-02-07 11:33:38: Button26 (0x00124b0023207d11): SNZB-01 - SONOFF Wireless button (EndDevice)
Zigbee2MQTT:info  2022-02-07 11:33:38: MiCube (0x00158d000708191f): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
Zigbee2MQTT:info  2022-02-07 11:33:38: Movement (0x00158d0002ae86bd): PECMS01 - Perenio Motion sensor (EndDevice)
Zigbee2MQTT:info  2022-02-07 11:33:38: Door (0x00158d0001994207): PECWS01 - Perenio Door sensor (EndDevice)
Zigbee2MQTT:info  2022-02-07 11:33:38: Button89 (0x00124b0022fdefab): SNZB-01 - SONOFF Wireless button (EndDevice)
Zigbee2MQTT:info  2022-02-07 11:33:38: Zigbee: disabling joining new devices.
2022-02-07T07:33:38.785Z zigbee-herdsman:controller:log Disable joining
2022-02-07T07:33:38.793Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":1,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,2,2,10,0,0]}}
2022-02-07T07:33:38.796Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,1,0,30,6,0,25,2,2,10,0,0,23]
2022-02-07T07:33:38.805Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2022-02-07T07:33:38.805Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2022-02-07T07:33:38.806Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2022-02-07T07:33:38.806Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2022-02-07T07:33:38.807Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-02-07T07:33:38.880Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,1,52]
2022-02-07T07:33:38.881Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,1,52]
2022-02-07T07:33:38.881Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,1] - 52
2022-02-07T07:33:38.882Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":1}
2022-02-07T07:33:38.882Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-02-07T07:33:39.011Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":0,"tcsignificance":0}
2022-02-07T07:33:39.011Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,0,0,26]
2022-02-07T07:33:39.027Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3,69,182,0,0,0,240,254,10,69,255,0,0,0,54,128,0,89,0,0,0,95]
2022-02-07T07:33:39.028Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3,69,182,0,0,0,240,254,10,69,255,0,0,0,54,128,0,89,0,0,0,95]
[... 34 lines cut ...]
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'homeassistant/sensor/0x00158d000708191f/action_side/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/MiCube/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d000708191f"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"MiCube"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/MiCube","name":"MiCube action side","state_topic":"zigbee2mqtt/MiCube","unique_id":"0x00158d000708191f_action_side_zigbee2mqtt","value_template":"{{ value_json.action_side }}"}'
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'homeassistant/sensor/0x00158d000708191f/action_to_side/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/MiCube/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d000708191f"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"MiCube"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/MiCube","name":"MiCube action to side","state_topic":"zigbee2mqtt/MiCube","unique_id":"0x00158d000708191f_action_to_side_zigbee2mqtt","value_template":"{{ value_json.action_to_side }}"}'
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'homeassistant/sensor/0x00158d000708191f/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/MiCube/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d000708191f"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"MiCube"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/MiCube","name":"MiCube action","state_topic":"zigbee2mqtt/MiCube","unique_id":"0x00158d000708191f_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'homeassistant/sensor/0x00158d000708191f/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/MiCube/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d000708191f"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"MiCube"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/MiCube","name":"MiCube linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/MiCube","unique_id":"0x00158d000708191f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
[... cut ...]
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'zigbee2mqtt/Movement', payload '{"battery":26,"battery_low":false,"last_seen":"2022-02-07T07:23:50.395Z","linkquality":null,"occupancy":false,"tamper":false}'
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'zigbee2mqtt/Door', payload '{"battery":19,"battery_low":false,"contact":true,"last_seen":"2022-02-07T07:20:23.596Z","linkquality":null,"tamper":false}'
Zigbee2MQTT:info  2022-02-07 11:33:39: MQTT publish: topic 'zigbee2mqtt/Button89', payload '{"action":null,"battery":100,"last_seen":"2022-02-07T06:56:49.464Z","linkquality":null,"voltage":3100}'

linker3000, 2022-03-13:
I also had this issue. I was running 1.21.1 which seemed to be the last version that worked with this sensor, and then I upgraded to 1.21.2 for Sonoff power switch (S26R2ZB) support and everything was still fine, plus the power switch worked.

I then upgraded the node-red dashboard, restarted node-red and this sensor stopped reporting any data. In the node object I could only select 'Complete state payload', and triggering the associated GET node just returned 'null'.

Things tried: 

Upgrading to the latest version (1.24.0) - no change

Regressing to 1.21.2 - no change

Regressing to 1.21.1 - Sensor was configurable and worked, but the power switch was again not supported.

Upgraded from 1.21.1 to 1.21.2 - SNZB-02 working, S26R2ZB working.

Through all these steps I tried removing the sensor and re-associating it, with no luck until I went back to 1.21.1 then forwards to 1.21.2

eloo, 2022-04-18:
just run again into this issue and the following comment helped me
https://github.com/Koenkk/zigbee2mqtt/issues/8971#issuecomment-1030698488

maybe this can be added to the docs of the device?

Koenkk, 2022-04-19:
This bug has been fixed in z2m 1.25.0, so upgrade to 1.25.0 and repair the device.

eloo, 2022-04-19:
@Koenkk 
unfortunately i'm running 1.25.0 and run into this issue.. so it looks like the fix is working as expected :(

Koenkk, 2022-04-19:
After upgrading, did you force remove the device via the frontend, restart z2m and repair the device?

eloo, 2022-04-19:
yes. remove, restart, pair has fixed the issues.. 
and thats what is mentioned in the comment i have linked. 

so this should be added to the docs IMHO

closed 2022-05-27 by github-actions[bot]

lazou, 2022-07-14:
@eloo Since I also had this issue and could not find the solution after looking into the docs, here the PR to add the instructions: #1480

swampjedi, 2023-03-01:
I am having this issue on 1.30.1 using EZSP v8 rev 6.10.3.0 build 297. Force drop / restart / pair instructions haven't been successful and often results in an "unsupported device".

dmsergeevich63, 2023-04-30:
@swampjedi I also encounter the problem of pairing the sensor with a stick, but based on CC2652P. Have you managed to figure out what the problem is?

nebhead, 2023-11-19:
FWIW - Also ran into issues, but just did a force remove, restart, then pair and it did come back.  It took a long time to update the data, but it did finally work.

ThierryDez, 2024-04-14:
I have the same problem.

My configuration had been working for several months but I update it regularly.
I had to change the battery in my SNZB-02
and since then, it hasn't been sending any information back.

https://github.com/Koenkk/zigbee2mqtt.io/commit/fd2285ca15ec45d394502af866d9a977e9791cef
1. Force remove the device via the frontend
2. Restart Zigbee2MQTT
3. Repair the device
doesn't work for me!

Worse, the information that comes back never seems to be the same.
Now I only have Linkquality et Battery, Temperature, Humidity, Voltage = N/A
"last seen" increases since the pairing date and "availibiliy" = active

what I've been doing lately:
updated the system in
Core 2024.4.3
Supervisor 2024.04.0
Operating System 12.2
Interface utilisateur 20240404.2
I've added the "last seen" and "availability" columns

my configuration ZB :
Bridge : Zigbee2MQTT
Micrologiciel : 1.36.1
Matériel : EZSP v8 6.10.3.0 build 297
[image]

ThierryDez, 2024-04-16:
> I have the same problem.
> 
> My configuration had been working for several months but I update it regularly. I had to change the battery in my SNZB-02 and since then, it hasn't been sending any information back.
> 
> [Koenkk/zigbee2mqtt.io@fd2285c](https://github.com/Koenkk/zigbee2mqtt.io/commit/fd2285ca15ec45d394502af866d9a977e9791cef)
> 
> 1. Force remove the device via the frontend
> 2. Restart Zigbee2MQTT
> 3. Repair the device
> 
> doesn't work for me!
> 
> Worse, the information that comes back never seems to be the same. Now I only have Linkquality et Battery, Temperature, Humidity, Voltage = N/A "last seen" increases since the pairing date and "availibiliy" = active
> 
> what I've been doing lately: updated the system in Core 2024.4.3 Supervisor 2024.04.0 Operating System 12.2 Interface utilisateur 20240404.2 I've added the "last seen" and "availability" columns
> 
> my configuration ZB : Bridge : Zigbee2MQTT Micrologiciel : 1.36.1 Matériel : EZSP v8 6.10.3.0 build 297 ![2024-04-14 184320](https://private-user-images.githubusercontent.com/6362466/322293094-1cef0e41-47d8-4d18-88f2-8fade2e01250.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxNjgwMDMsIm5iZiI6MTcxMzE2NzcwMywicGF0aCI6Ii82MzYyNDY2LzMyMjI5MzA5NC0xY2VmMGU0MS00N2Q4LTRkMTgtODhmMi04ZmFkZTJlMDEyNTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTVUMDc1NTAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDgyZjFkYjM2MmNjMmM4MjU3YzFhMzBmNDA1ZTgxODcxMjEyZDQzZGU1NjFiZGY3YTdiMzAzODBjMjBiNWE5NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.8JCML05SdCJd2SsQS5INmFaz2b418FSkRV0WGZPuDK0)

persevering, I took the time
and by doing it properly / quietly, the process works.
That's why I'm leaving my message: apply and take the time **between each step** and it works.

MordechaiN, 2024-12-21:
same issue.. all the remove>resetart>add didnt worked for me..
battery is good and nothing happend