Koenkk/zigbee2mqtt #11283: Zigbee model 'SML003' is NOT supported | Hue Motion Sensor

peterge1998, 2022-02-07:
### What happened?

When i try to join my hue sensor this error shows up:
[image]

Sometimes it does connect but it does not report any data:
[image]

### What did you expect to happen?

Connect and report data

### How to reproduce it (minimal and precise)

Join a Hue motion sensor

Coordinator type
    zStack3x0

Coordinator revision

Frontend version
    0.3.76

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

20211217

### Adapter

Slaeshs Zigbee Stick

### Debug log

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[16:39:34] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[16:39:35] INFO: MQTT available, fetching server detail ...
[16:39:35] INFO: MQTT server settings not configured, trying to auto-discovering ...
[16:39:35] INFO: Configuring 'mqtt://core-mosquitto:1883' mqtt server
[16:39:36] INFO: MQTT credentials not configured, trying to auto-discovering ...
[16:39:36] INFO: Configuring'addons' mqtt user
[16:39:36] INFO: Previous config file found, checking backup
[16:39:36] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[16:39:36] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[16:39:36] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
Zigbee2MQTT:warn  2022-02-07 16:40:56: Device '0x001788010bcdad0c' with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:warn  2022-02-07 16:41:00: Device '0x001788010bcdad0c' left the network
Zigbee2MQTT:warn  2022-02-07 16:44:39: Device '0x001788010bcdad0c' with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:warn  2022-02-07 16:44:47: Device '0x001788010bcdad0c' left the network
Zigbee2MQTT:warn  2022-02-07 16:45:03: Device '0x001788010bcdad0c' with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:warn  2022-02-07 16:46:35: Received message from unsupported device with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-02-07 16:46:35: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2022-02-07 16:46:45: Received message from unsupported device with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-02-07 16:46:45: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2022-02-07 16:46:55: Received message from unsupported device with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-02-07 16:46:55: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2022-02-07 16:47:05: Received message from unsupported device with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-02-07 16:47:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2022-02-07 16:47:15: Received message from unsupported device with Zigbee model 'SML003' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-02-07 16:47:15: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2022-02-07:
Supported in z2m 1.23.0

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2022-02-07 by Koenkk

peterge1998, 2022-02-07:
> Supported in z2m 1.23.0
> 
> If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

Sry did not notice, it's working now.