Koenkk/zigbee2mqtt #11285: [New device support]: Aqara E1 Water Leak Sensor SJCGQ13LM

cwm77, 2022-02-07:
### Link

https://www.aqara.com/cn/productInner/Water%20Leak%20Sensor%20E1?on=0

### Database entry

"lumi.flood.acn001"

### Comments

Is there any possibility that this leak sensor will support sometime in the future ? The two predecessor models are already supported.

Koenkk, 2022-02-07:
@z-x-f might be on it already 😄

Otnow, 2022-02-07:
Try this [external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device), made similar to the [SJCGQ12LM](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3a581af48ba7d97444fdc71f6e384a9f34fb8bb3/devices/xiaomi.js#L964) model:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.flood.acn001'],
    model: 'SJCGQ13LM',
    vendor: 'Xiaomi',
    description: 'Aqara E1 water leak sensor',
    meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
    fromZigbee: [fz.xiaomi_battery, fz.ias_water_leak_alarm_1],
    toZigbee: [],
    exposes: [e.battery(), e.water_leak(), e.battery_low(), e.tamper(), e.battery_voltage()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

closed 2022-03-18 by github-actions[bot]