Koenkk/zigbee2mqtt #11286: TS0601 _TZE200_fzo2pocs - position:100 doesn't work correctly

dmitry-gerasimov, 2022-02-07:
### What happened?

Setting `{"position": 100}` completely closes the curtain.

### What did you expect to happen?

`{"position": 100}` should open the curtain all the way. All other positions work as expected. {"position": 0} - completely closes, `{"position": 10}` - slightly opens, etc.  

The only way to fully open the curtain is to send `{"state": "OPEN"}`

### How to reproduce it (minimal and precise)

Publish `{"position": 100}` to `zigbee2mqtt/Tuya motor/set`. Or just move slider to 100 in UI.

### Zigbee2MQTT version

1.18.1

### Adapter firmware version

not specified

### Adapter

zStack12 (WBE2R-R-ZIGBEE for WirenBoard controller)

### Debug log

```
debug 2022-02-07 23:00:42: Received MQTT message on 'zigbee2mqtt/Tuya motor/set' with data '{"position":100}'
debug 2022-02-07 23:00:42: Publishing 'set' 'position' to 'Tuya motor'
debug 2022-02-07 23:00:42: Received Zigbee message from 'Tuya motor', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":0,"transid":10}' from endpoint 1 with groupID 0
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264042381,"linkquality":15,"motor_speed":true,"position":100,"running":true,"state":"OPEN"}'
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'true'
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '100'
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'OPEN'
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264042381'
info  2022-02-07 23:00:42: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '15'
debug 2022-02-07 23:00:44: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264044485,"linkquality":18,"motor_speed":true,"position":0,"running":false,"state":"CLOSE"}'
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'false'
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264044485'
info  2022-02-07 23:00:44: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '18'
debug 2022-02-07 23:00:45: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
debug 2022-02-07 23:00:45: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264045148,"linkquality":39,"motor_speed":true,"position":0,"running":false,"state":"CLOSE"}'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264045148,"linkquality":39,"motor_speed":true,"position":0,"running":false,"state":"CLOSE"}'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'false'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'false'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264045148'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264045148'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '39'
info  2022-02-07 23:00:45: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '39'
debug 2022-02-07 23:00:46: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
debug 2022-02-07 23:00:46: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
debug 2022-02-07 23:00:46: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
debug 2022-02-07 23:00:46: Received Zigbee message from 'Tuya motor', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
info  2022-02-07 23:00:46: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264046391,"linkquality":36,"motor_speed":true,"position":0,"running":false,"state":"CLOSE"}'
info  2022-02-07 23:00:46: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264046391,"linkquality":36,"motor_speed":true,"position":0,"running":false,"state":"CLOSE"}'
info  2022-02-07 23:00:46: MQTT publish: topic 'zigbee2mqtt/Tuya motor', payload '{"last_seen":1644264046391,"linkquality":36,"motor_speed":true,"position":0,"running":false,"state":"CLOSE"}'
[... 2 lines cut ...]
info  2022-02-07 23:00:46: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'false'
info  2022-02-07 23:00:46: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'false'
info  2022-02-07 23:00:46: MQTT publish: topic 'zigbee2mqtt/Tuya motor/running', payload 'false'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/position', payload '0'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/state', payload 'CLOSE'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/motor_speed', payload 'true'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264046391'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264046391'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264046391'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/last_seen', payload '1644264046391'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '36'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '36'
info  2022-02-07 23:00:47: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '36'
info  2022-02-07 23:00:48: MQTT publish: topic 'zigbee2mqtt/Tuya motor/linkquality', payload '36'

```

Koenkk, 2022-02-08:
Please check with the latest z2m.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

dmitry-gerasimov, 2022-02-08:
Unfortunately, I can't easily upgrade since z2m comes as a part of my controller (wirenboard) and this would require updating the firmware of the zigbee stick, etc. 

Is there a way to try fixing this issue with template tweaks or something while still staying on 1.18.1?

Koenkk, 2022-02-08:
>  and this would require updating the firmware of the zigbee stick, etc.

latest z2m works with any firmware ever released by me.

closed 2022-03-19 by github-actions[bot]