Koenkk/zigbee2mqtt #11288: "GENERATE" option doesn't work on addon

OldShaterhan, 2022-02-07:
### What happened?

I try to set `advanced.network` to `GENERATE`, but I get an error:
`Nie udało się zapisać konfiguracji dodatku, Invalid list for option 'network_key' in Zigbee2mqtt (45df7312_zigbee2mqtt). Got {'data_path': '/config/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': False, 'mqtt': {'base_topic': 'zigbee2mqtt'}, 'serial': {'port': 'tcp://xxx.xxx.xxx.xxx:6638'}, 'advanced': {'log_level': 'warn', 'pan_id': 6754, 'channel': 11, 'network_key': 'GENERATE', 'availability_blocklist': [], 'availability_passlist': []}, 'device_options': {}, 'blocklist': [], 'passlist': [], 'queue': {}, 'frontend': {'port': 8099}, 'ex...` when I try to save to it

### What did you expect to happen?

Trying to change configuration (note - addon was already run, but no real Zigbee devices except coordinator are added right now)

### How to reproduce it (minimal and precise)

Try to change network key to `GENERATE`

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

 20211217

### Adapter

CC2652P

Koenkk, 2022-02-08:
Please create issues related to the addon here: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues

closed 2022-02-08 by Koenkk

OldShaterhan, 2022-02-08:
Okay, I moved it there.