Koenkk/zigbee2mqtt #11298: [New device support]: Tuya Blind Motor: TM616EGBZTN

milobt, 2022-02-08:
### Link

https://www.aliexpress.com/item/1005003404420688.html?spm=a2g0o.9042311.0.0.3a124c4dNjHzmy

### Database entry

{"id":46,"type":"EndDevice","ieeeAddr":"0xb4e3f9fffea70446","nwkAddr":3715,"manufId":4098,"manufName":"_TZE200_gubdgai2","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_gubdgai2","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1644315912222,"defaultSendRequestWhen":"immediate"}

### Comments

Zigbee model: TS0601
Model: TM616EGBZTN
Zigbee Manufacturer: _TZE200_gubdgai2
Manufacturer: EGBZTN

Interview: complete
Blind control: UP, DOWN, POSITION.
Link quality: works
Motor speed: not working

[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID
    model: 'TM616EGBZTN', // Vendor model number, look on the device for a model number
    vendor: 'EGBZTN', // Vendor of the device (only used for documentation and startup logging)
    manufacturerName: '_TZE200_gubdgai2',
    description: 'Roller Blind Motor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
    toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
    exposes: [
        e.cover_position().setAccess('position', ea.STATE_SET),
        exposes.composite('options', 'options')
            .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(255)
                .withDescription('Motor speed'))],
};

module.exports = definition;
```

tenkaibr, 2022-02-23:
I ve the zigbee model

Zigbee model: TS0601
Model: M515EGBZTN
Zigbee Manufacturer: _TZE200_gubdgai2
Manufacturer: EGBZTN

Interview: complete
Blind control: UP, DOWN, POSITION.
Link quality: works
Motor speed: not working (does it have?)
Cant set limits

External converter:

`const fz = require( 'zigbee-herdsman-converters/converters/fromZigbee' );
const tz = require( 'zigbee-herdsman-converters/converters/toZigbee' );
const exposes = require( 'zigbee-herdsman-converters/lib/exposes' );
const reporting = require( 'zigbee-herdsman-converters/lib/reporting' );
const extend = require( 'zigbee-herdsman-converters/lib/extend' );
const tuya = require( 'zigbee-herdsman-converters/lib/tuya' );
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [
            'owvfni3\u0000', 'owvfni3', 'u1rkty3', 'aabybja', // Curtain motors
            'mcdj3aq', 'mcdj3aq\u0000', // Tubular motors
            ],
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE200_gubdgai2' },
            ],
        model: 'TS0601_cover',
        vendor: 'TuYa',
        description: 'Curtain motor/roller blind motor/window pusher/tubular motor',
        whiteLabel: [
            {vendor: 'Tuya', model: 'M515EGZT'},
            ],
        fromZigbee: [
            fz.tuya_cover,
            fz.ignore_basic_report,
            ],
        toZigbee: [
            tz.tuya_cover_control,
            tz.tuya_cover_options,
            ],
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            ],
    };

module.exports = definition;`

ianByrne, 2022-03-26:
My TS0601 is still not yet supported - commenting to remove the `stale` label

closed 2022-04-03 by github-actions[bot]