Koenkk/zigbee2mqtt #11311: Unresolved Error: SRSP - SYS - ping after 6000ms with Sonoff Zigbee 3.0 Dongle

afaucogney, 2022-02-08:
### What happened?

I'm in a phase of migrating my adaptor from CC2530 to Sonoff Dongle.
I succeed to flash thank to ZigStarZW, I also write the CC2530 IEEE into the Sonoff stick

```
➜  /  ls -l /dev/serial/by-id
total 0
lrwxrwxrwx    1 root     root            13 Oct 12 17:24 usb-0658_0200-if00 -> ../../ttyACM0lrwxrwxrwx    1 root     root            13 Oct 12 17:24 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_d0f394ad7829ec11ad666d7840c9ce8d-if00-port0 -> ../../ttyUSB0

The plungin doesn't start

### What did you expect to happen?

To start

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

CC1352P2_CC2652P_launchpad_router_20210128.hex

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_

### Debug log

```
debug 2022-02-08 22:06:38: Loaded state from file /share/zigbee2mqtt/state.json
info  2022-02-08 22:06:38: Logging to console and directory: '/share/zigbee2mqtt/log/2022-02-08.22-06-33' filename: log.txt
debug 2022-02-08 22:06:38: Removing old log directory '/share/zigbee2mqtt/log/2022-02-08.22-00-55'
info  2022-02-08 22:06:38: Starting Zigbee2MQTT version 1.23.0 (commit #unknown)
info  2022-02-08 22:06:38: Starting zigbee-herdsman (0.14.10)
debug 2022-02-08 22:06:38: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
error 2022-02-08 22:06:58: Error while starting zigbee-herdsman
error 2022-02-08 22:06:58: Failed to start zigbee
error 2022-02-08 22:06:58: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2022-02-08 22:06:58: Exiting...
error 2022-02-08 22:06:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)

afaucogney, 2022-02-08:
Here is my config:

```
data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: ***:1883
  user: ****
  password: ***
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  availability_blacklist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: false
socat:
  enabled: false
  master: pty,raw,echo=0,link=/dev/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
  restartdelay: 1
  initialdelay: 1
ban: []
whitelist: []

sorentorp, 2022-02-09:
I have the same problem... made some comments here: https://github.com/Koenkk/zigbee2mqtt/issues/2308#issuecomment-1034270906

Minims, 2022-02-10:
Same issue since 2 days. 
I have a CC2531. Tested with Home 1.2 and source routing firmware, same issue. 
I have not change anything in my network. Both Stable and Edge version have the issue. 
Can it it be a raspbian upgrade issue on USB management ?  I have a PI4

orxaz, 2022-02-11:
My case was for a failing stick, don't know why the first time I tested with the other one it has failed too, but now is working well with the second one.

joohann, 2022-02-11:
> Same issue since 2 days.
> I have a CC2531. Tested with Home 1.2 and source routing firmware, same issue.
> I have not change anything in my network. Both Stable and Edge version have the issue.
> Can it it be a raspbian upgrade issue on USB management ? I have a PI4

Same here just out of the blue.

matteopiccioni, 2022-02-21:
Same issue since 4-6 days.
Some days ago I add a usb-hub connected to raspy, maybe is related?

My CC2531 is still connected to raspy (no hub)

tlse-vins, 2022-03-25:
I have the same issue since 10 days.
Several users of Gladys assistant also have the problem.

FrandsAndersen, 2022-04-09:
I have had the same problem for a month.

I have tried:
- Flashing firmware on Sonoff USB stick
- Stop zigbee2mqtt, replug usb stick, start zigbee2mqtt
- Reinstall zigbee2mqtt
- Restarting raspberry pi running home assistant
- Restarting only home assistant
- Change pan_id
- Change channel
- Running ls -l /dev/serial/by-id/ on rpi to make sure correct port it used

Nothing works!

My config:

`data_path: /config/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.1.109:1883
  user: ****
  password: ****
serial:
  port: /dev/ttyUSB0
advanced:
  log_level: warn
  pan_id: 6760
  channel: 25
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8080
experimental:
  new_api: true
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
`

MihailRizov, 2022-04-12:
I am also having the same issue. I brought new Sonoff stick, flashed it with "CC1352P2_CC2652P_launchpad_router_20220125" firmware. My version of zigbee2mqtt is 1.25.0

FrandsAndersen, 2022-04-12:
> I am also having the same issue. I brought new Sonoff stick, flashed it with "CC1352P2_CC2652P_launchpad_router_20220125" firmware. My version of zigbee2mqtt is 1.25.0

Exactly the same version og zigbee2mqtt that I use, and also the same firmware.

renatoccosta, 2022-04-23:
I have the same problem. zigbee2mqtt is 1.25.0-1. Router firmware CC1352P2_CC2652P_launchpad_coordinator_20220219

renatoccosta, 2022-04-23:
Just rebooted everything and it seems to be working fine (for now)

dewet22, 2022-04-23:
I've seen an uptick in this happening as well, for at least the past few months. It'll go days, sometimes a week or two, without incident, but mostly I need to restart z2m every day or two when it gets stuck with endless SRSP errors. Usually (9 out of 10 times, roughly) just restarting the docker container works, but rarely it doesn't seem to recover until restarting the USB device (either rebooting or pulling & replugging). This happens with two different CC2652P-based sticks: zzh on zStack3x0-20220219 and Sonoff on factory-default 20210708.

Is there any way to just have z2m treat SRSP errors as fatal so we can auto-restart the service as necessary?

radinsky, 2022-05-04:
recently I see the same (i suspect the supervisor upgrade to 2022.5 - not sure)

dewet22, 2022-05-04:
> i suspect the supervisor upgrade to 2022.5

FYI I'm not using the supervisor, I run everything (HA, z2m, etc.) in containers.

noderat, 2022-05-05:
I've been bitten by this now as well, my network and all devices work fine for about 5-10 minutes and then everything fails. I'm running on amd64 with home assistant is install. I've tried different USB ports, and I've tried all 6 of my Sonoff dongles as well as my backup CC2652P Circuitsetup.us dongle all running the same latest firmware available and all tested using default gain setting of 5 dBm

Each device was reflashed to a blank state prior to test, host PC running home assistant OS (latest release 2022.4 and 2022.5 releases) freshly rebooted and zigbee2mqtt was allowed to restore the network to the adapter on startup.

I have disabled usb powersave through cmdline.txt on my boot partition and confirmed it has a value of `-1`

eladiogalvez, 2022-05-08:
Here same issue in a OrangePi and  CC2652R (VS201 - Blue led):

- Linux 4.19.62-sunxi #5.92 SMP Wed Jul 31 22:07:23 CEST 2019 armv7l GNU/Linux
- Docker version 20.10.15, build fd82621
- Homeassistant core-2022.3.5
- Homeassistant supervisor-2022.05.0

Some tests:
- With cc2652r and firmware CC2652R_coordinator_20220219 similar timeouts errors

- With cc2652r and firmware CC2652R_coordinator_20210120 try to book, but after some logs about information communicated with Zigbee devices the usb coordinator device is restarted and also service.

Finally I tried to install all needed software to boot only zigbee2mqtt (dev branch some hours from last commit) in a windows, connected dongle to the windows pc, copied all needed db/configurations files from OrangePi updating port device to com and Mqtt server, and the result:

- With cc2652r and firmware CC2652R_coordinator_20220219 => boots correctly, read information but when try to send for example an open blind, is not working and result in timeouts in communication with devices. This issue I suppose that is related with firmware.

- With cc2652r and firmware 20210120  => work stable and correctly

eladiogalvez, 2022-05-08:
I upgraded orangepi where is working hasssio-supervisor from Debian 10 (Buster) to Debian 11 (Bullseye) and component "Zigbee2mqtt Edge" is working again correctly with last firmware "CC2652R_coordinator_20220219".

closed 2022-06-15 by github-actions[bot]

lazarvgd, 2022-09-05:
I have exact same problem with Sonoff zigbee adapter. Is there a fix for this?

klemic, 2022-09-13:
after new installation of my IOBROKER and pivCCU on bullseye lite I also have these error messages, spent hours but no idea how to solve
`<html>
<body>

zigbee.0 | 2022-09-13 10:53:35.280 | error | Error herdsman start
-- | -- | -- | --
zigbee.0 | 2022-09-13 10:53:35.279 | error | Failed to start Zigbee
zigbee.0 | 2022-09-13 10:53:35.277 | error | Starting zigbee-herdsman problem : "Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'"
zigbee.0 | 2022-09-13 10:53:34.938 | info | Installed Version: ioBroker/ioBroker.zigbee#1fff5415828c773bb5d1dd3f54bbffd39b0483c8
zigbee.0 | 2022-09-13 10:53:34.909 | info | Starting Zigbee npm ...
zigbee.0 | 2022-09-13 10:53:34.908 | info | Try to reconnect. 1 attempts left
zigbee.0 | 2022-09-13 10:53:24.904 | error | Error herdsman start
zigbee.0 | 2022-09-13 10:53:24.903 | error | Failed to start Zigbee
zigbee.0 | 2022-09-13 10:53:24.902 | error | Starting zigbee-herdsman problem : "Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)"

</body>
</html>`

artem-zinnatullin, 2023-03-25:
Getting same issue with zigbee2mqtt `1.30.2` running in k8s on OrangePi 5 and providing access to Sonoff USB Zigbee stick via https://github.com/squat/generic-device-plugin

```js
Creating configuration file...
Zigbee2MQTT:info  2023-03-25 04:05:42: Logging to console and directory: '/app/data/log/2023-03-25.04-05-42' filename: log.txt
Zigbee2MQTT:info  2023-03-25 04:05:42: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea)
Zigbee2MQTT:info  2023-03-25 04:05:42: Starting zigbee-herdsman (0.14.96)
Zigbee2MQTT:error 2023-03-25 04:06:03: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-03-25 04:06:03: Failed to start zigbee
Zigbee2MQTT:error 2023-03-25 04:06:03: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-03-25 04:06:03: Exiting...
Zigbee2MQTT:error 2023-03-25 04:06:03: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

However right before switching to Zigbee2Mqtt I was able to set up ZHA with Homeassistant running on same hardware in k8s (privileged mode) and it was running totally fine, so it seems to be an issue of Zigbee2Mqtt so far judging by other comments here..

I'll be trying to run it in privileged mode in k8s, will report back if it works.

artem-zinnatullin, 2023-03-25:
Got it working! Had to specify `adapter` in `configuration.yaml` for my Sonoff Zigbee adapter:

```yaml
serial:
      port: /dev/ttyACM0
      adapter: ezsp
```

Works in k8s in both privileged mode and through https://github.com/squat/generic-device-plugin!

bastienjalbert, 2023-12-13:
> Got it working! Had to specify `adapter` in `configuration.yaml` for my Sonoff Zigbee adapter:
> 
> ```yaml
> serial:
>       port: /dev/ttyACM0
>       adapter: ezsp
> ```
> 
> Works in k8s in both privileged mode and through https://github.com/squat/generic-device-plugin!

Real fix here !

Redforce04, 2024-08-13:
For people finding this in 2024 - adapter `ezsp` has been deprecated and you should now use `ember`.

maximouche, 2024-11-05:
wow. Thanks !!!!
It took me days to finally find this solution.
I personally have a zigate, so I added `adapter: zigate` according to the doc here : https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html
The list of accepted values is : 
```
    # Optional: adapter type, specify if you are experiencing startup problems (default: shown below, options: zstack, deconz, ember, zigate)
```

My conf looks like this : 
```
homeassistant: true
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
serial:
  port: /dev/ttyUSB0
  adapter: zigate
...
```

I was missing the adapter line before and it was not working.
Now it does.