Koenkk/zigbee2mqtt #11312: IKEA LED1949C5N OTA firmware update failing

ralphhendriks, 2022-02-08:
### What happened?

I bought a few IKEA LED1949C5N devices (TRADFRI LED bulb E14 470 lumen). I tried to update firmware through the frontend _Check for new updates button_ on the _OTA_ tab. After a short while a message is shown

> Failed to check if update available for 'XXX' (No image available for imageType '8708')

I am **not** using the IKEA TRADFRI test URL.

### What did you expect to happen?

I expected the firmware check to complete successfully, either showing a new firmware available or indicating the device is already running the latest firmware.

### How to reproduce it (minimal and precise)

In the web UI, click _Check for new updates_ on the device.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

### Debug log

```
Zigbee2MQTT:debug 2022-02-08 21:53:31: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"XXX","transaction":"u6bgu-1"}'
Zigbee2MQTT:info  2022-02-08 21:53:31: Checking if update available for 'XXX'
Zigbee2MQTT:debug 2022-02-08 21:53:31: Check if update available for '0xabcdabcdabcdabcd' (TRADFRIbulbE14WScandleopal470lm)
Zigbee2MQTT:debug 2022-02-08 21:53:31: Using endpoint '1'
Zigbee2MQTT:debug 2022-02-08 21:53:31: Received Zigbee message from 'XXX', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-08 21:53:31: Device 'XXX' requested OTA
Zigbee2MQTT:debug 2022-02-08 21:53:31: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":8708,"fileVersion":65586}'
Zigbee2MQTT:info  2022-02-08 21:53:32: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"XXX"},"error":"Failed to check if update available for 'XXX' (No image available for imageType '8708')","status":"error","transaction":"u6bgu-1"}'
Zigbee2MQTT:error 2022-02-08 21:53:32: Failed to check if update available for 'XXX' (No image available for imageType '8708')
Zigbee2MQTT:debug 2022-02-08 21:53:32: Error: No image available for imageType '8708'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/lib/ota/tradfri.js:18:15)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:265:18)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:256:23)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:193:39)
Zigbee2MQTT:debug 2022-02-08 21:53:34: Received Zigbee message from 'XXX', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65586,"imageType":8708,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-02-08 21:53:34: Device 'XXX' requested OTA
Zigbee2MQTT:debug 2022-02-08 21:53:34: Check if update available for '0xabcdabcdabcdabcd' (TRADFRIbulbE14WScandleopal470lm)
Zigbee2MQTT:debug 2022-02-08 21:53:34: Failed to check if update available for 'XXX' (No image available for imageType '8708')
Zigbee2MQTT:info  2022-02-08 21:53:34: MQTT publish: topic 'zigbee2mqtt/XXX', payload '{"brightness":254,"color_mode":"color_temp","color_temp":380,"linkquality":112,"power_on_behavior":"previous","state":"ON","update":{"state":"idle"},"update_available":null}'
```

MattWestb, 2022-02-09:
The device image type 0x2204 / 8708 have being seen in the test feed bore IKEA was closing / moving (little lesser then one year ago) it but i have never seen seen it in the production feed. `10080506-zingo_kt_bulb_hwpwmcs_ws-1.0.032.ota.ota.signed`
Its one 3 gen devices so very likely its coming one update snore or later but for the moment its not released by IKEA.

ralphhendriks, 2022-02-12:
Thanks @MattWestb for confirming this is a genuine error message and not something in the update mechanism or device template in zigbee2mqtt going wrong. I will stay patient and try again at a later time.

closed 2022-02-12 by ralphhendriks

MattWestb, 2022-02-21:
I have found in the IKEA production feed:
| OTA File Name                | Image Type Hex/Dec | Product String in FW                | HW gen  / Zigbee ver | Version Prod       | Version Test |
|------------------------------|--------------|-------------------------------------------|---------------|------------------------|--|
| zingo_kt_bulb_hwpwmcs_ws | 0x2204 8708 | Not Used  | 3 ZB3 | 1.1.003 ✅ | 🔲 |
| tradfri_sy5882_driver_ws     | 0x4203 16899 | FLOALT panel WS 30x30 (PG2ZBID12C0400P)  |  1 ZLL > ZB3 | 2.3.091 ✅ | 🔲 |
| tradfri_sy5882_bulb_ws       | 0x4204 16900 | TRÅDFRI bulb GU10 WS 400lm  | 1 ZLL > ZB3 | 2.3.091 ✅ | 🔲 |

so your light shall having one OTA update waiting for being done !!!
I have looking on my GU10 WS and they is already updated then i was coming home from work so likely its working for your light 2.

Jeepmg, 2023-02-04:
Good morning,

still no way to update the firmware of this bulb with Z2M?

THANKS