Koenkk/zigbee2mqtt #11324: Frontend won't start without MQTT

danielvandenberg95, 2022-02-09:
### What happened?

> @parshivets I believe the issue is that startup is blocked because it cannot to the MQTT server, please check your MQTT settings, you should see `Connected to MQTT server` in your logs

I'm currently in a bit of a situation where this is actually a slight issue.
Due to some remodeling work, power to my MQTT server is turned off.
I want to replace the wall plug to my server with a different one.
I can reach my Pi running zigbee2mqtt. However, I can not pair my new plug, because the interface won't start, because the MQTT server is down.

So, I vote for starting the frontend before connecting to MQTT.

_Originally posted by @danielvandenberg95 in https://github.com/Koenkk/zigbee2mqtt/issues/4266#issuecomment-1033668221_

### What did you expect to happen?

Frontend to start up

### How to reproduce it (minimal and precise)

Bring down MQTT
Start zigbee2mqtt

### Zigbee2MQTT version

Ha ha very funny. "Can be found in the frontend -> settings -> about -> Zigbee2MQTT version"

### Adapter firmware version

3.2.6.0

### Adapter

CCsomething, but not related.

### Debug log

Zigbee2MQTT:info  2022-02-09 12:34:50: Starting Zigbee2MQTT version 1.23.0-dev (commit #7fe803a)

Zigbee2MQTT:info  2022-02-09 12:34:50: Starting zigbee-herdsman (0.14.13)

Zigbee2MQTT:info  2022-02-09 12:34:51: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'

Koenkk, 2022-02-09:
Due to the internal architecture, the frontend/z2m isn't able to function without being connected to an MQTT server. I therefore consider this a feature request to allow Z2M to start without being connected to a MQTT server.

danielvandenberg95, 2022-03-12:
Remove stale

closed 2022-05-20 by github-actions[bot]