Koenkk/zigbee2mqtt #11330: [New device support]: new revision of Livolo TI0001

setfire2, 2022-02-09:
### Link

https://www.livolo.com/cz/catalogDetail/VL-FC1Z-2G

### Database entry

{"dont":"know"}

### Comments

Today I received new Zigbee light switches from Livolo. I already have some older ones. The new ones have a cubic case while my old ones have a round case. For comparison:
Old: https://www.livolo-deutschland.de/p/zigbee-smart-lichtschalter-1-fach-vl-c701z-11-weiss
New: https://www.livolo.com/cz/catalogDetail/VL-FC1Z-2G

During pairing (I use iobroker.zigbee), the new ones are detected as TI0001 like the old ones. But they don't react on any Zigbee state changes.

I contacted Livolo and they confirmed that the chip has been changed in the new revision as well as the case.

I attached my logfile of one of the pairing tries as html to this request. I could supply a switch for testing purposes or could also investigate further when given some instructions,

[iobroker.2022-02-09.log](https://github.com/Koenkk/zigbee2mqtt/files/8035158/iobroker.2022-02-09.log)
d

systemiii, 2022-02-11:
If it helps you, you can find my external converter, which supports the new Livolo devices, here #8075

closed 2022-03-21 by github-actions[bot]

setfire2, 2023-02-26:
Update on the new revision of the Livolo switch: I tried one again yesterday and they work now.

kenvals, 2023-03-28:
I became the owner of a new revision and I will say that I have problems.
Zigbee2mqtt worked fine, and then the livolo switch started blinking red/blue and stopped being controlled.

I tried to sync it again with my zigbee network, now it always shows as an unidentified device... from neighboring topics I tried to install an external javascript converter, and also updated to the latest version of zigbee2mqtt docker:latest - did not help

botany112, 2024-03-09:
yeah, i also got a new one with dimmer. its connect and everything, then start blinking red and blue then device goes offline

botany112, 2024-09-09:
> Update on the new revision of the Livolo switch: I tried one again yesterday and they work now.

how did you get it to work?

setfire2, 2024-09-09:
Well, it just worked at some point. But it was one without a Dimmer.