Koenkk/zigbee2mqtt #11332: Aqara RTCGQ11LM drops when sleeping

MRobi1, 2022-02-09:
### What happened?

I setup 20 of these motion sensors throughout the house. As long as I keep moving regularly in front of the sensor, it will stay connected and report motion all day long. As soon as I let them sit with no motion in front of them, they will no longer report until they are re-joined onto the network. When I look at the reported lux and temperature history, there is an updated value sent 30m after the last reported motion event, and then nothing. The device is completely dead. In the FAQ it states it is normal for xiaomi devices not to show a link in the map because they sleep for a too long time. It appears as soon as they start to sleep, they are not waking up again.

### What did you expect to happen?

I expect the device to continue reporting motion without having to run through the house and trigger them every 30 minutes.

### How to reproduce it (minimal and precise)

- Join devices to network
- Allow them to sit

- Have joined all of them in their final placement.
- Have used Permit join (all) as well as joining through specific devices. I have a Sinope TH1124ZB in every room of the house which is listed in the compatible with Aqara devices list. I also have an ikea tradfri signal repeater and ikea tradfri wireless control outlet in every room, both on the known compatible list.
- I have no devices on the network at all that are on the known incompatible list.
- Currently have 41 router devices on the network.

### Zigbee2MQTT version

1.23.0-dev commit: ddcb8e1

### Adapter firmware version

20211217

### Adapter

Tube CC2652

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8035305/log.txt)

Koenkk, 2022-02-10:
Probably not something we can solve from z2m, https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html#troubleshooting-device-stops-sending-messages-disconnects-from-network might help

MRobi1, 2022-02-20:
I've had these stable by creating a second zigbee network with ONLY ikea repeaters. I suspect the issue was with the Sinope TH1124ZB thermostats even though they're on the list of compatible with these devices.

Although I'm having a second issue popping up. I've done the hardware mod on all of them and set the occupancy_timeout to 5s. After joining they all work at a 5s cooldown, but shortly after they all go back to 60s cooldowns. They continue detecting no problem at 60s cooldowns. To get back to 5s cooldown I have to rejoin again.

closed 2022-03-30 by github-actions[bot]