Koenkk/zigbee2mqtt #11333: Legrand 412015 not pairing with CC2652P but works fine with CC2531

PaoloTK, 2022-02-09:
### What happened?

I've been using my Legrand 412015 Power Meter with no issues for the last year. I recently decided to upgrade from a CC2531 stick to the new Sonoff Zigbee 3.0 USB Dongle Plus, which is CC2652P based, but unfortunately the device fails to pair. If I install back the CC2531 it pairs just fine on the first try.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

VOVAiS, 2022-02-11:
The device supports only channel 11

closed 2022-03-22 by github-actions[bot]

HaraldGrhein, 2022-03-23:
Have a new BTicino F20T60 (which is Legrand 412015) received today. 

Zigbee pairing with Sonoff Zigbee 3.0 USB Dongle Plus inside iobroker: I will the F20T60 show up an disappear during pairing and it show pairing.

"New device joined '0x00047400009263eb' model 412015
Device '0x00047400009263eb' announced itself
Interview started: 0x00047400009263eb
....
Pairing time left: 58
Interview successful: {"friendly_name":"0x00047400009263eb","model":"412015","vendor":"Legrand","description":"DIN power consumption module","supported":true}
New device joined '0x00047400009263eb' model 412015
Device '0x00047400009263eb' announced itself
Interview started: 0x00047400009263eb
Pairing time left: 59
Pairing time left: 60
Pairing started"

That show that the F20T60 does communicate with Sonoff Zigbee 3.0 USB Dongle Plus.

But finally failed joining with error:

Starting interview of '0x00047400009263eb'
Device '0x00047400009263eb' announced itself
Successfully interviewed '0x00047400009263eb', device has succesfully been paired
Device '0x00047400009263eb' is supported, identified as: Legrand DIN power consumption module (412015)
Configuring 0x00047400009263eb DIN power consumption module
DeviceConfigure:DeviceConfigure failed 0x00047400009263eb DIN power consumption module, attempt 1 (Error: Read 0x00047400009263eb/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/opt/iobroker/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.readEletricalMeasurementMultiplierDivisors (/opt/iobroker/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5) at Object.configure (/opt/iobroker/node_modules/zigbee-herdsman-converters/devices/legrand.js:264:13) at DeviceConfigure.doConfigure (/opt/iobroker/node_modules/iobroker.zigbee/lib/zbDeviceConfigure.js:145:9) at DeviceConfigure.configure (/opt/iobroker/node_modules/iobroker.zigbee/lib/zbDeviceConfigure.js:124:17))

HaraldGrhein, 2022-03-23:
Oh, after writing I finally could pair it using the description from "https://zigbee.blakadder.com/Legrand_412015.html" . And yes, during pairing I did make successive clicks on the reset button (every second), until the LED turns green.

fred-c1, 2022-08-27:
This helped me:
- This device ONLY works on channel 11  => shame on Legrand / BTicino
- If the paring process does not start (no pop up messages on zigbee2mqtt): Restart the coordinator (for example, unplug + plug the USB dongle / coordinator)
- If the device leave the zigbee network just after pairing/joining: Press the "reset" button on the device during the pairing process at a rate of about 1 per second