Koenkk/zigbee2mqtt #11338: Livolo 1M Switch

Ric199021, 2022-02-10:
### What happened?

I'm able to control the switch but can't receive the status back from the switch.

### What did you expect to happen?

To control the switch and also to have the status if it's on off

### How to reproduce it (minimal and precise)

if you switch the channel to 26 it will recognize this one as the big switch and it's not working you need to add a new config.js file and the changes what I have into the description (livolo.js) but I can't get the status

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

systemiii, 2022-02-11:
After my experience for the same situation, I have learned that the Livolo devices cannot work with the ConBee II adapter because they use malformed Zigbee packages discarded by the firmware inside the ConBee II.
I recommend using the new Sonoff USB Zigbee adapter which, in my opinion, has the same performance as the ConBee II and works great with Livolo.

However, I warn you that the new 1M and similar devices are not yet fully supported by Z2M, in fact they do not remain equalized for more than 1 minute.
I created an external converter that solves this problem, you can find it here inside #8075

Ric199021, 2022-02-11:
I understood that you must have the permit join on true from the zigbee2mqqt config file. Otherwise it will not work. I have only conbee II adapter, and I really needed that 1m device because I have a lot of Gewiss switches. Do you know something like this 1m livolo switch?

systemiii, 2022-02-12:
I have searched all over the web there Is nothing like Livolo devices unfortunately

Ric199021, 2022-02-12:
ok, thanks. Then i think I can use it without knowing the state if it's on or off. But it's strange that the big device it's working [image] and should be the same configuration. But for the 1m it's not working. Tbh i didn't test the big one, but I read on some forums that they make it work but i can't remember if they used conbee II or about another adapter.

systemiii, 2022-02-12:
New devices can't works with ConBee because Livolo has changed packet structure and ConBee discard most of commands, like final stage in pairing process and switch status.
Infact your switch should not be able to keep paired along time or after power cycle.

closed 2022-03-22 by github-actions[bot]