Koenkk/zigbee2mqtt #11339: [New device support]: EcoDim Eco-Dim.10

lvanjaarsveld, 2022-02-10:
### Link

https://www.ecodim.nl/eco-dim10-zigbee.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x0c4314fffe237f1b","nwkAddr":12910,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"EcoDim-Zigbee 3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"EcoDim-Zigbee 3.0","manufacturerName":"EcoDim BV","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20210228","swBuildId":"3.09"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237b9d68","endpointID":242},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00237b9d68","endpointID":242}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20210228","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1644489583411,"defaultSendRequestWhen":"immediate"}

### Comments

The device itself is actually discovered and recognized as ECO-DIM.07 which is not correct, it's a ECO-DIM.10. In my Zigbee network, I'm using 9 ECO-DIM.07 modules flawlessly. Based on their performance and bad experience with iCasa modules I decided to buy and add support for the ECO-DIM.10 module. 

I setup a isolated zigbee network running from my laptop on the zigbee2mqtt-dev branch for debugging purposes. Any suggestions to try are welcome.

### External converter

```shell
Non jet.
```

lvanjaarsveld, 2022-02-10:
Here is the log of the interview and after that I tried interacting with the device.

`debug 2022-02-10 11:38:48: Can't load state from file C:\zigbee2mqtt-dev\data\state.json (doesn't exist)
info  2022-02-10 11:38:48: Logging to console and directory: 'C:\zigbee2mqtt-dev\data\log\2022-02-10.11-38-46' filename: log.txt
info  2022-02-10 11:38:48: Starting Zigbee2MQTT version 1.23.0-dev (commit #unknown)
info  2022-02-10 11:38:48: Starting zigbee-herdsman (0.14.13)
debug 2022-02-10 11:38:48: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"C:\\zigbee2mqtt-dev\\data\\coordinator_backup.json","databaseBackupPath":"C:\\zigbee2mqtt-dev\\data\\database.db.backup","databasePath":"C:\\zigbee2mqtt-dev\\data\\database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"\\\\.\\COM3"}}'
info  2022-02-10 11:39:16: zigbee-herdsman started (reset)
info  2022-02-10 11:39:16: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'
debug 2022-02-10 11:39:16: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b00237b9d68","panID":6754}
info  2022-02-10 11:39:16: Currently 0 devices are joined:
info  2022-02-10 11:39:16: Zigbee: disabling joining new devices.
info  2022-02-10 11:39:16: Connecting to MQTT server at mqtt://localhost
debug 2022-02-10 11:39:16: Using MQTT anonymous login
info  2022-02-10 11:39:16: Connected to MQTT server
info  2022-02-10 11:39:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2022-02-10 11:39:16: Started frontend on port 0.0.0.0:8080
debug 2022-02-10 11:39:30: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"rrid4-1","value":true}'
info  2022-02-10 11:39:30: Zigbee: allowing new devices to join.
info  2022-02-10 11:39:30: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"rrid4-1"}'
info  2022-02-10 11:39:40: Device '0x0c4314fffe237f1b' joined
info  2022-02-10 11:39:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe237f1b","ieee_address":"0x0c4314fffe237f1b"},"type":"device_joined"}'
info  2022-02-10 11:39:41: Starting interview of '0x0c4314fffe237f1b'
info  2022-02-10 11:39:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe237f1b","ieee_address":"0x0c4314fffe237f1b","status":"started"},"type":"device_interview"}'
debug 2022-02-10 11:39:41: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"modelId":"EcoDim-Zigbee 3.0"}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:41: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:41: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"EcoDim BV"}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:41: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:41: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:41: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:41: Device '0x0c4314fffe237f1b' announced itself
info  2022-02-10 11:39:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe237f1b","ieee_address":"0x0c4314fffe237f1b"},"type":"device_announce"}'
debug 2022-02-10 11:39:41: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:42: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:42: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"appVersion":3}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:42: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:42: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":6}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:42: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:42: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:42: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:42: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20210228"}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:42: Skipping message, definition is undefined and still interviewing
debug 2022-02-10 11:39:42: Received Zigbee message from '0x0c4314fffe237f1b', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"3.09"}' from endpoint 1 with groupID 0
debug 2022-02-10 11:39:42: Skipping message, definition is undefined and still interviewing
[... cut ...] riteUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (C:\zigbee2mqtt-dev\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:414:23)
    at Queue.executeNext (C:\zigbee2mqtt-dev\node_modules\zigbee-herdsman\src\utils\queue.ts:32:32)
debug 2022-02-10 11:48:48: Saving state to file C:\zigbee2mqtt-dev\data\state.json
debug 2022-02-10 11:53:48: Saving state to file C:\zigbee2mqtt-dev\data\state.json
`

lvanjaarsveld, 2022-02-10:
Ok, strange update. Was trying to utilize the dev console, initially failing quite a few commands. Then I was suddenly able to read onoff and levelcontrol. From that point I was also able to fully control the device trough the exposed commands, on/off, brightness.

So in this state the device ECO-DIM.10 is fully functional, just wrongly reported as ECO-DIM.07.

lvanjaarsveld, 2022-02-10:
Ok, support looks solid beyond wrong identification. There is a required step to properly add the ecodim.10. I moved the initial device successfully to my main network and after that successful added 3 more devices. 

First the eco-dim.10 needs a reset after power up, hold the reset button for 5-6 seconds. Indication light goes red and turns of when finished. Behavior is identical to eco-dim.07.
Second, press reset two times shortly. The indicator led wil blink blue. Again similar behavior as evo-dim.07. the device wil now join your network.

Koenkk, 2022-02-10:
Device uses the same zigbee model so we cannot distinguish them, changed the model to `Eco-Dim.07/Eco-Dim.10`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-10 by Koenkk