Koenkk/zigbee2mqtt #11342: Lutron Aurora Brightness not Working on 1.23.0 (main)

nebhead, 2022-02-10:
### What happened?

Since switching to the v1.23.0 (Main) branch (and tried the development branch as well), the Lutron Aurora dimmer no longer seems to be able to dim the cluster or bulbs that it is associated with.  I've tried removing the device and re-adding, refreshing, adding it to the groups that it is intended to dim, but nothing seems to bring back the old behavior.  

### What did you expect to happen?

I expect that when the Lutron Aurora is bound to specific cluster or bulb that the rotary dimmer will adjust the brightness on that cluster or bulb.  

### How to reproduce it (minimal and precise)

- Pair a Lutron Aurora to Zigbee2MQTT
- Add a cluster (group) to the binds with levelCtrl, onOff selected. 
- Add Lutron Aurora to same group

Lights turn on and off with button press, but do not dim with the rotary dial.  

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20220103

### Adapter

Sonoff Zigbee 3.0 Dongle Plus

### Debug log

``` 
MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":32,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
[... 1 lines cut ...]
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"brightness_move_to_level","action_level":255,"action_transition_time":0.02,"brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
[... cut ...] "update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B', payload '{"action":"","brightness":null,"linkquality":36,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Playroom Dimmer B/action', payload 'brightness_move_to_level'
Info MQTT publish: topic 'zigbee2mqtt/Smart Plug Dining Room', payload '{"linkquality":21,"state":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/3D Printer', payload '{"linkquality":36,"state":"OFF"}'
Info 
```

twinkie0101, 2022-02-22:
Zigbee2MQTT "locks" up on me when I try to bind to a light group with the levelCtrl and onOff selected - using v1.23.0-1 (Main).

bravesfan171, 2022-02-25:
@nebhead @Koenkk I am noticing the same issue on 1.23.0 with both of my Auroras. I am also on the Sonoff Zigbee 3.0 Plus for my coordinator and I am flooded in the log from each press. This log here is two presses of the button:
```
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":255,"transtime":7}' from endpoint 1 with groupID 57507
Info MQTT publish: topic 'zigbee2mqtt/Living Room Dimmer Switch', payload '{"action":"brightness_move_to_level","action_group":57507,"action_level":255,"action_transition_time":0.07,"brightness":0,"linkquality":138,"update":{"state":"idle"},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Living Room Dimmer Switch', payload '{"action":"","brightness":0,"linkquality":138,"update":{"state":"idle"},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/Living Room Dimmer Switch/action', payload 'brightness_move_to_level'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genLevelCtrl', data '["currentLevel"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genLevelCtrl' and type 'read' and data '["currentLevel"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genLevelCtrl', data '["currentLevel"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genLevelCtrl' and type 'read' and data '["currentLevel"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genLevelCtrl', data '["currentLevel"]' from endpoint 1 with groupID 0
Debug No converter available for 'Z3-1BRL' with cluster 'genLevelCtrl' and type 'read' and data '["currentLevel"]'
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'read', cluster 'genLevelCtrl', data '["currentLevel"]' from endpoint 1 with groupID 0
[... cut ...]
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":255,"transtime":2}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Living Room Dimmer Switch', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":255,"transtime":2}' from endpoint 1 with groupID 0
```

I have legacy turned off for each switch too. I can't seem to get the blueprint to work at all that is designed for the switch.

nebhead, 2022-02-28:
The strange thing in my case is that I have two Lutron Aurora's in the house and only one of them seems to have lost the dimming capability.  The other still works as expected, though it seems like there is an issue with it possibly losing connection with one of the bulbs it is dimming.  I've looked at the configuration file and there really isn't any difference between the two and how they are configured there.  Very odd.

twinkie0101, 2022-03-01:
I updated to 1.24.0 a few minutes ago and can confirm LQI is reporting as well as state and brightness are updating accurately and very quickly (less than 1 second).  I haven't tried binding to a group yet, will try that later.

deepcoder, 2022-03-01:
What model and firmware do your devices have? Thx.
```
Manufacture : Lutron	
Model : Z3-1BRL	
Firmware Build Date: 20181019	
Firmware Version : 3.8
```

twinkie0101, 2022-03-01:
@deepcoder 

`Manufacture : Lutron	
Model : Z3-1BRL	
Firmware Build Date: 20181019	
Firmware Version : 3.8`

nebhead, 2022-03-01:
Firmware build date: 20181019
Firmware version: 3.8
Model
: Z3-1BRL

twinkie0101, 2022-03-01:
I just use the bind to the lights I want to control in Zigbee2mqtt and they’re working great.  The issue I was having was the lightbulb state was getting updated timely so if a light was tied to a motion sensor it would trigger at like 2lux.  No longer an issue with the update.  But when I tried to bind to a group Zigbee2mqtt would lock up.  Hoping to try this later tonight to see if it works.

nebhead, 2022-03-02:
> I just use the bind to the lights I want to control in Zigbee2mqtt and they’re working great. 

This is good news.  Did you remove the devices and re-pair them in your system?  Or did you just go through the binding process again?

twinkie0101, 2022-03-02:
I had my dimmers already setup in zigbee2mqtt (binding) prior to the update.

nebhead, 2022-03-02:
> I had my dimmers already setup in zigbee2mqtt (binding) prior to the update.

Thanks for the response.  OK, so it sounds like it worked for you through just updating the package. 

Good news on my front as well.  While initially it did _not_ work after updating to v1.24.0, I went back to my configuration and tried removing the dimmer and re-adding it.  This still did not yield results.  

Finally, I tried **binding the dimmer to the bulbs directly, instead of the group of bulbs.**  And presto, this works!  **So it seems that you cannot bind to groups, for this specific dimmer at this time.**  May require you to bind directly to bulbs only.  

With that, I'm going to close this issue. I hope this thread helps others.  

Thanks!

closed 2022-03-02 by nebhead

PaulMcMillan, 2022-04-11:
Based on my experimenting with this device this afternoon, I'm pretty convinced there's some kind of race condition during binding (maybe bugs in the device firmware?) that sometimes results in the wrong attributes getting bound to the wrong endpoint.

Specifically, I managed to (somehow) get one of my devices bound such that on/off was working with direct group control, but brightness control worked for a single device. This only happened AFTER I gave up on group control, and started binding level control to that individual device.

As with the earlier poster, I was getting the "commandMoveToLevelWithOnOff" spam in my Z2M debug logs if I enabled levels to the controller. So I made sure not to enable those on my second device ever. HOWEVER, after quickly binding 4 separate hue lightbulbs to the same Aurora, it magically started spewing those errors towards the controller. I'm absolutely positive this switch was never instructed to set that binding.

I ended up leaving it alone, since the spam is relatively harmless, these switches aren't used too much, and I plan to replace them soon. I'll be curious to see whether (now that they're bound) they drop off the network in six hours like they were doing before.

nebhead, 2022-04-11:
> I ended up leaving it alone, since the spam is relatively harmless, these switches aren't used too much, and I plan to replace them soon. I'll be curious to see whether (now that they're bound) they drop off the network in six hours like they were doing before.

Keep us updated.  I've had some similar issues to what you mentioned above.  I do really like these switches and felt that before the 1.23.0 update, they were working pretty flawlessly.

PaulMcMillan, 2022-04-15:
Update - they drop off the network still. They didn't do that before, so my best guess is that the most recent patch to their behavior was wrong.

nebhead, 2022-04-15:
> Update - they drop off the network still. They didn't do that before, so my best guess is that the most recent patch to their behavior was wrong.

Which version are you currently running?  I'm on [1.25.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.25.0) commit: [6f1460e](https://github.com/Koenkk/zigbee2mqtt/commit/6f1460e) in the Docker container.  Right now, I haven't seen the device drop off the network, unless I wait several days without using the device.  And even then, it seems when I use it (even though it shows offline in the WebUI), it shows back up.  I did have previous issues with binding being a bit wonky, but I changed my mesh so that I removed a repeating bulb which would frequently be turned off by a wall switch - and so the lutron wouldn't mesh to that device.  That seems to have improved the situation greatly.  

Side note:  I still recently have been having issues with the docker container crashing and restarting.  Not sure what that is all about though.  Need to gather the logs and investigate.

deepcoder, 2022-04-15:
I have my Lutron Aurora stable and working for 4 weeks nonstop no issues, LQI values right around 110, I am using it's events via MQTT in Home Assistant, so no zigbee direct control via it :
```
zigbee2mqtt 1.24.0 commit: 7a2ddf2 in docker container
TubesZB cc26xx coordinator direct USB connected with 20211217 firmware
```
Not with the Aurora, however on a test network, I have had a lot of issues with zigbee2mqtt 1.25.0 and 20220219 firmware, routing and container crashing. Once I rolled this test environment back to the 1.24.0 /20211217 it is as solid as the parallel production zigbee2mqtt config referenced above.

fractalcounty, 2024-01-08:
Sorry to comment on an issue that was closed 2 years ago, but is it possible to bind the Lutron Aurora dimmer switch (Z3-1BRL) to groups yet? There's also some additional annoying quirks that I have yet to figure out and I'm curious if it's user error or not. This is what I've noticed:

- The Lutron Aurora dimmer switch cannot be bound to groups, only individual lights via the LevelCtr cluster.
- Attempting to bind the dimmer to groups (or any other cluster that isn't LevelCtr) causes the Lutron Aurora to lock up.
- The only way to restore functionality after this point is to factory reset the Lutron Aurora by quickly tapping the button three times, holding for around 20 seconds on the third tap. and rapidly pressing it a bunch more when the light on the Aurora begins rapidly blinking until it enters pairing mode.
- Besides the factory reset process, none of this is really mentioned in the zigbee2mqtt device page for the [Lutron Z3-1BRL](https://www.zigbee2mqtt.io/devices/Z3-1BRL.html), leading me to believe that this isn't a common issue or is perhaps user error.

Any ideas?

nebhead, 2024-01-09:
For what it is worth, when I was using these dimmer knobs, I was able to bind them to multiple Zigbee bulbs.  After clicking bind, you just need to click the Aurora button and it should complete the binding process successfully.  No issues with them dropping off of the network or needing to reset them to factory settings.  

I had lots of other issues with this product though, which is a shame given their price.  Things like simply not controlling brightness, or controlling brightness on only two of the bulbs it was bound to.  Or simply not working at all.  I've since moved on to Leviton switches [DG6HD-1BW](https://www.zigbee2mqtt.io/devices/DG6HD-1BW.html#leviton-dg6hd-1bw) which control dumb bulbs.  They _also_ have their quirks, but I have been able to work around their weirdness.  

Anyway, best of luck, and report back if you have success.