Koenkk/zigbee2mqtt #11353: Xiaomi Aqara SSM-U01 not show specific settings

dtalens, 2022-02-11:
### What happened?

The documentation for Xiaomi Aqara SSM-U01 device shows that  temperature can be calibrated, but then nothing is shown in the Settings (specific) section.
[image]
I added this directly to the `device.yaml` file but it has no effect.: 
```
'0x54ef44xxxxxxxxxx':
  friendly_name: Estufa tovalloles
  temperature_calibration: -4
```

### Zigbee2MQTT version

1.23.0-dev commit: 7fe803a

### Adapter firmware version

20220103

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

McGiverGim, 2022-02-11:
Is strange, I see it, but I see too a illuminance calibration, that has no sense...

[image]

[image]

McGiverGim, 2022-02-11:
I don't know why I see the temperature and illuminance (maybe a bug?) but both of them appear in the documentation of the device too:

[image]

What I see is that some devices, that use the `temperature` converter, define the "options" with precision/calibration, but the SSM-U01 uses the `device_temperature` converter, but there is no option for the precision/calibration. Maybe the problem is here? the `device_temperature` can be calibrated?

https://github.com/Koenkk/zigbee-herdsman-converters/blob/06c3c11082af424afe86dfc670cce7bd83f63b72/converters/fromZigbee.js#L342-L360

McGiverGim, 2022-02-11:
@dtalens looking at the code, it differentiates between `temperature` and `device_temperature`, and only the first can be calibrated. I don't know if it has sense to calibrate the `device_temperature`, but if it is, maybe is better to ask for a `feature request` for it.

Another thing is why I have the possibility to calibrate temperature and illuminance. I think at some point it failed at this condition:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/06c3c11082af424afe86dfc670cce7bd83f63b72/converters/fromZigbee.js#L5229-L5241

Why? I don't know...

closed 2022-03-21 by github-actions[bot]