Koenkk/zigbee2mqtt #11360: Hive TRV keeps resetting to default on z2m

fryc88, 2022-02-11:
### What happened?

Hi,
Last night i was migrating from deconz to z2m, in general it went ok, except of 3 devices, 2 are some chinese bulb, so that's not so important, but bigger issue is with Hive TRV.
I have 4 Hive TRVs installed (danfoss ally but branded by Hive as far as i know). 3 are working fine, no issues at all. 4th one keeps resetting to default after pairing in z2m.
Debug log in z2m does not say a lot, tells me about new device and that's it. When i disable pairing new devices just after pairing TRV it does not change anything, still resets itself.
```
MQTT publish: topic 'zigbee2mqtt/0xccccccfffe1d9e78', payload '{"algorithm_scale_factor":null,"battery":null,"day_of_week":null,"external_measured_room_sensor":null,"heat_available":null,"heat_required":null,"keypad_lockout":null,"last_seen":"2022-02-11T12:41:34+00:00","linkquality":255,"load_balancing_enable":null,"load_estimate":null,"load_room_mean":null,"mounted_mode_active":null,"mounted_mode_control":null,"pi_heating_demand":null,"programming_operation_mode":null,"radiator_covered":null,"running_state":null,"setpoint_change_source":null,"thermostat_vertical_orientation":null,"trigger_time":null,"update":{"state":null},"update_available":null,"viewing_direction":null,"window_open_external":null,"window_open_internal":null}'
info MQTT publish: topic 'zigbee2mqtt/0xccccccfffe1d9e78', payload '{"algorithm_scale_factor":null,"battery":null,"day_of_week":null,"external_measured_room_sensor":null,"heat_available":null,"heat_required":null,"keypad_lockout":null,"last_seen":"2022-02-11T12:42:20+00:00","linkquality":255,"load_balancing_enable":null,"load_estimate":null,"load_room_mean":null,"mounted_mode_active":null,"mounted_mode_control":null,"pi_heating_demand":null,"programming_operation_mode":null,"radiator_covered":null,"running_state":null,"setpoint_change_source":null,"thermostat_vertical_orientation":null,"trigger_time":null,"update":{"state":null},"update_available":null,"viewing_direction":null,"window_open_external":null,"window_open_internal":null}'
debug Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"2qo2u-7","value":false}'
info Zigbee: disabling joining new devices.
info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"2qo2u-7"}'
```
I thought maybe it's some TRV problem, so i paired them all to Hive hub, updated to latest 227 firmware (that's what Hive is showing), paired them back to z2m and again same issue, but got it even worse at end. Haven't had that issue in Hive or Deconz previously.
So NOW, after updating it to 227 and trying to pair them all back to z2m ALL of them are resetting to default, just few seconds after pairing. 
To reset it to default i had to press and hold button and reinstall battery, then it goes to installer mode to be able to pair. After successful pairing usually i have to press and hold button without taking out batteries, to switch from installer mode to normal mode, could do it last night with other 3 TRVs, but now can't do it with any of them, because as i said, it keeps resetting to default when paired with z2m.
Another strange thing is, z2m (even if hive tells me all got latest 227 firmware) shows me different firmware versions and build dates.
[image]

Now i've no idea what to do with it, got them non usable in my HA at this moment :/

### What did you expect to happen?

pair&work

### How to reproduce it (minimal and precise)

update in Hive gateway, and simply pair to z2m

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26720700

### Adapter

conbee 2

### Debug log

info  11-02-2022 12:40:58: Succesfully changed options
info  11-02-2022 12:40:58: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"2qo2u-5"}'
debug 11-02-2022 12:41:01: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"2qo2u-6","value":true}'
info  11-02-2022 12:41:01: Zigbee: allowing new devices to join.
info  11-02-2022 12:41:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"2qo2u-6"}'
debug 11-02-2022 12:41:04: Saving state to file /app/data/state.json
info  11-02-2022 12:41:34: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe1d9e78', payload '{"algorithm_scale_factor":null,"battery":null,"day_of_week":null,"external_measured_room_sensor":null,"heat_available":null,"heat_required":null,"keypad_lockout":null,"last_seen":"2022-02-11T12:41:34+00:00","linkquality":255,"load_balancing_enable":null,"load_estimate":null,"load_room_mean":null,"mounted_mode_active":null,"mounted_mode_control":null,"pi_heating_demand":null,"programming_operation_mode":null,"radiator_covered":null,"running_state":null,"setpoint_change_source":null,"thermostat_vertical_orientation":null,"trigger_time":null,"update":{"state":null},"update_available":null,"viewing_direction":null,"window_open_external":null,"window_open_internal":null}'
info  11-02-2022 12:42:20: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe1d9e78', payload '{"algorithm_scale_factor":null,"battery":null,"day_of_week":null,"external_measured_room_sensor":null,"heat_available":null,"heat_required":null,"keypad_lockout":null,"last_seen":"2022-02-11T12:42:20+00:00","linkquality":255,"load_balancing_enable":null,"load_estimate":null,"load_room_mean":null,"mounted_mode_active":null,"mounted_mode_control":null,"pi_heating_demand":null,"programming_operation_mode":null,"radiator_covered":null,"running_state":null,"setpoint_change_source":null,"thermostat_vertical_orientation":null,"trigger_time":null,"update":{"state":null},"update_available":null,"viewing_direction":null,"window_open_external":null,"window_open_internal":null}'
debug 11-02-2022 12:42:22: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"2qo2u-7","value":false}'
info  11-02-2022 12:42:22: Zigbee: disabling joining new devices.
info  11-02-2022 12:42:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"2qo2u-7"}'
debug 11-02-2022 12:46:04: Saving state to file /app/data/state.json
debug 11-02-2022 12:46:27: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"2qo2u-8","value":true}'
info  11-02-2022 12:46:27: Zigbee: allowing new devices to join.
info  11-02-2022 12:46:27: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"2qo2u-8"}'
info  11-02-2022 12:46:30: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe1d9e78', payload '{"algorithm_scale_factor":null,"battery":null,"day_of_week":null,"external_measured_room_sensor":null,"heat_available":null,"heat_required":null,"keypad_lockout":null,"last_seen":"2022-02-11T12:46:30+00:00","linkquality":255,"load_balancing_enable":null,"load_estimate":null,"load_room_mean":null,"mounted_mode_active":null,"mounted_mode_control":null,"pi_heating_demand":null,"programming_operation_mode":null,"radiator_covered":null,"running_state":null,"setpoint_change_source":null,"thermostat_vertical_orientation":null,"trigger_time":null,"update":{"state":null},"update_available":null,"viewing_direction":null,"window_open_external":null,"window_open_internal":null}'
info  11-02-2022 12:47:11: Device '0x14b457fffebed36d' joined
info  11-02-2022 12:47:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x14b457fffebed36d","ieee_address":"0x14b457fffebed36d"},"type":"device_joined"}'
info  11-02-2022 12:47:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x14b457fffebed36d"},"type":"device_connected"}'
info  11-02-2022 12:47:11: MQTT publish: topic 'homeassistant/sensor/0x14b457fffebed36d/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x14b457fffebed36d"],"manufacturer":"Hive","model":"Radiator valve based on Danfoss Ally (014G2461) (UK7004240)","name":"0x14b457fffebed36d","sw_version":"01.15.0006 01.15"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic" [... cut ...] algorithm scale factor","state_topic":"zigbee2mqtt/0xccccccfffe1d9e78","unique_id":"0xccccccfffe1d9e78_algorithm_scale_factor_zigbee2mqtt","value_template":"{{ value_json.algorithm_scale_factor }}"}'
debug 11-02-2022 12:50:08: Received MQTT message on 'homeassistant/number/0xccccccfffe1d9e78/algorithm_scale_factor/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0xccccccfffe1d9e78/set/algorithm_scale_factor","device":{"identifiers":["zigbee2m

closed 2022-03-21 by github-actions[bot]

robertalexa, 2022-04-14:
Are you still experiencing this issue? I have 8 of these TRVs and they all pair as expected. The frmware version at my end:
[image]

And the 8th of them which i have literally just removed from my main network and paired with my testing network, everything went as expected.
[image]

MagicalTrev89, 2022-05-10:
@robertalexa I have just updated my coordinator to latest version and all my hive TRVs have stopped behaving, it looks like there is new functionality. I notice the firmware version mine are running is older than yours but i am running latest 227 hive update.

When did you buy your hive TRVs ?

robertalexa, 2022-05-10:
@MagicalTrev89 Official Hive website, but i got them originally with the Hive hub before i migrated to z2m about a year later. So the firmware i have is the latest that was available at that time (almost 2 years ago i think), but it might still be the latest available now. Unfortunately Hive does not publicise their firmware, so we can't do OTA via z2m. In that regard, the original Danfoss Ally and the Popp TRV are better solutions as they are the same physical devices but with the official Danfoss Ally firmware.

I have also attempted updating the firmware to the official Danfoss one (with a bit of hacking) but the devices deny the image, so that is also a dead end.