Koenkk/zigbee2mqtt #11362: [New device support]: Profalux covers support

rclsilver, 2022-02-11:
### Link

https://www.profalux-pro.com/documentations/1.%20Notices,%20Plans,%20Technique/1.%20Volets%20roulants/3.%20Moteurs%20Commandes%20et%20Accessoires/1.%20Moteur/Moteur%20Profalux%20Zigbee/

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0x20918a00000dacdb","nwkAddr":7417,"manufId":4368,"powerSource":"Unknown","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":513,"inClusterList":[0,3,21],"outClusterList":[3,4,5,6,8,256,64544,64545],"clusters":{"genBasic":{"attributes":{"powerSource":0,"zclVersion":1,"dateCode":"20170309 NA 064"}}},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"20170309 NA 064","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1644576713553,"defaultSendRequestWhen":"immediate"}

### Comments

Profalux covers come with one remote per cover and a main remote (ZOE).

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

let tzCover = {
    key: ['position', 'state'],
    options: [exposes.options.invert_cover()],
    convertSet: async (entity, key, value, meta) => {
        if (typeof value !== 'number') {
            switch(value.toLowerCase()) {
                case 'stop':
                    await entity.command('genLevelCtrl', 'stop', {}, utils.getOptions(meta.mapped, entity));
                    return;

                case 'open':
                    value = 100;
                    break;

                case 'close':
                    value = 0;
                    break;

                default:
                    return;
            }
        }

        const invert = utils.getMetaValue(entity, meta.mapped, 'coverInverted', 'allEqual', false)
            ? !meta.options.invert_cover
            : meta.options.invert_cover;
        const position = invert ? 100 -  value : value;
        const level = utils.mapNumberRange(Number(position), 0, 100, 0, 255).toString();

        await entity.command(
            'genLevelCtrl',
            'moveToLevelWithOnOff',
            {
                level: level,
                transtime: 0
            },
            utils.getOptions(meta.mapped, entity),
        );

        return {state: {position: value}, readAfterWriteTime: 0};
    },
    convertGet: async (entity, key, meta) => {
        await entity.read('genLevelCtrl', ['currentLevel']);
    },
};

module.exports = [
    {
        fingerprint: [
            { manufId: 4368, endpoints: [ { ID: 1, profileID: 260, deviceID: 513, inputClusters: [ 0, 3, 21 ], outputClusters: [ 3, 4, 5, 6, 8, 256, 64544, 64545 ] } ] },
        ],
        model: 'Remote',
        vendor: 'Profalux',
        description: 'Store profalux',
        fromZigbee: [],
        toZigbee: [],
        exposes: [],
    },
    {
        fingerprint: [
            { manufId: 4368, endpoints: [ { ID: 1, profileID: 260, deviceID: 512, inputClusters: [ 0, 3, 4, 5, 6, 8, 10, 21, 256, 64544, 64545 ], outputClusters: [ 3, 64544 ] } ] },
        ],
        model: 'Store',
        vendor: 'Profalux',
        description: 'Store profalux',
        fromZigbee: [fz.cover_position_via_brightness, fz.cover_state_via_onoff],
        toZigbee: [tzCover],
        exposes: [e.cover_position().setAccess('state', ea.ALL)],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);

            await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl']);
            await reporting.brightness(endpoint);
        },
    },
];
```

rclsilver, 2022-02-12:
Hi @Koenkk, the converters PR has been merged.

Could you close this issue once the new version of the converters will be used?

deliwyn, 2022-07-16:
Hi! I have Profalux Zigbee covers but the previous generation. As per vendor, it should be referenced as "NSAV029". I currently used Deconz, but I'd like to switch to Zigbee2MQTT. In Deconz, my Profalux covers are managed as "bulb" (up/down is managed through brightness). Do you fix I would be able to pair them as covers using Zigbee2MQTT? Thanks.

deliwyn, 2022-07-16:
I did a test with one of my Profalux cover. Both cover and remote are discovered and appear to be supported out of the box.
[image]
However, I cannot control the cover, nothing happens when I try to open/close it.

rclsilver, 2022-07-16:
You should have two devices when linking your covers:
- one for the remote
- one for the cover

deliwyn, 2022-07-16:
Yes, I can see both cover and remote.
[image]

What do you mean by metadata? Sorry, I'm very new to Zigbee2MQTT but I already like it very much :) 

Here is what I captured in debug mode while pairing again the cover/remote :

```
info 2022-07-16 12:38:08Succesfully changed options
info 2022-07-16 12:38:08MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"k2qla-1"}'
debug 2022-07-16 12:38:19Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"k2qla-2","value":true}'
info 2022-07-16 12:38:19Zigbee: allowing new devices to join.
info 2022-07-16 12:38:20MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"k2qla-2"}'
info 2022-07-16 12:38:39Device '0x20918affff00728d' joined
info 2022-07-16 12:38:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918affff00728d","ieee_address":"0x20918affff00728d"},"type":"device_joined"}'
info 2022-07-16 12:38:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x20918affff00728d"},"type":"device_connected"}'
info 2022-07-16 12:38:39MQTT publish: topic 'homeassistant/cover/0x20918affff00728d/cover/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x20918affff00728d/set","device":{"identifiers":["zigbee2mqtt_0x20918affff00728d"],"manufacturer":"Profalux","model":"Cover (NSAV061)","name":"0x20918affff00728d"},"json_attributes_topic":"zigbee2mqtt/0x20918affff00728d","name":"0x20918affff00728d","position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0x20918affff00728d","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x20918affff00728d/set","unique_id":"0x20918affff00728d_cover_zigbee2mqtt"}'
info 2022-07-16 12:38:39MQTT publish: topic 'homeassistant/sensor/0x20918affff00728d/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x20918affff00728d"],"manufacturer":"Profalux","model":"Cover (NSAV061)","name":"0x20918affff00728d"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x20918affff00728d","name":"0x20918affff00728d linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x20918affff00728d","unique_id":"0x20918affff00728d_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2022-07-16 12:38:39Starting interview of '0x20918affff00728d'
info 2022-07-16 12:38:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918affff00728d","ieee_address":"0x20918affff00728d","status":"started"},"type":"device_interview"}'
info 2022-07-16 12:38:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x20918affff00728d"},"type":"pairing"}'
debug 2022-07-16 12:38:39Received MQTT message on 'homeassistant/cover/0x20918affff00728d/cover/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x20918affff00728d/set","device":{"identifiers":["zigbee2mqtt_0x20918affff00728d"],"manufacturer":"Profalux","model":"Cover (NSAV061)","name":"0x20918affff00728d"},"json_attributes_topic":"zigbee2mqtt/0x20918affff00728d","name":"0x20918affff00728d","position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/0x20918affff00728d","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/0x20918affff00728d/set","unique_id":"0x20918affff00728d_cover_zigbee2mqtt"}'
debug 2022-07-16 12:38:39Received MQTT message on 'homeassistant/sensor/0x20918affff00728d/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x20918affff00728d"],"manufacturer":"Profalux","model":"Cover (NSAV061)","name":"0x20918affff00728d"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x20918affff00728d","name":"0x20918affff00728d linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x20918affff00728d","unique_id":"0x20918affff00728d_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
debug 2022-07-16 12:38:40Received Zigbee message from '0x20918affff00728d', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
info 2022-07-16 12:38:42Device '0x20918affff00728e' joined
info 2022-07-16 12:38:42MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x20918affff00728e","ieee_address":"0x20918affff00728e"},"type":"device_joined"}'
info 2022-07-16 12:38:42MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x20918affff00728e"},"type":"device_connected"}'
info 2022-07-16 12:38:42MQTT publish: topic 'homeassistant/sensor/0x20918affff00728e/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x20918affff00728e"],"manufacturer":"Profalux","model":"Cover remote (NB102)","name":"0x20918affff00728e"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attrib [... cut ...]
info 2022-07-16 12:39:07MQTT publish: topic 'zigbee2mqtt/smartswitch_chambre_nous', payload '{"action":null,"battery":100,"click":null,"device_temperature":30,"linkquality":255,"power_outage_count":11,"voltage":3035}'
debug 2022-07-16 12:39:59Received Zigbee message from 'motion_chambre_nous', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null
info 2022-07-16 12:39:59MQTT publish: topic 'zigbee2mqtt/motion_chambre_nous', payload '{"battery":48,"battery_low":false,"linkquality":255,"occupancy":false,"tamper":false,"voltage":2800}'
```

deliwyn, 2022-07-17:
Hi @rclsilver !

I did some additional tests today. I'm able to link my Profalux covers and remotes with Zigbee2MQTT.
However, when trying to open or close the cover, it only does a quick up & down (like when it confirms something).
In the other hand, when using the remote to open or close the cover, if I use "STOP" function in Zigbee2MQTT, it stops the cover correctly.

Is there anything I can do or I can share with you so you can help me? Thank you :)

rclsilver, 2022-07-18:
Hello, you have to resync again and calibrate the cover.
I written a procedure (in french) available hère: https://community.jeedom.com/t/tutoriel-integration-volet-profalux/73689/28

deliwyn, 2022-07-18:
Thank you @rclsilver ! I missed to do the calibration process (2/3 up & down cycles).
I successfully paired and managed 4 of my 11 Profalux covers using Z2M ! We'll include the others tomorrow. Thank you again.