Koenkk/zigbee2mqtt #11368: Xiaomi Aqara SSM-U01 not reporting power after update to 1.23.0-1

miljbee, 2022-02-11:
### What happened?

Hello, I have a few Xiaomi Aqara SSM-U01. After upgrading to 1.23, the power is not reported anymore. 
Back to 1.22, power is back.
Before rolling back, I went to dev console and requested haElectricalMeasurement/activePower. I got a correct raw value.
I also tried to repost the report interval without any luck.

Can it be related to the changes implemented here ?
https://github.com/Koenkk/zigbee2mqtt/issues/10709

### Zigbee2MQTT version

1.23

### Adapter firmware version

20211217

### Adapter

CC2652RB

### Debug log

Sorry, forgot to save the log before rolling back

McGiverGim, 2022-02-11:
Maybe you have a different firmware version than me? Because the changes made my device work correctly.

Koenkk, 2022-02-12:
Could you try to repair the device? If you paired the device with an older coordinator fw (before you updated to 20211217) the Xiaomi device will not be in "Xiaomi mode"

miljbee, 2022-02-12:
just pairing again by pressing the pairing button for 8s ? or do I need to unpair / delete the devices before ?

miljbee, 2022-02-12:
upgraded to 1.23, the one I repaired now correctly reports power.
I will confirm with the other ones that a repair is needed with these modules when upgrading to 1.23.

miljbee, 2022-02-12:
Just repaired another one in 1.23. It's ok, I get the power. But in the report tab, I lost power and currentSummDelivred :

[image]

The one I repaired in 1.22 still have them :

[image]

I can probably add them manualy

McGiverGim, 2022-02-12:
The temperature and illuminance is fixed in current master. It will be fixed in next version.

Koenkk, 2022-02-12:
> It would be great if you could bring back the report interval for active power and currentSummDelivred. New users of this device will have a better experience with this.

This device doesn't use the standard clusters for power reporting anymore when using it together with coordinator fw >= 20211217, with this fw the Xiaomi device will think it has been paired to a Xiaomi gateway which enables all features.

miljbee, 2022-02-12:
> This device doesn't use the standard clusters for power reporting anymore when using it together with coordinator fw >= 20211217, with this fw the Xiaomi device will think it has been paired to a Xiaomi gateway which enables all features.

You mean reporting interval can't be set anymore ? That's a regression ....

I had a look at the console and saw that the messages comes from AqaraOpple/attributeReport. Is there some way to set the reporting interval for this one ?
Is there some way to force an update of one or all the values ? A magic mqtt message ?

Also, while I was looking at the console, I saw that the data from AqaraOpple/attributeReport contains voltage, current and temperature (ambient) which are not exposed. It would be great to have some sensors for these attributes.

[image]

Koenkk, 2022-02-13:
> I had a look at the console and saw that the messages comes from AqaraOpple/attributeReport. Is there some way to set the reporting interval for this one ?
Is there some way to force an update of one or all the values ? A magic mqtt message ?

I'm not aware of this, we event don't configure this explicitly, the device just detects it has been connected to a Xiaomi bridge and configures itself internally.

> Also, while I was looking at the console, I saw that the data from AqaraOpple/attributeReport contains voltage, current and temperature (ambient) which are not exposed. It would be great to have some sensors for these attributes.

Added this.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-23 by github-actions[bot]

jkellerm, 2022-08-30:
I had the same problem. My SSM-U01s were added in version 1.24.x and not showing power. Still not showing power in 1.27.0. I tried resetting the devices as mentioned in this post and related posts about similar issues. This did not work. Today I upgraded the firmware of my sonoff zigbee 3.0 usb dongle plus to version 20220219 from version 20211217 and after pressing the button on the SSM-U01 for a few seconds it was re-added and power is reported! 

I now this post is closed, but it might save others with a similar issue a lot of time.

ghenzler, 2023-10-18:
I had the same problem and I also updated the firmware for my Sonoff *ZBDongle-P* (from a 2021 version to 20230507) and repaired just only the Aqara SSM-U01 device to fix it. 

There are a lot of guides out there on how to do it, but partly they rely on Windows software (I'm on a Mac) and seemed complicated with many steps. The best way I found for me was using https://github.com/JelmerT/cc2538-bsl - the nice thing is that it puts the device into boot mode *and* actually flushes the device (so no need to open the enclosure and press the button, no need for extra software except python, no need for a script that only starts the boot mode but doesn't actually flash). 

https://www.zigbee2mqtt.io/guide/adapters/#recommended
method 1 in https://sonoff.tech/wp-content/uploads/2021/12/SONOFF-Zigbee-3.0-USB-dongle-plus-firmware-flashing-1-1.pdf - 
https://github.com/JelmerT/cc2538-bsl
https://youtu.be/iCE5Z43EKpk?t=246

I was also unclear if the firmware update would maybe cause the whole network to lose existing connections to all the other devices (that would have been awful for doing it for 50+ devices) - but it worked fine without repairing as documented:
https://www.zigbee2mqtt.io/guide/faq/#doesn-t-require-repairing