Koenkk/zigbee2mqtt #11374: Innr SP 234 Smart Plug 15A leaves the network in less then 12 hours

kris-anderson, 2022-02-12:
### What happened?

I added the Innr SP 234 device to my network and everything works well. But within less then 12 hours, the device leaves my network.

I ordered a second plug of the same make/model to see if maybe the first plug was at fault, but the second plug also has the same issue. I also have 4 Innr SP 224 plugs and they don't have this issue. I also have 40 devices on my Zigbee network and none of them drop off except for these SP 234 plugs.

I read online somewhere that this plug has issues with power meter reports, but I disabled those reports and only reported the onOff attribute, but it still fell of my network.

When I check the logs, I see these showing up for that device when it leaves my network:
```txt
info  2022-02-11 02:19:03: MQTT publish: topic 'zigbee2mqtt/testtest', payload '{"current":0,"device":{"applicationVersion":40,"dateCode":"20210518","friendlyName":"testtest","hard
wareVersion":1,"ieeeAddr":"0x847127fffe980475","manufacturerID":4454,"manufacturerName":"innr","model":"SP 234","networkAddress":12027,"powerSource":"Mains (single phase)","softwar
eBuildID":"2.8","stackVersion":102,"type":"Router","zclVersion":3},"energy":null,"last_seen":"2022-02-11T02:19:03-08:00","linkquality":91,"power":0,"state":"OFF","voltage":0}'
info  2022-02-11 02:23:56: MQTT publish: topic 'zigbee2mqtt/testtest', payload '{"current":0,"device":{"applicationVersion":40,"dateCode":"20210518","friendlyName":"testtest","hard
wareVersion":1,"ieeeAddr":"0x847127fffe980475","manufacturerID":4454,"manufacturerName":"innr","model":"SP 234","networkAddress":12027,"powerSource":"Mains (single phase)","softwar
eBuildID":"2.8","stackVersion":102,"type":"Router","zclVersion":3},"energy":null,"last_seen":"2022-02-11T02:23:56-08:00","linkquality":94,"power":0,"state":"OFF","voltage":0}'
warn  2022-02-11 02:25:23: Device 'testtest' left the network
info  2022-02-11 02:25:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"testtest","ieee_address":"0x847127fffe980475"},"type":"device_leave"}'warn  2022-02-11 02:25:23: Device 'testtest' left the network
info  2022-02-11 02:25:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"testtest","ieee_address":"0x847127fffe980475"},"type":"device_leave"}'warn  2022-02-11 02:25:23: Device 'testtest' left the network
info  2022-02-11 02:25:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"testtest","ieee_address":"0x847127fffe980475"},"type":"device_leave"}'
```

Here's the plug details:
https://www.innr.com/en/product/smart-plug-15a/
https://www.zigbee2mqtt.io/devices/SP_234.html

### What did you expect to happen?

I expected my plug to stay on my network.

### How to reproduce it (minimal and precise)

1. Join the plug to the network
2. Verify the plug works
3. Wait
4. Within 12 hours the device will no longer be visible in the Zigbee2mqtt interface

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

Koenkk, 2022-02-12:
Do you maybe have a CC2531 dongle? If so it would be interesting to see the traffic at the moment the device leaves: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

If you are able to sniff please share the `.pcapng` file + network key here.

kris-anderson, 2022-02-12:
I'm using a CC2652P dongle flashed with the latest coordinator firmware from the master branch.

jschornick, 2022-03-14:
To add a data point, I'm also seeing a similar problem with my Innr SP 234. 

Two times it has fallen off the network and is unavailable in the device list until power cycling the plug and permitting it to rejoin the network.

The strange thing is that I had zero problems with this device before I made two changes. 
1) I added two Innr SP 224 plugs to the network (total of 3 routers, all Innr plugs)
2) I put a load on the SP 234 to test the power monitoring

I'm currently testing the SP 234 "unloaded" (nothing plugged into it) to see if it stops responding with only condition 1 present.

If I can reproduce reliably, please let me know how I can help debug.

Zigbee2MQTT: 1.24.0
Adapter: Sonoff Zigbee 3.0 USB Dongle Plus
Adapter Firmware: zStack3x0 (revision 20211217)
Innr firmware: 2.8 (build 20210518)

jschornick, 2022-03-16:
Experienced another network drop with nothing plugged into the SP 234. Not a huge surprise.

I realized that I also may have enabled the "availability" feature in zigbee2mqtt before the failures began. I've now disabled availability checking on the chance that the active checks are somehow responsible. Unlikely since this device checks in quite regularly on its own.

Standard logs at the "info" level don't reveal any interesting events prior to the device leaving the network. I'll enable debug logging if the problem persists.

jschornick, 2022-03-17:
After disabling the zigbee2mqtt's availability checks, I've now gone over 24 hours without having my SP 234 drop off the network. This is with the 2 additional SP 224 routers also connected.

It may be a red herring, but worth testing further. I'm going to give it another day to see if the problematic plug stays connected, then will re-enable availability checks to see if they again induce the plug to drop off. 

@kris-anderson It would be great to know if you had the availability feature enabled in zigbee2mqtt when you saw your drop-offs occur.

jschornick, 2022-03-17:
Thanks, @kris-anderson

Sadly, it looks like this was indeed a red herring. Even though disabling availability did seem to affect the frequency of the problem, the issue is clearly still present. 

As an alternate tactic, I ordered a second USB dongle, which I'll try to convert into a zigbee and/or 802.15.4 sniffer. Any guidance on captures or generally debugging in that regard would be appreciated.

srb37, 2022-03-28:
I'm having the same issue with the SP234.  

The SP234 joined at 17:18:05, published its last update at 23:51:19 and left at 23:51:27 so only stayed on the network about 6 hours.  

I'll try to run a sniffer, but I'm not sure how difficult it will be given the random disconnect time.

Zigbee2MQTT version
: 1.24.0
Adapter: Electrolama ZZH CC2652R
Coordinator type
: zStack3x0
Coordinator revision
: 20211217

srb37, 2022-03-29:
@Koenkk I was able to grab a capture at the moment the SP234 left the network.  See attached.

`info ^[[39m 2022-03-28 18:31:33: MQTT publish: topic 'zigbee2mqtt/WashingMachineOutlet', payload '{"current":0.05,"linkquality":109,"power":0,"state":"ON","voltage":126}'
warn ^[[39m 2022-03-28 18:31:38: Device 'WashingMachineOutlet' left the network
info ^[[39m 2022-03-28 18:31:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"WashingMachineOutlet","ieee_address":"0x84fd27fffe44f5ce"},"type":"device_leave"}'
info ^[[39m 2022-03-28 18:31:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84fd27fffe44f5ce"},"type":"device_removed"}'
warn ^[[39m 2022-03-28 18:31:39: Device 'WashingMachineOutlet' left the network
info ^[[39m 2022-03-28 18:31:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"WashingMachineOutlet","ieee_address":"0x84fd27fffe44f5ce"},"type":"device_leave"}'
info ^[[39m 2022-03-28 18:31:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84fd27fffe44f5ce"},"type":"device_removed"}'
warn ^[[39m 2022-03-28 18:31:39: Device 'WashingMachineOutlet' left the network
info ^[[39m 2022-03-28 18:31:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"WashingMachineOutlet","ieee_address":"0x84fd27fffe44f5ce"},"type":"device_leave"}'
info ^[[39m 2022-03-28 18:31:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x84fd27fffe44f5ce"},"type":"device_removed"}'
`
[sp234extract.zip](https://github.com/Koenkk/zigbee2mqtt/files/8367318/sp234extract.zip)

Koenkk, 2022-03-29:
@srb37 so what happens is that the device rejoins (not a problem) but the coordinator refuses to resend the network key. What devices are the ones with network address `0x58a8` and `0x5471` (you can find network address in the zigbee2mqtt forntend)? Could you try repairing those close to the coordinator and see if the issue happens again?

srb37, 2022-03-29:
Thanks @Koenkk.  Both of those devices are Innr SP224's.  I've gone ahead and removed them from the network and repaired.   
Just for some additional my Zigbee2MQTT installation first started with a CC2531, I've since migrated to a an Electrolama ZZH.  I copied the ieee address from the old adapter into the new one.  The SP224's are some of my oldest devices and were paired with the original CC2531, the SP234 is a newer device and was likely paired post adapter switch.

srb37, 2022-03-30:
@Koenkk the SP234 dropped off again last night.  I removed the two SP224 and actually only re-paired one of them, 0xccccccfffed05481 (0x7D1E).  I wonder if I need to remove and repair all routers?  Seems like it maybe related to my migration from the CC2531 to the ZZH, I feel like I've seen some other issues that mention problems with that migration.

See attached sniff.
[IssueRecurrance.zip](https://github.com/Koenkk/zigbee2mqtt/files/8380127/IssueRecurrance.zip)

Koenkk, 2022-03-30:
Same issue, device tries to pair via `0x7D1E` but this should succeed now because you repaired the device. Could you stop zigbee2mqtt and provide me a copy of your `coordinator_backup.json` file?

Koenkk, 2022-04-01:
The coordinator_backup.json looks good, but I'm not able to decrypt the "APS: command" (No. 149) in the sniff with the linkkey from the coordinator_backup.json
- Just to be sure; could you update to the latest coordinator firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
- Remove `0xccccccfffed05481` via frontend
- Repair `0xccccccfffed05481`
- Restart zigbee2mqtt
- Start sniffing
- Wait till the SP 234 disconnects again
- Provide the sniff + your `coordinator_backup.json` here.

srb37, 2022-04-02:
@Koenkk thanks for all of your help with this.  The issue recurred again last night.
I followed all of the directions above.  I've gone ahead an included the disconnect sniff (sp234left), sniffs from when I repaired 0xccccccfffed05481 (sp224pairing) and when I repaired the SP234 (sp234pairing).  I've also included the latest coordinator_backup.json. 
[2022-04-02 Issue.zip](https://github.com/Koenkk/zigbee2mqtt/files/8402447/2022-04-02.Issue.zip)

Koenkk, 2022-04-03:
Seems I overlooked a crucial part; it is actually the coordinator that asks the SP 234 to leave:

[image]

Currently I can think of 2 possible root causes:
- This time the SP 234 paired via 0x24fd5b00010afdf6 (0x24ee), looks to be a SmartThings device. Has this device already been repaired after the migration from CC2531 -> CC2652?
- Something is wrong on in the coordinator firmware

I propose the following next step:
1. Pair SP 234 very close to the coordinator, sniff this to verify that it joins directly via the coordinator. The source of the assocation response should be the iee address of your coordinator (0x00124b000e0ead84 in your case), example: <br/> [image]
1. Keep it there, so the SP 234 doesn't switch parents
2. Does the device still leave? If yes provide a sniff of that. In this case I can provide you a coordinator firmware with debug logging so I can backtrack where the problem occurs.

srb37, 2022-04-04:
@Koenkk The issue is still recurring.  Here's what I've done:
1. I paired the SP 234 right next to the coordinator and verified with the sniffer that the association response came from the coordinator.
2. I also removed and re-paired 0x24ee as it likely wasn't paired post migration to CC2652.  I verified that it's association response also came from the coordinator. 
3. I had one additional SmartThings plug (0x7CCA) and figured for good measure I'd remove and re-pair it, again verified it was paired via the coordinator.  
4. SP234 remained right next to the coordinator with the other outlets in a different room.  The issue recurred about 5-6 hours later with the coordinator asking the SP234 to leave.  

Attached is the sniff from when the SP234 left.

[234left2022-04-03.zip](https://github.com/Koenkk/zigbee2mqtt/files/8406077/234left2022-04-03.zip)

Koenkk, 2022-04-04:
Ok great,

- Could you flash the following fw on your coordinator: [znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8412280/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)
-  Start zigbee2mqtt with herdsman debug logging enabled: 
See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.
- Permit and disable joining once (this is for me to verify that the debug logging is working)
- Wait till the SP 234 leaves again
- Provide the log here.

srb37, 2022-04-06:
@Koenkk sorry for the delay, hopefully I did everything correctly.  I'm using PM2 to start zigbee2mqtt.  I went ahead and created a separate pm2 service with the herdsman debug commands.  It logs STDOUT and STDERR to separate files.   I've gone ahead and attached the Out & Error (herdsman debug) for the permit/disable join and when 234 left the network.  

NOTE: It appears the error/debug logging is in GMT while the OUT is in EDT, so there is a 4 hour off-set between the two log files.

I've also attached a sniff of 234 leaving the network.  Hope this gives you what you need.  Thanks again for your help.

[debuglogs.zip](https://github.com/Koenkk/zigbee2mqtt/files/8434656/debuglogs.zip)

Koenkk, 2022-04-07:
Perfect, seems that the firmware call the device remove function which triggers the leave. I've added some more logging to more accurately determine where things go wrong. Could you provide the log again of the leave? Just `leftthenetwork-debug.log` is enough, don't need any other logging or sniffs.

[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex 2.zip](https://github.com/Koenkk/zigbee2mqtt/files/8444683/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.2.zip)

srb37, 2022-04-08:
I flashed the new debug firmware last night and expected to wake up this morning to a disconnect--so far the 234 hasn't left the network.  It's been about 17 hours which is the longest I recall it staying on the network.  I'll keep the debugging running, but it's interesting that it hasn't left yet.

srb37, 2022-04-09:
@Koenkk uptime 36 hours and it still hasn't dropped off.  Could anything else have changed in the new firmware you provided?

Koenkk, 2022-04-09:
@srb37 I've seen this earlier with a CC2531, after adding debug logging for a particular problem it disappeared, probably because the timing gets different. Lets see if it stays connected.

srb37, 2022-04-15:
The SP234 remains connected, uptime at this point is 7 days, debug logging is still enabled.  @Koenkk is there anything else I should do at this point?  The outlet is still next to the controller and doesn't have a load plugged into it, I could move it to it's normal location, but not sure that would help with debugging.  I can also just keep debugging on forever, it doesn't seem to be impacting the usability of Zigbee2MQTT.

Koenkk, 2022-04-15:
@srb37 could you keep the SP234 at the same location and switch back to the default firmware again and see whether the problem pops up again?

srb37, 2022-04-15:
@Koenkk, to confirm I am to reflash with the default firmware and start Zigbee2MQTT without debugging enabled?

Koenkk, 2022-04-16:
@srb37 yes, and see if the problem reappears.

srb37, 2022-04-17:
@Koenkk after flashing with the regular firmware the issue recurred and the SP234 left the network about 6.5 hrs after Zigbee2MQTT was restarted 12:06 - 18:41.

Koenkk, 2022-04-17:
Can you try with
[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8502075/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip) 

This has some of the debug logging removed.

srb37, 2022-04-19:
@Koenkk I flashed the controller with the new firmware provided.  Attached is the log from when it left the network. 

[zigbee-error-2022-04-19.log](https://github.com/Koenkk/zigbee2mqtt/files/8512396/zigbee-error-2022-04-19.log)

Koenkk, 2022-04-20:
Can you try with this firmware? I don't need any sniffs or logging, just let me know if it stays connected or not.
[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8523362/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)

srb37, 2022-04-23:
@Koenkk I flashed the controller with the new firmware, but it didn’t fix the problem, saw a disconnect after about 6 hrs. Interestingly enough, about 30 mins after it left the network I lost communication with all zigbee devices and had to restart zigbee2mqtt. I haven’t had to do that in a while.

Koenkk, 2022-04-23:
Can you try with this fw?
[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8546520/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)

What I'm trying to do here is to replace the debug function (which creates some kind of delay) with a sleep function (10ms for now).

srb37, 2022-04-23:
@Koenkk thanks, it left again with the new firmware. Still after about 6 hours.

Koenkk, 2022-04-24:
Increased the sleep to 200ms (from 10ms), can you try again?
[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8549611/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)

laskatj, 2022-04-25:
I have a similar problem with my sp234. I contacted support and they acknowledged some firmware issues that were causing this behavior 

“ seems that the SP 234 does indeed have some issues where the devices keep disconnecting from the hub. This happens with all SP 234's. This issue needs to be fixed by an FW update.

Since we don't offer our Innr bridge in the US, we will also have to figure out how we will update the devices. Most likely it will be by replacing the smart plugs. We hope to have a new batch of smart plugs (with the correct FW) by the end of April/beginning of May. After these "new" smart plugs are in stock, I can provide you with the replacement.

Koenkk, 2022-04-25:
@laskatj if they can provide the OTA file we can support the OTA update through z2m

srb37, 2022-04-26:
@Koenkk it disconnected again after about 6 hours. Let me know if you want to try and increase the sleep time again. 

@laskatj interesting feedback from the manufacturer. Let us know if they will provide the OTA file @Koenkk referenced.

Koenkk, 2022-04-26:
@srb37 

1. Could you flash firmware `1.hex` and see if the device still leaves?
    - If it leaves, please provide me the herdsman debug logging (no need to do step 2)
    - If it does not leave continue with step 2
2. Could you flash firmware `2.hex` and see if the device still leaves? If it does let me know, no need to gather debug logging since this fw doesn't output it.

[fws.zip](https://github.com/Koenkk/zigbee2mqtt/files/8565640/fws.zip)

laskatj, 2022-04-26:
@Koenkk  @srb37   I asked them again to consider allowing the option for folks to download the firmware when it's ready but it didn't sound like they want to deal with that and are opting for replacing affected devices.  I pushed again suggesting that it would save them cost if the community were able to upgrade it themselves.  Will let you know if they respond.

srb37, 2022-04-28:
@Koenkk, flashed with 1.hex, device still left.  Attached is the debug.  I'll try to flash with 2.hex tonight and will let you know the result.
[2022-04-27Debug.log](https://github.com/Koenkk/zigbee2mqtt/files/8578618/2022-04-27Debug.log)

Koenkk, 2022-04-28:
great that you got some logging, I see where it goes wrong now (and don't understand why it didn't happen with https://github.com/Koenkk/zigbee2mqtt/issues/11374#issuecomment-1091907830). Can you provide the logging again with:
[3.zip](https://github.com/Koenkk/zigbee2mqtt/files/8584543/3.zip)

srb37, 2022-04-29:
@Koenkk yes perhaps it was a fluke that is stayed on with the prior debugging firmware.  Attached is the debug from firmware 3.
[2022-04-29Debug.log](https://github.com/Koenkk/zigbee2mqtt/files/8590701/2022-04-29Debug.log)

Koenkk, 2022-04-29:
Ah sorry, I forgot to revert back to the actual logging instead of the sleep; so there is not debug logging in your log. Could you try again with:
[4.zip](https://github.com/Koenkk/zigbee2mqtt/files/8594598/4.zip)

srb37, 2022-04-30:
@Koenkk attached is the debug log with firmware 4.  
[2022-04-30Debug.log](https://github.com/Koenkk/zigbee2mqtt/files/8596789/2022-04-30Debug.log)

Koenkk, 2022-05-02:
Could you check if the device stays connected with
[5.zip](https://github.com/Koenkk/zigbee2mqtt/files/8604282/5.zip) (if not provide the debug logging again)

Could you also again check the fw from https://github.com/Koenkk/zigbee2mqtt/issues/11374#issuecomment-1091907830 , I'm still curious if it stays connected with that fw (like it did before)

srb37, 2022-05-03:
@Koenkk it's been running on firmware 5 for 24 hours and hasn't disconnected.  Usually the disconnect occurs after 6 hours so I think we're in the clear.  Should I turn off herdsman debug logging and see if it remains connected?  I haven't tried the firmware from https://github.com/Koenkk/zigbee2mqtt/issues/11374#issuecomment-1091907830 yet.

Koenkk, 2022-05-04:
firmware 5 disables some security related things, so this seems to fix this but this is not really acceptable. Could you check again with the fw from https://github.com/Koenkk/zigbee2mqtt/issues/11374#issuecomment-1091907830 ?

Koenkk, 2022-05-11:
@srb37 unfortunately I lost the changes I did those this firmware, trying to reproduce it now. Can you try with this fw? If it fails provide me the herdsman debug logging.
[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8672506/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)

xconverge, 2022-05-22:
Happy to see you guys chugging away at this, it is definitely a shame for the product, but appreciated work @Koenkk !

I wanted to let you know I had this problem in deconz with a conbee II and had tested zigbee2mqtt (with a cc2531) to confirm it was device specific and not coordinator specific at the time. The cc2531 is now my sniffer and I had provided several dumps to innr. The power readings being off by a factor of 10 was also quite annoying but at least was acknowledged as a bug by innr, not sure if they have fixed it on later units.

Here was when I gave up, I had gone through a few devices and a lot of back and forth with innr since late 2021.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5256#issuecomment-1014844589

Koenkk, 2022-05-26:
@srb37 any updates?

srb37, 2022-05-26:
@Koenkk sorry for the delay. Uptime 12 days with no disconnects.  I still have debug logging enabled. Let me know if there is anything you’d want to see from the debug logs or if we should try with debugging off.

Koenkk, 2022-05-27:
Great, so lets see what prevents the issue from happening, please try out the firmwares in the zip starting by 1 and stop when the device starts to leave again (lets assume it stays working if it stays connected for at least a day)

[Archive.zip](https://github.com/Koenkk/zigbee2mqtt/files/8787465/Archive.zip)

srb37, 2022-05-28:
Thanks @Koenkk, will do.  Should I enable debugging or run without debugging?

Koenkk, 2022-05-29:
@srb37 without

srb37, 2022-05-29:
@Koenkk not sure if this was expected, but it left on firmware 1.  Should I move on to 2 or do I need a new batch?

Koenkk, 2022-05-29:
@srb37 does it stay connected for longer than a day with firmware 1? If yes, then move to 2, if that works move to 3.

srb37, 2022-05-29:
No, it left after 6.5 hours with firmware 1. I did have debugging on, not sure if that messed things up.

Koenkk, 2022-05-30:
Thats good, can you confirm that with this fw the device stays connected?

[znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8799458/znp_CC26X2R1_LAUNCHXL_tirtos_ccs.hex.zip)

srb37, 2022-06-03:
@Koenkk, no luck with that firmware, it disconnected after its normal 6 hours.

Koenkk, 2022-06-04:
I just noticed your comment from https://github.com/Koenkk/zigbee2mqtt/issues/11374#issuecomment-1083040230

> Seems like it maybe related to my migration from the CC2531 to the ZZH, I feel like I've seen some other issues that mention problems with that migration.

Have you already done this in the meantime? This could definitely cause such issue.

[21 more replies]

closed 2022-07-30 by github-actions[bot]