Koenkk/zigbee2mqtt #11377: Going from advanced availability to simple errors out

atv2016, 2022-02-12:
### What happened?

Errors
should be boolean
should be object
should match exactly one schema in oneOf
Availability
Checks whether devices are online/offline

Availability (simple)
should be boolean
should be object
should match exactly one schema in oneOf
should be boolean
should be object
should match exactly one schema in oneOf

### What did you expect to happen?

To go back to simple availability

### How to reproduce it (minimal and precise)

Go from advanced back to simple and hit save.

### Zigbee2MQTT version

Latest

### Adapter firmware version

0x26720700

### Adapter

Conbee-II

### Debug log

See above, shows nothing in debug log.

Koenkk, 2022-02-12:
I can reproduce this, seems to be a frontend bug (nothing is send to z2m when switching from advanced -> simple disabled, advanced -> simple disabled -> simple enabled -> simple disabled does work though.

@nurikk can you check this?

[image]

McGiverGim, 2022-02-17:
I've suffered the same, but I have another question about this: it seems that the values here are not in the "main config file" of the HA addon, and the inverse.
How it works? If I add something to the yaml in the HA addon:
```yaml
availability:
  active:
    timeout: 20
  passive:
    timeout: 100
```
It does not appear in the options inside the frontend, and it seems the values inside the frontend are the valid ones and this are ignored. Must the value appear the same in both sides?

closed 2022-03-27 by github-actions[bot]