Koenkk/zigbee2mqtt #11378: Incorrect reading from Develco EMI Norwegian HAN - EMIZB-132

janbanan007, 2022-02-12:
### What happened?

The value current_phase_b, is not correct. 
Using these values 6554 A x 238.8 V = 1 565 kW, which is way to high energy usage.

`info  2022-02-12 09: 58: 12: MQTT publish: topic 'zigbee2mqtt/HAN', payload '{
  "current": 4.1,
  "current_phase_b": 6554,
  "current_phase_c": 11,
  "energy": 57875.14,
  "linkquality": 39,
  "power": 2863,
  "update": {
    "state": "idle"
  },
  "update_available": null,
  "voltage": 238.8,
  "voltage_phase_b": 237,
  "voltage_phase_c": 237
}'`

### What did you expect to happen?

In addition to the wrong values to be corrected, there are some strange field names.

current should be renamed to current_phase_a
voltage should be renamed to voltage_phase_a

This will make all values to follow a similar pattern.

### Zigbee2MQTT version

1.23.0 commit: 6ea61bae

### Adapter firmware version

20211115

### Adapter

CC2531

janbanan007, 2022-02-12:
Cluster definition from [EMI Norwegian HAN Technical Manual](https://www.develcoproducts.com/media/1783/emizb-132-technical-manual-emi-norwegian-han.pdf)

4.3 Electrical Measurement – Cluster id 0x0B04

Id# | Name | Type | Man  /Opt | Relevance and ref.
-- | -- | -- | -- | --
0x0000 | MeasurementType | BitMap32 | O | Section 4.3.1
0x0304 | TotalActivePower | Int32 | O
0x0305 | TotalReactivePower | Int32 | O
0x0402 | PowerMultiplier | Uint32 | O | Fixed to 1
0x0403 | PowerDivisor | Uint32 | O | Fixed to 1000
0x0505 | RMSVoltagePhA | Uint16 | O | L1 - Volts (V)
0x0508 | RMSCurrentPhA | Uint16 | O | L1 - Amps (A)
0x0600 | ACVoltageMultiplier | Uint16 | O | Depending on attached meter
0x0601 | ACVoltageDivisor | Uint16 | O | Depending on attached meter
0x0602 | ACCurrentMultiplier | Uint16 | O | Depending on attached meter
0x0603 | ACCurrentDivisor | Uint16 | O | Depending on attached meter
0x0905 | RMSVoltagePhB | Uint16 | O | Fixed to 1
0x0908 | RMSCurrentPhB | Uint16 | O | Fixed to 1
0x0A05 | RMSVoltagePhC | Uint16 | O | Fixed to 1
0x0A08 | RMSCurrentPhC | Uint16 | O | Fixed to 1

<br class="Apple-interchange-newline">

mikke49, 2022-02-13:
I'm using the same Develco device, and I see readings like this from time to time, maybe twice a month. However all readings are then off at the same time, not just one like yours, i.e. all show default values. Energy and power drops to 0 and current and volt all show 6553.5. Looks like some kind of reset on the device or something. The next reading is then ok again. 

Do you have this value constant, or just one single reading (sometimes)? 

But I notice another problem in the MQTT message you attached. The readings for current and volt on phases B and C have no fractional digits. The reading of current phase B should actually have showed 6553.5 - not 6554. I have the same situation, phase A shows values with fractional digits while phases B and C are rounded to integers. I experience this problem after update Zigbee2mqtt to version 1.23.0. Before this version all values showed readings with fractional digits. I have started discussion https://github.com/Koenkk/zigbee2mqtt/discussions/11323#discussion-3865347, evaluating whether to open an issue. 

Have you noticed the same change with the latest zigbee2mqtt version, losing the digits after the decimal point?

janbanan007, 2022-02-13:
@mikke49 
Phase A is a float, as it has digits for almost all readings.
Phase B is constant 6554, phase.
Phase C is int.
Exactly the same as your observation, thanks for biting this out.

I have only measurements for 3 days, as I just added the Develco HAN adapter to my smarthome installation

janbanan007, 2022-03-04:
After using a zigbee sniffer and analyzed data transmitted, I have verified that value of 65535 is sent as Current Phase B.
This is not a issue to be fixed by zigbee2mqtt, value from HAN port on energy meter is void.

closed 2022-03-04 by janbanan007