Koenkk/zigbee2mqtt #11381: Unable to turn off the LED backlight on Aurora AOne Dimmer module

proffalken, 2022-02-12:
### What happened?

Aurora support have advised that sending an `OFF` command to Endpoint 3, Attribute "OnOff" should turn off the blue ring around the LED, however when I try this I get the following:

```
Error Publish 'set' 'write' to 'Dining main lights' failed: 'Error: Write 0x000d6ffffe7ef527/3 genOnOff({"onOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'READ_ONLY')'
```

### What did you expect to happen?

I expected the LED ring around the dimmer control to turn off

### How to reproduce it (minimal and precise)

[image]

Select the device, go to the dev console, select the Endpoint (3), the Cluster (OnOff), and set the Attribute OnOff to 0

### Zigbee2MQTT version

1.23.0 commit: 6ea61ba

### Adapter firmware version

20210708

### Adapter

Electrollama ZZH

### Debug log

```
Received MQTT message on 'zigbee2mqtt/Dining main lights/3/set' with data '{"write":{"cluster":"genOnOff","options":{},"payload":{"onOff":0}}}'
Publishing 'set' 'write' to 'Dining main lights'
```

Koenkk, 2022-02-12:
You are sending a write command, not an on/off command. Try with the following external converter, this should give you the option in the frontend -> exposes:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    aOneBacklight: {
        key: ['backlight_led'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = meta.device.getEndpoint(3);
            await endpoint.command('genOnOff', state, {});
            utils.validateValue(state, ['toggle', 'off', 'on']);
            return {state: {backlight_led: state.toUpperCase()}}
        },
    }
}

const definition = {
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module CUSTOM',
    exposes: [extend.light_onoff_brightness({noConfigure: true}).exposes, 
        exposes.binary('backlight_led', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable the blue backlight LED')],
    toZigbee: [extend.light_onoff_brightness({noConfigure: true}).toZigbee, tzLocal.aOneBacklight],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)

MJWMJW2, 2022-02-12:
Thank you for your help.

I may be tired, too stupid or some combination, I followed above, but can no longer start without removing the config.. Startup goes in to a loop:

`[21:56:24] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.23.0 start
> node index.js
Zigbee2MQTT:warn  2022-02-12 21:56:26: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-02-12 21:56:26: Allowing new devices to join.
Zigbee2MQTT:warn  2022-02-12 21:56:26: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:error 2022-02-12 21:56:27: Invalid message 'null', skipping...
Zigbee2MQTT:error 2022-02-12 21:56:28: Failed to call 'HomeAssistant' 'start' (Error: Unsupported exposes type: 'undefined'
    at HomeAssistant.exposeToConfig (/app/lib/extension/homeassistant.ts:788:19)
    at HomeAssistant.getConfigs (/app/lib/extension/homeassistant.ts:897:38)
    at HomeAssistant.discover (/app/lib/extension/homeassistant.ts:1029:14)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:105:18)
    at Controller.callExtensions (/app/lib/controller.ts:315:17)
    at Controller.start (/app/lib/controller.ts:152:9)
    at start (/app/index.js:101:5))
Error: Unsupported exposes type: 'undefined'
    at HomeAssistant.exposeToConfig (/app/lib/extension/homeassistant.ts:788:19)
    at HomeAssistant.getConfigs (/app/lib/extension/homeassistant.ts:897:38)
    at HomeAssistant.onMQTTMessage (/app/lib/extension/homeassistant.ts:1267:31)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:102:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:115:27)
    at MqttClient.emit (events.js:400:28)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1545:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:533:12)
    at work (/app/node_modules/mqtt/lib/client.js:436:12)
[21:56:28] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.23.0 start
> node index.js
Zigbee2MQTT:warn  2022-02-12 21:56:30: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-02-12 21:56:30: Allowing new devices to join.
Zigbee2MQTT:warn  2022-02-12 21:56:30: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:error 2022-02-12 21:56:31: Invalid message 'null', skipping...`

proffalken, 2022-02-13:
I disabled `homeassistant` in the config and that let me start z2m, and I can see the "Backlight on/off" option in `Exposes`, however when I click it I get this error in the logs:

```
Zigbee2MQTT:debug 2022-02-13 06:55:06: Received MQTT message on 'zigbee2mqtt/Dining main lights/set' with data '{"backlight_led":"OFF"}'
Zigbee2MQTT:error 2022-02-13 06:55:06: Error: Cannot read property 'includes' of undefined
Zigbee2MQTT:debug 2022-02-13 06:55:06: TypeError: Cannot read property 'includes' of undefined
    at /app/lib/extension/publish.ts:209:60
    at Array.find (<anonymous>)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:209:42)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:102:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:115:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27)
    at WebSocket.emit (events.js:400:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1098:20)
    at Receiver.emit (events.js:400:28)
```

I don't really know Javascript at all (I'm a python guy!), so I'm not quite sure why it's trying to include something that's undefined?

Koenkk, 2022-02-13:
Whoops, made a mistake, try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    aOneBacklight: {
        key: ['backlight_led'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = meta.device.getEndpoint(3);
            await endpoint.command('genOnOff', state, {});
            utils.validateValue(state, ['toggle', 'off', 'on']);
            return {state: {backlight_led: state.toUpperCase()}}
        },
    }
}

const definition = {
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module CUSTOM',
    exposes: [...extend.light_onoff_brightness({noConfigure: true}).exposes, 
        exposes.binary('backlight_led', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable the blue backlight LED')],
    toZigbee: [...extend.light_onoff_brightness({noConfigure: true}).toZigbee, tzLocal.aOneBacklight],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
};

module.exports = definition;
```

proffalken, 2022-02-13:
Getting closer:

```
Error Publish 'set' 'backlight_led' to 'Dining main lights' failed: 'ReferenceError: state is not defined'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'backlight_led' to 'Dining main lights' failed: 'ReferenceError: state is not defined'","meta":{"friendly_name":"Dining main lights"},"type":"zigbee_publish_error"}'
Debug Received MQTT message on 'zigbee2mqtt/Dining main lights/set' with data '{"backlight_led":"OFF"}'
```

Koenkk, 2022-02-13:
Another mistake, try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    aOneBacklight: {
        key: ['backlight_led'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = meta.device.getEndpoint(3);
            await endpoint.command('genOnOff', value.toLowerCase(), {});
            utils.validateValue(state, ['toggle', 'off', 'on']);
            return {state: {backlight_led: state.toUpperCase()}}
        },
    }
}

const definition = {
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module CUSTOM',
    exposes: [...extend.light_onoff_brightness({noConfigure: true}).exposes, 
        exposes.binary('backlight_led', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable the blue backlight LED')],
    toZigbee: [...extend.light_onoff_brightness({noConfigure: true}).toZigbee, tzLocal.aOneBacklight],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
};

module.exports = definition;
```

proffalken, 2022-02-13:
:tada: The backlight is now controllable, thanks for that.

I do still get an error though:

```
 Received MQTT message on 'zigbee2mqtt/Dining main lights/set' with data '{"backlight_led":"OFF"}'
Debug Publishing 'set' 'backlight_led' to 'Dining main lights'
Error Publish 'set' 'backlight_led' to 'Dining main lights' failed: 'ReferenceError: utils is not defined'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'backlight_led' to 'Dining main lights' failed: 'ReferenceError: utils is not defined'","meta":{"friendly_name":"Dining main lights"},"type":"zigbee_publish_error"}'
```

Given that it's working, if there's any way this can be rolled in to the main code and added as a control that can be exposed to Home Assistant that would be amazing!

MJWMJW2, 2022-02-13:
Huge thanks @Koenkk - confirm same behaviour as @prffalken, backlight toggle works, with same error, expanded debug:

'''
Debug Publishing 'set' 'backlight_led' to 'Office Top Light'
Error Publish 'set' 'backlight_led' to 'Office Top Light' failed: 'ReferenceError: utils is not defined'
Debug ReferenceError: utils is not defined at Object.convertSet (/app/dist/util/externally-loaded.js:15:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'backlight_led' to 'Office Top Light' failed: 'ReferenceError: utils is not defined'","meta":{"friendly_name":"Office Top Light"},"type":"zigbee_publish_error"}'
'''

MJWMJW2, 2022-02-14:
It is still the same,
Publish 'set' 'backlight_led' to 'Office Top Light' failed: 'ReferenceError: utils is not defined'

Koenkk, 2022-02-14:
And with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    aOneBacklight: {
        key: ['backlight_led'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = meta.device.getEndpoint(3);
            await endpoint.command('genOnOff', value.toLowerCase(), {});
            utils.validateValue(state, ['toggle', 'off', 'on']);
            return {state: {backlight_led: state.toUpperCase()}}
        },
    }
}

const definition = {
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module CUSTOM',
    exposes: [...extend.light_onoff_brightness({noConfigure: true}).exposes, 
        exposes.binary('backlight_led', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable the blue backlight LED')],
    toZigbee: [...extend.light_onoff_brightness({noConfigure: true}).toZigbee, tzLocal.aOneBacklight],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
};

module.exports = definition;
```

MJWMJW2, 2022-02-14:
slightly different: 
Publish 'set' 'backlight_led' to 'Office Top Light' failed: 'ReferenceError: state is not defined'

Koenkk, 2022-02-14:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    aOneBacklight: {
        key: ['backlight_led'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = meta.device.getEndpoint(3);
            await endpoint.command('genOnOff', value.toLowerCase(), {});
            utils.validateValue(state, ['toggle', 'off', 'on']);
            return {state: {backlight_led: value.toUpperCase()}}
        },
    }
}

const definition = {
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module CUSTOM',
    exposes: [...extend.light_onoff_brightness({noConfigure: true}).exposes, 
        exposes.binary('backlight_led', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable the blue backlight LED')],
    toZigbee: [...extend.light_onoff_brightness({noConfigure: true}).toZigbee, tzLocal.aOneBacklight],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
};

module.exports = definition;
```

Koenkk, 2022-02-14:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    aOneBacklight: {
        key: ['backlight_led'],
        convertSet: async (entity, key, value, meta) => {
            const state = value.toLowerCase();
            utils.validateValue(state, ['toggle', 'off', 'on']);
            const endpoint = meta.device.getEndpoint(3);
            await endpoint.command('genOnOff', state, {});
            return {state: {backlight_led: state.toUpperCase()}}
        },
    }
}

const definition = {
    zigbeeModel: ['WallDimmerMaster'],
    model: 'AU-A1ZB2WDM',
    vendor: 'Aurora Lighting',
    description: 'AOne 250W smart rotary dimmer module CUSTOM',
    exposes: [...extend.light_onoff_brightness({noConfigure: true}).exposes, 
        exposes.binary('backlight_led', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable or disable the blue backlight LED')],
    toZigbee: [...extend.light_onoff_brightness({noConfigure: true}).toZigbee, tzLocal.aOneBacklight],
    extend: extend.light_onoff_brightness({noConfigure: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
    },
};

module.exports = definition;
```

MJWMJW2, 2022-02-14:
Perfect!  Working.

Koenkk, 2022-02-15:
Awesome, added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koenkk, 2022-03-02:
@karlc28 added, could you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

karlc28, 2022-03-02:
@Koenkk  what exactly would I need to do Im already using the edge version in HASS add on and have an external converter folder created

karlc28, 2022-03-02:
[image]

karlc28, 2022-03-02:
after a full reboot z2m wont load and the log shows this

Error: Cannot find module '../lib/exposes'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:101:18)
    at /app/dist/util/externally-loaded.js:1:17
    at Script.runInContext (vm.js:144:12)
    at Script.runInNewContext (vm.js:149:17)
    at Object.runInNewContext (vm.js:303:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:2998) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2998) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[21:52:12] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.24.0-dev start
> node index.js
(node:3095) UnhandledPromiseRejectionWarning: internal/modules/cjs/loader.js:905
  throw err;
  ^

MJWMJW2, 2022-03-03:
I can’t be sure, however I have a feeling that I read that the endpoint for the socket’s switch back light is different to the dimmer’s

karlc28, 2022-03-03:
Aurora said endpoint 1 I believe

karlc28, 2022-03-03:
@Koenkk  the exposure is now there but fails  

log:

Debug Received MQTT message on 'zigbee2mqtt/Front Room Double Socket/set' with data '{"backlight_led":"OFF"}'
Debug Publishing 'set' 'backlight_led' to 'Front Room Double Socket'
Error Publish 'set' 'backlight_led' to 'Front Room Double Socket' failed: 'TypeError: Cannot read property 'command' of undefined'
Debug TypeError: Cannot read property 'command' of undefined at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/devices/aurora_lighting.js:17:28) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:122:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27) at WebSocket.emit (events.js:400:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1137:20) at Receiver.emit (events.js:400:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:528:14)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'backlight_led' to 'Front Room Double Socket' failed: 'TypeError: Cannot read property 'command' of undefined'","meta":{"friendly_name":"Front Room Double Socket"},"type":"zigbee_publish_error"}'

karlc28, 2022-03-03:
[image]

karlc28, 2022-03-03:
[image]

karlc28, 2022-03-03:
[image]

karlc28, 2022-03-04:
@Koenkk not sure if anything has been changed but it's still showing this when trying to turn the back light off
[image]

Koenkk, 2022-03-05:
@karlc28 can you provide the data/database.db entry of this device?

karlc28, 2022-03-05:
@Koenkk 
{"id":13,"type":"Router","ieeeAddr":"0x00158d000256f882","nwkAddr":24356,"manufId":4636,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"DoubleSocket50AU","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[6,5,4,0,3,2820,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"DoubleSocket50AU","manufacturerName":"Aurora","powerSource":1,"zclVersion":1,"appVersion":20,"stackVersion":2,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":3}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"nearestGateway":{"attributes":{}}},"binds":[{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":242},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":1},{"cluster":7,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[6,5,4,0,3,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":2}}},"binds":[{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":242},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d4023","endpointID":242}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":20,"stackVersion":2,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1646471088378,"defaultSendRequestWhen":"immediate"}

karlc28, 2022-03-05:
I think that's what you are after

Koenkk, 2022-03-05:
@karlc28 yes, pushed a possible fix, please check

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

karlc28, 2022-03-05:
@Koenkk  Still getting this error pal
[image]

karlc28, 2022-03-05:
[image]

karlc28, 2022-03-05:
[image]

karlc28, 2022-03-05:
[image]

Koenkk, 2022-03-06:
Did you update to the latest edge by uninstalling it and reinstalling it?

karlc28, 2022-03-06:
> Did you update to the latest edge by uninstalling it and reinstalling it?

Yes mate

karlc28, 2022-03-08:
@Koenkk ive tried again today to see if there's any change and still the same

Koenkk, 2022-03-08:
@karlc28 found the issue, please try again

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

karlc28, 2022-03-08:
@Koenkk i just tried then and still the same but not sure if the dev branch has changed yet I'll try again in an hour or so

karlc28, 2022-03-08:
@Koenkk so it's now doing something but it turns on/off the power to the right socket not the light

Koenkk, 2022-03-10:
@karlc28 are you sure your socket supports this functionallity? The difference with yours is that it doesn't have ep 3 (only 1 and 2), endpoint 1 and 2 are used to control the outlets itself (left/right) so I'm not sure what endpoint to use.

karlc28, 2022-03-10:
@Koenkk yes if I connect it to my echo 4th gen it gives the option to dim the light with a slider and if it's connected to an aurora hub it has the same feature I spoke with Aurora support and they said it was endpoint 1 [image]

karlc28, 2022-03-10:
@Koenkk 
IT WORKED hallelujah 😂 thank you so much mate