Koenkk/zigbee2mqtt #11385: No reporting from QBKG34LM

keniji, 2022-02-12:
### What happened?

No reporting from a device (model: QBKG34LM), attributes like power, energy, temperature, action are shown "N/A".

### What did you expect to happen?

When I click on the device, I should see a report from the frontend(log).

### How to reproduce it (minimal and precise)

Not sure why/how this happen. The device used to work fine and suddenly I can not see any report from it after I re-paired it mistakenly. It looks like the device is "dead", but I am still able to control the device on the frontend, like click on the button, but when I press a button to switch it "ON", frontend still shows "OFF". I can not see any report from it in log (except some about I control it on frontend) so the device attributes like power, energy, temperature, action are shown "N/A" forever:
[image]

I am not sure if there is any problem with the device, but I've tried to pair it to Aqara E1 gateway and it seems work fine. Does anyone has any idea about what I can try to fix this problem? Thanks in advance!

### Zigbee2MQTT version

1.23.0 commit: 6ea61ba

### Adapter firmware version

20210319

### Adapter

CC2652P LAN adapter

### Debug log

NA

z-x-f, 2022-02-12:
Is it the latest version of the firmware?

keniji, 2022-02-12:
> Is it the latest version of the firmware?

Do you mean QBKG34LM's firmware? It's not up to date when it worked fine before, and I've updated its firmware after I got this issue, but nothing changes.

z-x-f, 2022-02-12:
Try removing the device and pairing the device again.
Maybe you need to update the firmware of your Adapter.

keniji, 2022-03-12:
> Try removing the device and pairing the device again. Maybe you need to update the firmware of your Adapter.

Yes! I update the firmware today and it finally works! It's really strange because the device works fine with the previous firmware and this is the reason I was trying to find a way besides updating the firmware... Anyway, thanks for your help again!

closed 2022-03-12 by keniji