Koenkk/zigbee2mqtt #11401: Noticed two spellings of same expose: power_on_behavior and power_on_behaviour

flatsiedatsie, 2022-02-13:
### What happened?

Just looking through the device database I noticed that there are two spellings of the same expose property. I don't know if it's an issue, but it might be nice to streamline on one language?

Most devices use the American spelling. Only three use the Brittish spelling:
https://www.zigbee2mqtt.io/supported-devices/#e=power_on_behaviour

These were added by @danieledwardgeorgehitchcock

Is this something to streamline? I could create a pull request.

### What did you expect to happen?

I was hoping that a search in the device database would give all devices power this option in one search

### How to reproduce it (minimal and precise)

Use the search feature of the website.

### Zigbee2MQTT version

not relevant

### Adapter firmware version

not relevant

### Adapter

not relevant

sjorge, 2022-02-13:
Given most other spelling is American English (e.g. color, ...) it's probably something that should indeed updated to reflect the American English spelling.

cc @Koenkk there was some discussion about American vs British English for the frontend and most people there wan ted British English, while z2m itself is mostly American English... do we want to have both of these aligned to British English?

It's kind of tricky as some of that effects attribute naming.

Koenkk, 2022-02-13:
If most is already in American English we should keep it such (to prevent changes as much as possible). Since power_on_behavior/power_on_behaviour is typically a one time setting it is fine to change it

closed 2022-02-13 by Koenkk