Koenkk/zigbee2mqtt #11406: Xiaomi Honeywell gas detector JTQJ-BF-01LM/BW reset and repair?

jrhbcn, 2022-02-13:
### What happened?

It is possible to reset the honeywell gas detector to repair it to the network? I noticed it was not working properly (several days without reporting status) so I force remove it from the network using the zigbee2mqtt remove tool (i did not check "Block from joining again"). However, now i cannot pair it again, i have tried to press 3 times the button as stated in the instructions (and that worked the first time) but now it does not seem to start the pairing process. 

Is there a way to reset the device to factory setting so I can repair it? Thanks,

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20201026

### Adapter

zzh

closed 2022-03-23 by github-actions[bot]

nediam, 2022-06-22:
Hey @jrhbcn, I have similar problem, did you manage to make it working again?

jrhbcn, 2022-06-22:
No @nediam.

It is a paperweight at the moment for me. I have been unable to find how to reset the thing. :(