Koenkk/zigbee2mqtt #11411: Timout while trying to command NodOn SIN-4-RS-20 roller shutter

PEGARifrouf, 2022-02-13:
### What happened?

NodOn Roller Shutter module do not take into account commands (timeout is reported) when OPEN, STOP or CLOSE command is requested.
Manual commands are working from the wired buttons (wired to the Nodon module), position opened and closed are seen within HA entity when commanded manually with historical wall buttons

### What did you expect to happen?

Roller shutter shall OPEN, STOP or CLOSE following the command set on HA or Z2M dahsboard

### How to reproduce it (minimal and precise)

SIN-4-RS-20 wired, has joined the zigbee network and is calibrated
Click to OPEN on Z2M dashboard of the device
No mouvement from the roller shutter and timeout error recieved 10s later

### Zigbee2MQTT version

1.23.0-1

### Adapter firmware version

20210319

### Adapter

CC2652P

### Debug log

debug 2022-02-12 18:08:46: Received MQTT message on 'zigbee2mqtt/VR_SDB/set' with data '{"state":"CLOSE"}'
debug 2022-02-12 18:08:46: Publishing 'set' 'state' to 'VR_SDB'
debug 2022-02-12 18:09:02: Received Zigbee message from 'Carillon', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":257,"imageType":0,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2022-02-12 18:09:02: Device 'Carillon' requested OTA
debug 2022-02-12 18:09:03: Received Zigbee message from 'Carillon', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65543,"imageType":1,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2022-02-12 18:09:03: Device 'Carillon' requested OTA
error 2022-02-12 18:09:09: Publish 'set' 'state' to 'VR_SDB' failed: 'Error: Command 0x9035eafffec88e03/242 closuresWindowCovering.downClose({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22632 - 242 - 58 - 258 - 11 after 10000ms)'
debug 2022-02-12 18:09:09: Error: Command 0x9035eafffec88e03/242 closuresWindowCovering.downClose({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22632 - 242 - 58 - 258 - 11 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
info  2022-02-12 18:09:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'VR_SDB' failed: 'Error: Command 0x9035eafffec88e03/242 closuresWindowCovering.downClose({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 22632 - 242 - 58 - 258 - 11 after 10000ms)'","meta":{"friendly_name":"VR_SDB"},"type":"zigbee_publish_error"}'
debug 2022-02-12 18:09:33: Saving state to file /config/zigbee2mqtt/state.json

closed 2022-03-26 by github-actions[bot]

Jassevanloe, 2022-07-03:
I had the same problem as described above. 

The solution in my case:
Changing the zigbee channel to 11
(Before i was using 12) 

It now works flawless.

AlexH269, 2022-11-09:
I am facing this exact problem, but my network is already on channel 11 :/

PEGARifrouf, 2022-11-15:
For info I finally switched to legrand wired shutter switch, works fine. Partial positions must be calculated by timing the movement.
https://www.zigbee2mqtt.io/devices/067776.html

wucke13, 2023-09-13:
I'm on channel 11, and I do have the same issue.

bobrossthepainter, 2023-11-03:
Same issue for me as well. I'm as well on channel 11 and using Sonoff Dongle P. Anybody got a solution for this issue?