Koenkk/zigbee2mqtt #11418: Trying to add a CC2531 as router to the network returns unsupported error

urkoin, 2022-02-14:
### What happened?

I flashed an old CC2531 with the router firmware. All went fine. Trying to add it to the network, it is not working. 
I got the following error in zigbee2mqtt logs:

**Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'**

My coordinator is a SONOFF ZIGBEE 3.0 USB DONGLE-P.
I have other SONOFF ZIGBEE 3.0 USB DONGLE-P  with router firmware working fine.

### What did you expect to happen?

The CC2531 with the router firmware is added correctly to the network

### Zigbee2MQTT version

1.21.1-dev

### Adapter firmware version

20211217

### Adapter

SONOFF ZIGBEE 3.0 USB DONGLE-P as coordinator and CC2531 as router

deepcoder, 2022-02-14:
I've only had it running about 24 hours, however I am successful with the setup below. The CC2531 router below is successfully routing between a battery powered CC2530 device and the SONOFF 3.0 dongle. I've been sending test control functions to the CC2530 at a rate of about one message every 0.25 seconds and it seems to be staying connected and in sync pretty well. Good hunting!

I have a zigbee2mqtt 1.23.0 docker running with the Sonoff Zigbee 3.0 dongle flashed with this firmware:

CC1352P2_CC2652P_launchpad_coordinator_20211217.hex

I just flashed one of these: 
Acxico 1Pcs USB CC2531 Sniffer Board Bluetooth 4.0 Wireless Zigbee Analyzer Module with External Antenna
https://www.amazon.com/gp/product/B081WRGL31/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

with this firmware:
CC2531 based router (firmware)
https://ptvo.info/cc2531-based-router-firmware-136/
[image]

urkoin, 2022-02-16:
> 

Tried with the same firmware for the CC2531 but not luck. I get the same error on the zigbee2mqtt logs....

deepcoder, 2022-02-16:
Has it ever worked in a zigbee2mqtt environment, with any firmware? You might want to review a couple of ptvo's blog posts on cc25xx hardware, there do appear to be some non-standard configurations in 'sheeps clothing'  : 
https://ptvo.info/how-to-select-and-flash-cc2530-144/

urkoin, 2022-02-16:
> Has it ever worked in a zigbee2mqtt environment, with any firmware? You might want to review a couple of ptvo's blog posts on cc25xx hardware, there do appear to be some non-standard configurations in 'sheeps clothing' :
> https://ptvo.info/how-to-select-and-flash-cc2530-144/
> My little test rig continues to route between the with the cc2531 routing between the cc2530 and the sonoff 3.0 dongle non-stop. I'm doing some simple battery life tests, so I am sending messages non-stop via the router successfully. You might see if you can upgrade the version of zigbee2mqtt and sonoff dongle firmware.
> [image]

The cc2531 was working fine as coordinator. But after flashing the router firmware it never worked as router. Now I have the sonoff as coordinator and want the cc2531 to work as router, but it seems zigbee2mqtt is not accepting it as router…

closed 2022-03-26 by github-actions[bot]

Jackfritt, 2024-04-05:
I only got one of my 2 CC2531 with external Antenna running with 
git clone https://github.com/Koenkk/Z-Stack-firmware
unzip Z-Stack-firmware/router/Z-Stack_Home_1.2/bin/CC2531_router_2020_09_29.zip
md5sum *.hex
265664a30a1d0c912fe374aadb5d581a  CC2531ZNP-Prod.hex
f7d8374a806672b10d3676787999456a  router-cc2531-diag.hex
c67ee2a4c84c14f5a67d3b80b76f6a85  router-cc2531-diag-usb.hex
4786f1b287a7846951c8f106ee8f3cf0  router-cc2531-std.hex
(Same as above firmware from deepcoder....)

sudo cc-tool/cc-tool -e -w router-cc2531-diag-usb.hex
CC2531_SOURCE_ROUTING_20211116 and FW router-cc2531-diag-usb.hex

I cant get the second one to connect to Zigbee2MQTT
1.36.1-dev commit: [3e913495](https://github.com/Koenkk/zigbee2mqtt/commit/3e913495)

After changing the coordinator to a Sonoff they connect like a charm. 
Now I have to investigate which firmrware the old CC2531 coordinator without external antenna had...
Maybe just this was the problem...