Koenkk/zigbee2mqtt #11419: [New device support]: ROBB Smarrt Zigbee thermostat

Tenkc, 2022-02-14:
### Link

https://www.robbshop.nl/robb-smarrt-zigbee-thermostaat

### Database entry

{"id":44,"type":"Router","ieeeAddr":"0xffff030201030259","nwkAddr":31103,"manufId":4644,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"ZG9092","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,9,513,1794,2820],"outClusterList":[3,10,1030],"clusters":{" genBasic":{"attributes":{"modelId":"ZG9092","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"appVersion":7,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"6.9.1.0_r4"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":7,"inClusterList":[0,3],"outClusterList":[ 3,25],"clusters":{"genBasic":{"attributes":{"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":7,"stackVersion":0,"hwVersion":1,"dateCode":"NULL" ,"swBuildId":"6.9.1.0_r4","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1644845741592,"defaultSendRequestWhen":"immediate"}

### Comments

Pairing with Z2M was no problem, but I already got stuck at step 2: Adding your device. 

Not sure what add in the file or how to proceed, as this is a bit above my knowledge level...

Tenkc, 2022-03-17:
I've been tinkering a bit (and learning a lot). The basic converter looks as follows and is working in HA.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZG9092'],
    model: 'ROB_200-040-0',
    vendor: 'Robb Smarrt',
    description: 'Robb Heating Thermostat',
    fromZigbee: [fz.thermostat],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_running_state, tz.thermostat_system_mode],
    exposes: [
        exposes.climate().withSetpoint('occupied_heating_setpoint', 0.0, 30, 0.1)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat'])
            .withRunningState(['idle', 'heat'])
        ],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(3) || device.getEndpoint(1);
            const binds = ['genBasic', 'genIdentify', 'genPowerCfg', 'genTime', 'hvacThermostat', 'hvacUserInterfaceCfg'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.thermostatSystemMode(endpoint);
            await reporting.thermostatTemperature(endpoint);
            await reporting.thermostatRunningState(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);

        },
};

module.exports = definition;`

Tenkc, 2022-04-04:
It appears that this thermostat is a white label of the Namron thermostat which is being supported now: https://www.zigbee2mqtt.io/devices/4512737.html

Dragonheart8567, 2022-04-15:
Well @Tenkc  unfortunately it doesn't

Tenkc, 2022-04-20:
@Dragonheart8567 indeed, I'm running into some errors regarding the child_lock, and the energy (sum of consumed energy) readout stays null or N/A. Power, current and voltage do give a reading.

The error is: 

`Failed to configure 'Keuken vloerverwarming', attempt 4 (Error: Read 0xffff030201030259/1 seMetering(["multiplier","divisor"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.readMeteringMultiplierDivisor (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:29:5) at Object.configure (/app/dist/util/externally-loaded.js:56:13) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
`

When I use the dev console I can get a reading for currentSummDelivered:

Read result of 'seMetering': {"currentSummDelivered":[0,439]}

owangen, 2022-04-23:
Found out my problem with writing zigbee to the device, it was caused by a bug caused by me. Everything working fine now. Will soon create a PR with full support of the Namron Thermostat, should be copy and paste to get the ROBB thermostat supported.

owangen, 2022-04-24:
I created a PR for supporting the Namron thermostat https://github.com/Koenkk/zigbee-herdsman-converters/pull/4154 so use that code to implement support for the ROBB thermostat. Both stems from the sunricher thermostat, but seems like the ROBB is whitelabel of Sunricher, and Namron thermostat is private label of the sunricher. Namron thermostat reports own make and modelid (but reported mfgcode is sunricher) . I guess you can reuse alle the to and from zigbee converters. The custom from and to zigbee are prefixed with namron, but if they works for ROBB and sunricher they should be renamed to sunricher_

Tenkc, 2022-04-25:
Nice work, and thanks!

closed 2022-06-03 by github-actions[bot]