Koenkk/zigbee2mqtt #11422: Nyce motion sensor NCZ-3043-HA bateery level incorrect, and temperature/humidity not updating

35gh, 2022-02-14:
### What happened?

The [Nyce motion sensor NCZ-3043-HA](https://www.zigbee2mqtt.io/devices/NCZ-3043-HA.html#nyce-ncz-3043-ha) is properly detected by v1.23.0. There are however several problems:

1. Battery level is incorrectly reported (142% in the example below)
```
{
    "battery": 142,
    "battery_low": false,
    "humidity": 44,
    "linkquality": 255,
    "occupancy": false,
    "tamper": false,
    "temperature": 22.9,
    "voltage": 2700
}
```

2. Temperature and humidity are not updating. However they do when the sensor is manually queried.
Before query: see above
After manual query:
`Read result of 'msTemperatureMeasurement': {"measuredValue":2220}`
`Read result of 'msRelativeHumidity': {"measuredValue":3800}`

```
{
    "battery": 142,
    "battery_low": false,
    "humidity": 38,
    "linkquality": 255,
    "occupancy": false,
    "tamper": false,
    "temperature": 22.2,
    "voltage": 2700
}
```

3. Picture is wrong
Current picture is from wall sensor [NCZ-3041-HA](https://www.nycesensors.com/product/motion-sensor)
[image]

while the [NCZ-3043-HA](https://www.nycesensors.com/product/ceiling-motion-sensor) is ceiling mounted:
[image]

Note: motion detection works properly
Note 2: battery, humidity and temperature are correctly reported and updating when using DeconZ

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26660700

### Adapter

Conbee II

35gh, 2022-02-14:
Thanks! Battery seems OK now, but temperature/humidity (and I believe tamper as well) are still not updating.

```
{
    "battery": 70,
    "battery_low": false,
    "humidity": 41,
    "linkquality": 255,
    "occupancy": false,
    "tamper": false,
    "temperature": 23.3,
    "voltage": 2700
}
```

Tried to remove and re-add device as well.

Note: I see the following error in logs:  Failed to configure 'Sensor name', attempt 3 (ReferenceError: overides is not defined at Object.configure (/opt/zigbee2mqtt/dist/util/externally-loaded.js:22:47) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13))

35gh, 2022-02-15:
Update 2: now temperature and humidity are updating about every 2 hours.

Koenkk, 2022-02-15:
Made a mistake in the external converter which gives you the failed to configure error, try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['3043'],
    model: 'NCZ-3043-HA',
    vendor: 'Nyce',
    description: 'Ceiling motion sensor CUSTOM',
    fromZigbee: [fz.occupancy, fz.humidity, fz.temperature, fz.ignore_basic_report, fz.ignore_genIdentify, fz.ignore_poll_ctrl,
        fz.battery, fz.ignore_iaszone_report, fz.ias_occupancy_alarm_2],
    toZigbee: [],
    exposes: [e.occupancy(), e.humidity(), e.temperature(), e.battery(), e.battery_low(), e.tamper()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msTemperatureMeasurement', 'msRelativeHumidity']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.temperature(endpoint);
        await reporting.humidity(endpoint);
    },
};

module.exports = definition;
```

After this configure the device via the z2m frontend -> device page -> yellow refresh button (make sure to wakeup the device right before clicking the button)

35gh, 2022-02-15:
Thank you! Device configuration works now, but temperature/humidity still don't update until I query the clusters manually.

Koenkk, 2022-02-15:
So it says configured successfully? Can you wait a few more hours and see if it works after that?

35gh, 2022-02-15:
Sensor still updates temperature/humidity every 2 hours only...
[image]
.

Koenkk, 2022-02-16:
Then I expect this sensor does not support change the report interval.

35gh, 2022-02-16:
Well my point is that this sensor was working perfectly when using DeconZ (reporting temp/humifidty as they change), while with Zigbee2mqtt I only get updates every 2 hours.

Koenkk, 2022-02-16:
Interesting, could you play with the configure reporting option in the frontend? In the frontend -> click on your device -> reporting, try e.g. setting the min rep change to 0 and min rep interval to 1. Before pressing apply make sure to wakeup the device.

35gh, 2022-02-17:
It works thanks! I used Min rep interval=20 for humidity, and Min rep interval=60 for temperature, Min rep change=0 for both.

This sensor has a 0.1°C precision for temperature. Is it possible to report temperature changes every 0.5°C? Min rep change does not seem to have any effect.

[image]

closed 2022-03-29 by github-actions[bot]