Koenkk/zigbee2mqtt #11423: New Firmware for Ubisys D1

Danit2, 2022-02-14:
### What happened?

There is a new Firmware Online for the Ubisys D1 but Z2M dose not shwo the Firmware. 
https://www.ubisys.de/en/support/firmware/
https://www.ubisys.de/wp-content/uploads/10F2-7B21-0000-0006-0193020E-spo-fmd.ota.zigbee

The newest Firmware Online is 1.9.3 and in Z2M is only 1.9.2.

Can this be Updated?

Thanks. 

### What did you expect to happen?

The new Firmware in Z2M

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

-

Koenkk, 2022-02-15:
@zivillian could you check this?

zivillian, 2022-02-15:
@Danit2 have you already updated to firmware version 1.9.2? The [image type was changed in 1.9.2](https://www.ubisys.de/en/support/firmware/changelog-d1/#page%231-2-0) so 1.9.3 is not a valid update for D1 devices with older firmware. Version 1.9.3 should automatically be available after updating to 1.9.2.

When I check for new firmwares, both images are listed in the debug output:
>  OTA ubisys: got firmware page, status: 200, data.length: 80716
 OTA ubisys: image found: /wp-content/uploads/10F2-7B29-0000-0004-01930221-spo-fmi4.ota.zigbee
 OTA ubisys: image found: **/wp-content/uploads/10F2-7B21-0000-0006-0193020E-spo-fmd.ota.zigbee**
 OTA ubisys: image found: /wp-content/uploads/10F2-7B28-0000-0004-0194020E-spo-rmd.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B04-0000-0007-0191020D-spo-fmsh.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B07-0000-0004-0191020D-spo-rmsh.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B2A-0000-0005-02000230-m7b-r0.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B02-0002-0007-0192020D-spo-fms.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B05-0000-0004-0191020D-spo-rms.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B03-0000-0006-0191020D-spo-fms2.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B06-0000-0004-0191020D-spo-rms2.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B0B-0000-0001-01900210-m7b-h10.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B09-0000-0004-0192020D-spo-fmi4.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B09-0000-0004-01090206-spo-fmi4.ota.zigbee
 OTA ubisys: image found: **/wp-content/uploads/10F2-7B01-0000-0006-0192020D-spo-fmd.ota1.zigbee**
 OTA ubisys: image found: /wp-content/uploads/10F2-7B01-0000-0006-01110206-spo-fmd.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B08-0000-0004-0192020D-spo-rmd.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B08-0000-0004-01110206-spo-rmd.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B04-0000-0007-01160206-spo-fmsh.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B07-0000-0004-01160206-spo-rmsh.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B0A-0000-0005-0193020D-m7b-r0.ota1.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B0A-0000-0005-010A0206-m7b-r0.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B02-0000-0001-010F0206-spo-fms.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B02-0002-0007-010F0206-spo-fms.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B05-0000-0004-010F0206-spo-rms.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B03-0000-0006-010E0206-spo-fms2.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B06-0000-0004-010E0206-spo-rms2.ota.zigbee
 OTA ubisys: image found: /wp-content/uploads/10F2-7B0B-0000-0001-01000206-m7b-h10.ota.zigbee

Danit2, 2022-02-15:
Ahhh.... 
yes i see. 
Now i have the new update. Sorry for that.

closed 2022-02-15 by Danit2