Koenkk/zigbee2mqtt #11430: MQTT failed to connect: Connection refused: Not authorized

henrikrox, 2022-02-15:
### What happened?

After updating dev edge today I get 

Zigbee2MQTT:error 2022-01-31 12:51:56: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-01-31 12:51:56: Exiting...

I havnt changed anything with the mosquito user, so I don't know why I get this error.

### What did you expect to happen?

For z2m to work

### How to reproduce it (minimal and precise)

Update z2m edge

### Zigbee2MQTT version

edge latest

### Adapter firmware version

N/A

### Adapter

deconz

henrikrox, 2022-02-15:
Tried making new users, same issue, havnt touched my Mqtt config, just updated z2m

closed 2022-03-26 by github-actions[bot]

jjy9331, 2025-02-23:
I create an ID and password in MQTT
User and password were added in Z2M configuration. So the permission problem disappeared. 

Please refer to the documentation for information on how to configure the username and password you want to add.