Koenkk/zigbee2mqtt #11431: Zlinky not reading cluster attributes (UNSUPPORTED_ATTRIBUTE)

vk496, 2022-02-15:
### What happened?

Hello,

**UNSUPPORTED_ATTRIBUTE** is a message that come from the zigbee device itself when you try to read something that is somehow wrong (AFAIK).

For some reason, attributes from a custom cluster (lixee, `0xFF66`) stopped working for me from one day to other. Trying to read or suscribe, I get always a message like this from Z2M:

```
Read 0xDEVICE/1 liXeePrivate(["currentTarif"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

At the begining I thought that is directly related with the firmware of the device and opened a issue there (https://github.com/fairecasoimeme/Zlinky_TIC/issues/46). However, after closely inspecting, I noticed that reading that attribute using the deCONZ works correctly:

[image]

That's the attribute `0xFF66/0x0000` (currentTarif) that doesn't work in Z2M and does in deCONZ.

Moreover, the attribute `0xFF66/0x0217`(statusRegister) works in Z2M. So there is something weird that I don't understand completely

### Zigbee2MQTT version

dd2fef5 (1.23.0)

### Adapter firmware version

0x26580700

### Adapter

ConBee2

vk496, 2022-02-15:
@Koenkk should I open this issue in zigbee-herdsman maybe?

vk496, 2022-02-15:
More info about it. This is what **deCONZ** send to read `0xFF66/0x0000`:

Request:
[image]
Response:
[image]

Compared to what Z2M does:
Request:
[image]
Response:
[image]

Koenkk, 2022-02-15:
Z2M send the manufacturer code while Deconz doesn't. Try to disable this with `            await endpoint.read('liXeePrivate', ['currentTarif'], {manufacturerCode: null});
`

closed 2022-02-15 by vk496