Koenkk/zigbee2mqtt #11432: [New device support]: TS0601 Temperature Humidity Sensor '_TZE_bjawzodf'

maxfok, 2022-02-15:
### Link

https://www.aliexpress.com/item/1005003501561951.html?spm=a2g0o.9042311.0.0.605f4c4dSpWLqB

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0xa4c13832afaa50fd","nwkAddr":43513,"manufId":4417,"manufName":"_TZE200_bjawzodf","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0019��)f\u001a��)\u0012","65506":49,"65508":0,"stackVersion":0,"dateCode":"","appVersion":67}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1644934282108,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to connect to Zigbee module CC2531 but I didn't manage to connect it. I get unsupported device message.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');

const definition = {
      fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'}],
      model: 'TS0601_temperature_humidity_sensor',
      vendor: 'TuYa',
      description: 'Temperature and humidity sensor with display',
      fromZigbee: [fz.tuya_temperature_humidity_sensor],
      toZigbee: [],
      exposes: [e.temperature(), e.humidity(), e.battery()],
};

module.exports = definition;
```

jenko1080, 2022-02-26:
Started having a look at adding some code and this was the only hit on Google, is there any reason for this to be closed?

Either way, what I've got so far:
```
# <SimpleDescriptor endpoint=1, profile=260, device_type=81
# device_version=1
# input_clusters=[4, 5, 61184, 0]
# output_clusters=[25, 10]>
```

Haven't got very far on the messages it's sending on `61184`, most of the attempts to make this work have been guess / check.

Message: `19830100a40102000400000118`, 99% sure that this is 28.0 degrees
Message: `19840100a5020200040000022c`, 99% sure that this is 55.6 humidity
Message: `19850100a60402000400000064`, likely battery, 100%

After this point I ran out of time for the session to troubleshoot, using the base class of `TuyaManufClusterAttributes` I was expecting `_update_attribute()` to be called... but it didn't seem to be.

Adding an `__init__` function with a `_LOGGER.warning` shows that it's at least using the class.

Hope that helps someone if I don't get back on to this soonish.

maxfok, 2022-02-26:
Sorry I closed the issue, but I did so because no-one seemed to have referred to it before, so I thought it was me the problem. I managed to get it work, under CC2531 and still running under a conbee ii as I upgraded my adapter due to weak signal issues. The only mistake is that I don't get a decimal number on humidity, so it looks like 436% (lol) instead of of 43.6% but since I don't much of coding decided to let it as it is. Issue is opened again for any further configuration. Thanks and excuse me.

andre68723, 2022-03-05:
@maxfok https://github.com/Koenkk/zigbee-herdsman-converters/pull/3957 was merged, please check next release of zigbee2mqtt

closed 2022-04-12 by github-actions[bot]