Koenkk/zigbee2mqtt #11436: Is Zigbee2MQTT seriously this buggy, unusable, and broken all the time?

mortenmoulder, 2022-02-15:
### What happened?

I'm not here to bash Koenkk or any of the developers on Zigbee2MQTT. What you guys have developed is huge, but I need to rant for a bit here. I've been playing with Z2M for a while now with just a handful of devices, because I want to migrate away from ZHA. I like Z2M because of the web interface and because of MQTT, but that's about it.

I've been experiencing a bunch of issues. [I created an issue over a month ago](https://github.com/Koenkk/zigbee2mqtt/issues/10704), because for some weird reason, my routers won't pair again after being removed from Z2M. I provided a lot of details, and some people reported they experienced the same thing. However, I got no reply, so I kind of disbanded the idea of using Z2M.

Just today I was doing some server maintenance, so I took my server offline for about an hour. After booting the server up again and starting the Z2M Docker container, not a single one of my devices paired with Z2M work. No routers or no end devices. Poof. Completely dead. I found similar issues from other people, but no real answer was given.

So far I have not been impressed with Z2M at all, whereas ZHA just freaking works. Even using my 2+ year old CC2531 stick. I have been using for ZHA for a couple of years now, and I haven't experienced any alarming issues what so ever.

I must be doing something utterly wrong, since I'm having this many issues, but when I ask for help in a detailed issue post here on Github, I don't get any answers. What if I decide to move 50+ devices to Z2M, and all of a sudden all of my devices disconnect, because I reboot my server? Would I get any help then, or should I pair them all to ZHA again, until someone replies?

I'm not sure how to end off this rant, so I'd just like to say sorry for sounding grumpy.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Sonoff CC2652P

KaHeMu, 2022-02-17:
I'm running Z2M for 2 years over various revisions and my network grew to 60+ devices but I never had the issue that all devices disconnected after a server reboot. I re-setup the server a couple of times and I even migrated from a CC2531 stick to CC2652 and zStack3.0 without having to repair any device!  Yes, converters are sometimes buggy especially if you use brand new devices but mostly this gets fixed by the authors quickly within a couple of revisions.  The only case when I completed disconnected by devices was in the early days when I played around with channel numbers and pan IDs without knowing what I was doing. All in all I cannot second your opinion.

wb441, 2022-02-17:
Sorry to hear you're having a bad experience.
I am all too familiar with the sensation, but the important thing to keep in mind is that it is a hobby more than anything else.
Unfortunately, for me at least, that sometimes means trying 10+ times to do something, and finding out that it was mostly a massive waste of time :-)

As I'm sure you understand, every item should be tackled separately.
If suddenly, all the devices don't respond anymore, I suspect interference.
As for the router not connecting to the network, there unfortunately I don't have experience to share, as I have never run a board just as router.

My honest suggestion would be to use a plug or other zigbee device which is a router.
I agree it is in no way a solution or answer to the fact that it doesn't work.
There is definitely a reason, we just haven't found it yet.

And man, when it works, it's f* awesome!

thebatfink, 2022-02-20:
I had to take the power off my lights recently to do some electrical lights and also ended up having to rebind them so you are definitely not alone there.

closed 2022-03-30 by github-actions[bot]