Koenkk/zigbee2mqtt #11439: [New device support]: PSE03-v1.1.0

mickadam29, 2022-02-15:
### Link

https://www.leroymerlin.fr/produits/electricite-domotique/alarme-camera-de-surveillance-et-detecteur-de-fumee/alarme-maison/sirene-alarme/sirene-interieure-connectee-evology-80123668.html

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x000d6f000f1957ad","nwkAddr":25240,"manufId":65440,"manufName":"","powerSource":"DC Source","modelId":"PSE03-v1.1.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,9,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"PSE03-v1.1.0","manufacturerName":"","powerSource":4,"zclVersion":1,"appVersion":0,"stackVersion":0,"hwVersion":0,"swBuildId":"1.1.0"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00158d0001ed3267","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"swBuildId":"1.1.0","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1644946473349,"defaultSendRequestWhen":"immediate"}

### Comments

Branch : DEV

Clusters :

cluster genbasic
cluster ssIasZone
cluster powercfg
cluster genalarms
cluster ssIasWd
cluster ota

From Plugin Zigbee  / Jeedom 

Armement / Désarmement

Logical ID = 1::ias_wd::squawk:: A :: B :: C, avec :

A = squawk mode (0 = system is armed / 1 = system is disarmed)
B = Strobe (0 = no strobe / 1 = use strobe)
C = Squawk level (0 = low level / 1 = medium level / 2 = high level / 3 = very high level)

J’ai fait plein d’essais et l’équipement permet uniquement d’envoyer une valeur sur A (0 ou 1). Cela a pour action de faire sonner ou arrêter l’alarme (avec la son « urgence ») et de faire clignoter le flash lumineux.

Les 2 autres paramètres (B et C) ne changent rien au comportement de la sirène.

J’ai donc créé 2 commandes :

Armement : 1::ias_wd::squawk::0::0::0
Désarmement : 1::ias_wd::squawk::1::0::0
Choix de la sonnerie, durée,…

Logical ID = 1::ias_wd::start_warning:: A :: B :: C :: D :: E :: F, avec :

A = Type de sonnerie
B = Strobe
C = Level Siren
D = Durée de la sirène
E = Strobe cycle
F = Strobe level

Idem, j’ai fait de nombreux essais pour en arriver au fonctionnement suivant :

A : valeur possible de 0 à 9 avec 0 = stop / 1 = urgence / 2 = incendie / 3 = police / 4 = pompier / 5 = sonnette / 6 = Double ding / 7 = Flash seul / 8 = Ding 1 / 9 = Ding 2.

B : 0 ou 1 => ne change rien, le flash lumineux s’active quel que soit la valeur.

C : 0 = low level / 1 = medium level / 2 = high level / 3 = very high level => ne change rien le volume de la sirène reste le même quel que soit la valeur.

D : de 1 à 180 sec (la valeur max_duration est donnée dans les informations brutes du module) => cette valeur modifie bien la durée de la sirène de 1 à 180 sec.

                {
                    "id": 1282,
                    "name": "IAS Warning Device",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "max_duration",
                            "value": 180
                        }

E : de 0 à 100% => aucune action sur le flash lumineux, même fréquence de clignotement.

F : 0 à 10 => aucune action sur le flash lumineux, l’intensité lumineuse reste la même.

With external converter Mode and Duration work fine
Strobe Cycle and Level N/A

Mode are not all available with this conveter

Don't receive ACK, so commands are send TWICE

Device send status : 0 or 1 Armed/Disarmed

It seems squawk is OK, not in the converter

Thank's for help
 

[Manual PSE03-1.pdf](https://github.com/Koenkk/zigbee2mqtt/files/8073689/Manual.PSE03-1.pdf)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['PSE03-v1.1.0'],
        model: 'PSE03-v1.1.0',
        vendor: 'EVOLOGY',
        description: 'Sound and flash siren',
        fromZigbee: [fz.battery],
        exposes: [e.warning(), e.battery()],
        toZigbee: [tz.warning],
    
    },
];
```

mickadam29, 2022-02-16:
This payload works (mode and duration)

{"warning":{"duration":5,"level":"low","mode":"burglar","strobe":false,"strobe_duty_cycle":0,"strobe_level":"low"}}

So a model with only mode and duration is nessary.

closed 2022-03-26 by github-actions[bot]

thibp, 2023-02-12:
Hello,
Could you please use 
`zigbeeModel: ['PSE03-V1.1.0', 'PSE03-v1.1.0'],`
Instead, because mine is not recognized?
Thanks in advance,
Regards