Koenkk/zigbee2mqtt #11442: Options cannot be set in the configure_reporting

wb-adegtyarev, 2022-02-15:
### What happened?

Hi, please help me.

I send to the topic `zigbee2mqtt/bridge/request/device/configure_reporting`:
```
{
  "id": "0x842e14fffe8b184e/1",
  "options": {"manufacturerCode":26214},
  "cluster": "msOccupancySensing",
  "attribute": "sprutOccupancyLevel",
  "minimum_report_interval": 5,
  "maximum_report_interval": 10,
  "reportable_change": 10
}
```
But the manufacturerCode option is not passed and in response it writes that the attribute is not supported. How to transfer the manufacturer's code?

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211115

### Adapter

CC2530

Koenkk, 2022-02-16:
Added this now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

wb-adegtyarev, 2022-02-17:
> Added this now.

Thanks, I'll test. Tell me more, what should I do to add clusters of my device to the report settings?
[image]

Koenkk, 2022-02-18:
I believe the frontend doesn't support clusters via ID (number), only the ones in https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts so this would be a feature request for https://github.com/nurikk/zigbee2mqtt-frontend/issues

wb-adegtyarev, 2022-02-21:
> I believe the frontend doesn't support clusters via ID (number), only the ones in https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts so this would be a feature request for https://github.com/nurikk/zigbee2mqtt-frontend/issues
Hi.
The clusters and attributes of my device are described in cluster.ts. I expected them to be added to the frontend automatically, but they didn't.

The options in the configure_reporting work, thanks.

[image]
[image]

Koenkk, 2022-02-21:
@wb-adegtyarev regarding the clusters please create an issue here: https://github.com/nurikk/zigbee2mqtt-frontend/issues

Since the options are working now I will close this.

closed 2022-02-21 by Koenkk