Koenkk/zigbee2mqtt #11445: Home Assistant error reporting switch as light

ksya, 2022-02-15:
### What happened?

With the latest version of Home Assistant (2022.2), I get the following error in HA and the light switch is shown as unavailble and cannot be controlled:
`Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Hobbykamer/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Hobbykamer/set', 'device': {'identifiers': ['zigbee2mqtt_0x804b50fffe0a6717'], 'manufacturer': 'SONOFF', 'model': 'Zigbee two way smart switch (ZBMINI)', 'name': 'Hobbykamer'}, 'json_attributes_topic': 'zigbee2mqtt/Hobbykamer', 'name': 'Hobbykamer', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Hobbykamer', 'unique_id': '0x804b50fffe0a6717_light_zigbee2mqtt', 'value_template': '{{ value_json.state }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 41, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: The 'value_template' option has been removed, please remove it from your configuration`

This is the config I use in devices.yaml:
`'0x804b50fffe0a6717':
  friendly_name: Hobbykamer
  homeassistant:
    switch:
      type: light
      object_id: light`

### What did you expect to happen?

Light not "unavailable" and controllable

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20200925

### Adapter

Slaesh stick

adizag, 2022-02-16:
@Koenkk 
seen all the issues for this problem, fixed the code according to your fix (#11211), but still can't get the light entities after upgrading to 2022.2.

What shall I do??

TNX!

HA entities:
[image]

my setup:
```
  '0xec1bbdfffe2c55be':
    friendly_name: Mamad 2nd
    homeassistant:
      switch_l1:
        type: light
        object_id: mamad_2nd_light_l1
      light_l1:
        name: mamad_2nd_l1
        value_template: null
        state_value_template: '{{ value_json.state_l1 }}'
      switch_l2:
        type: light
        object_id: mamad_2nd_light_l2
      light_l2:
        name: mamad_2nd_l2
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'
```

HA log:
```
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Stairs Down/l2/set', 'device': {'identifiers': ['zigbee2mqtt_0x5c0272fffe8ef707'], 'manufacturer': 'TuYa', 'model': '2 gang switch (TS0002)', 'name': 'Stairs Down'}, 'json_attributes_topic': 'zigbee2mqtt/Stairs Down', 'name': 'Stairs Down stairs down light 2', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Stairs Down', 'unique_id': '0x5c0272fffe8ef707_stairs_down_light_2_zigbee2mqtt', 'value_template': '{{ value_json.state_l2 }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 41, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: The 'value_template' option has been removed, please remove it from your configuration
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Mamad 2nd/l1/set', 'device': {'identifiers': ['zigbee2mqtt_0xec1bbdfffe2c55be'], 'manufacturer': 'TuYa', 'model': '2 gang switch (TS0002)', 'name': 'Mamad 2nd'}, 'json_attributes_topic': 'zigbee2mqtt/Mamad 2nd', 'name': 'Mamad 2nd mamad 2nd light l1', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Mamad 2nd', 'unique_id': '0xec1bbdfffe2c55be_mamad_2nd_light_l1_zigbee2mqtt', 'value_template': '{{ value_json.state_l1 }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/p [... cut ...]
Zigbee2MQTT:debug 2022-02-16 23:14:15: Received Zigbee message from 'motion_sensor_girls', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3045,"10":45750,"100":0,"11":8,"3":28,"4":5032,"5":26,"6":[0,7]}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-02-16 23:14:15: MQTT publish: topic 'zigbee2mqtt/motion_sensor_girls', payload '{"battery":100,"illuminance":8,"illuminance_lux":8,"linkquality":70,"occupancy":false,"temperature":28,"voltage":3045}'
Zigbee2MQTT:debug 2022-02-16 23:14:45: Received Zigbee message from 'Mamad', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":66}' from endpoint 1 with groupID 0
```

Koenkk, 2022-02-18:
Can you provide the debug log from starting z2m till 10 seconds after that?

adizag, 2022-02-19:
@Koenkk , here:

```
Zigbee2MQTT:debug 2022-02-19 23:52:57: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2022-02-19 23:52:57: Logging to console and directory: '/app/data/log/2022-02-19.23-52-54' filename: log.txt
Zigbee2MQTT:debug 2022-02-19 23:52:57: Removing old log directory '/app/data/log/2022-02-16.09-47-22'
Zigbee2MQTT:info  2022-02-19 23:52:57: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)
Zigbee2MQTT:info  2022-02-19 23:52:57: Starting zigbee-herdsman (0.14.10)
Zigbee2MQTT:debug 2022-02-19 23:52:57: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2022-02-19 23:52:58: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-19 23:52:58: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:debug 2022-02-19 23:52:58: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b001cdd3b52","panID":6754}
Zigbee2MQTT:info  2022-02-19 23:52:58: Currently 11 devices are joined:
Zigbee2MQTT:info  2022-02-19 23:52:58: Stairs Up (0xbc33acfffec80607): ZM-L03E-Z - Zemismart Smart light switch - 3 gang with neutral wire (Router)
Zigbee2MQTT:info  2022-02-19 23:52:58: Stairs Down (0x5c0272fffe8ef707): TS0002 - TuYa 2 gang switch (Router)
Zigbee2MQTT:info  2022-02-19 23:52:58: Mamad 2nd (0xec1bbdfffe2c55be): TS0002 - TuYa 2 gang switch (Router)
Zigbee2MQTT:info  2022-02-19 23:52:58: Mamad (0x5c0272fffe8ef714): TS0002 - TuYa 2 gang switch (Router)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_master_bedroom (0x00158d0006a261df): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_bathroom (0x00158d0006a157a6): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_shower_horim (0x00158d0006f7805b): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_girls (0x00158d0006a25d21): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_guests_room (0x00158d0006f25853): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_laundry_room (0x00158d0006f25cb9): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:info  2022-02-19 23:52:58: motion_sensor_salon_kitchen (0x00158d0006a25f66): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
Zigbee2MQTT:warn  2022-02-19 23:52:58: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-02-19 23:52:58: Allowing new devices to join.
Zigbee2MQTT:warn  2022-02-19 23:52:58: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-02-19 23:52:58: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-02-19 23:52:59: Connecting to MQTT server at mqtt://10.0.0.251
Zigbee2MQTT:debug 2022-02-19 23:52:59: Using MQTT keepalive: 60
Zigbee2MQTT:debug 2022-02-19 23:52:59: Using MQTT anonymous login
Zigbee2MQTT:info  2022-02-19 23:52:59: Connected to MQTT server
Zigbee2MQTT:info  2022-02-19 23:52:59: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:debug 2022-02-19 23:52:59: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"6ea61ba","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"last_seen":"disable","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"blocklist":[],"device_options":{},"devices":{"0x00158d0006a157a6":{"friendly_name":"motion_sensor_bathroom"},"0x00158d0006a25d21":{"friendly_name":"motion_sensor_girls"},"0x00158d0006a25f66":{"friendly_name":"motion_sensor_salon_kitchen","occupancy_timeout":10},"0x00158d0006a261df":{"friendly_name":"motion_sensor_master_bedroom"},"0x00158d0006f25853":{"friendly_name":"motion_sensor_guests_room"},"0x00158d0006f25cb9":{"friendly_name":"motion_sensor_laundry_room"},"0x00158d0006f7805b":{"friendly_name":"motion_sensor_shower_horim"},"0x5c0272fffe8ef707":{"friendly_name":"Stairs Down","homeassistant":{"light_l1":{"name":"stairs_down_light_1","state_value_template":"{{ value_json.state_l1 }}","value_template":null},"light_l2":{"name":"stairs_down_light_2","state_value_template":"{{ value_json.state_l2 }}","value_template":null},"switch_l1":{"object_id":"stairs_down_light_1","type":"light"},"switch_l2":{"object_id":"stairs_down_light_2","type":"l [... cut ...] ery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/motion_sensor_salon_kitchen","name":"motion_sensor_salon_kitchen_battery","state_class":"measurement","state_topic":"zigbee2mqtt/motion_sensor_salon_kitchen","unique_id":"0x00158d0006a25f66_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:debug 2022-02-19 23:53:00: Received MQTT message on 'homeassistant/binary_sensor/0x00158d0006a25f66/occupa

Koenkk, 2022-02-20:
Can you try with the following config:

```yaml
  '0xec1bbdfffe2c55be':
    friendly_name: Mamad 2nd
    homeassistant:
      switch_l1:
        type: light
        object_id: mamad_2nd_light_l1
        value_template: null
        state_value_template: '{{ value_json.state_l1 }}'
      light_l1:
        name: mamad_2nd_l1
        value_template: null
        state_value_template: '{{ value_json.state_l1 }}'
      switch_l2:
        type: light
        object_id: mamad_2nd_light_l2
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'
      light_l2:
        name: mamad_2nd_l2
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'
```

adizag, 2022-02-20:
unfortunately not working. not recognized in HA.

HA logs:
```
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Mamad 2nd/l1/set', 'device': {'identifiers': ['zigbee2mqtt_0xec1bbdfffe2c55be'], 'manufacturer': 'TuYa', 'model': '2 gang switch (TS0002)', 'name': 'Mamad 2nd'}, 'json_attributes_topic': 'zigbee2mqtt/Mamad 2nd', 'name': 'Mamad 2nd mamad 2nd light l1', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Mamad 2nd', 'unique_id': '0xec1bbdfffe2c55be_mamad_2nd_light_l1_zigbee2mqtt', 'value_template': '{{ value_json.state_l1 }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 41, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: The 'value_template' option has been removed, please remove it from your configuration
Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Mamad 2nd/l2/set', 'device': {'identifiers': ['zigbee2mqtt_0xec1bbdfffe2c55be'], 'manufacturer': 'TuYa', 'model': '2 gang switch (TS0002)', 'name': 'Mamad 2nd'}, 'json_attributes_topic': 'zigbee2mqtt/Mamad 2nd', 'name': 'Mamad 2nd mamad 2nd light l2', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Mamad 2nd', 'unique_id': '0xec1bbdfffe2c55be_mamad_2nd_light_l2_zigbee2mqtt', 'value_template': '{{ value_json.state_l2 }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator raise vol.Invalid(warning % arguments) voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 41, in validate_mqtt_light_discovery return schemas[value[CONF_SCHEMA]](value) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: The 'value_template' option has been removed, please remove it from your configuration
```

Koenkk, 2022-02-20:
I see your config is wrong, `object_id: mamad_2nd_light_l1` should be `object_id: light_l1`;

```yaml
devices:
  '0xec1bbdfffe2c55be':
    friendly_name: Mamad 2nd
    homeassistant:
      switch_l1:
        type: light
        object_id: light_l1
      light_l1:
        name: mamad_2nd_l1
        value_template: null
        state_value_template: '{{ value_json.state_l1 }}'
      switch_l2:
        type: light
        object_id: light_l2
      light_l2:
        name: mamad_2nd_l2
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'
````

adizag, 2022-02-21:
Working!
Strange thing - how could that work in the former version of HA...?
anyway, thanks. maybe better to have a more clear explanation about the headers and variables in the WIKI for the coding-amature? 

tnx again!