Koenkk/zigbee2mqtt #11446: [Feature request]: Include a Device Name and TimeStamp in the message payload.

nothingmn, 2022-02-16:
### Is your feature request related to a problem? Please describe

No.

### Describe the solution you'd like

I would like to see two additional properties added to the JSON payload sent to MQTT.  

DeviceName
- Right now im parsing the topic to get the device name, but I would rather skip that step and just have it provided.
 
Timestamp
- In UTC format in ISO 8601 Format

The primary scenario is to avoid any pre-processing of the prior to storage or event triggering...

### Describe alternatives you've considered

Right now I pre-process all incoming data and then republish it for my primary workflows.  

### Additional context

None.

sjorge, 2022-02-16:
You can configure advanced -> last_seen or advanced -> elapsed to get the timestamp you want:
https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#server-connection

nothingmn, 2022-02-17:
Thanks @sjorge 

I also noticed there is a flag in the UI to send "Device Information" as well.  That solves both of my questions

Great project.  Love it.

closed 2022-02-17 by nothingmn