Koenkk/zigbee2mqtt #11449: No converter available for

rwb196884, 2022-02-16:
### What happened?

Log is full of these `No converter available for...` errors and the UI screen is full of red boxes for them.

What's going on?

### What did you expect to happen?

Nothing.

### How to reproduce it (minimal and precise)

Try to use a [TuYa WSD500A](https://www.zigbee2mqtt.io/devices/WSD500A.html#tuya-wsd500a).

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20190608

### Adapter

TI CC2531 USB CDC

### Debug log

...
info  2022-02-16 12:39:18: Changed config for device 0xa4c1385fc72de22e
info  2022-02-16 12:39:18: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{},"id":"0xa4c1385fc72de22e","to":{"optimistic":true,"retain":true}},"status":"ok","transaction":"2eyy2-3"}'
error 2022-02-16 12:39:18: No converter available for 'battery' (0)
error 2022-02-16 12:39:18: No converter available for 'temperature' (0)
error 2022-02-16 12:39:18: No converter available for 'humidity' (0)
error 2022-02-16 12:39:18: No converter available for 'voltage' (0)
error 2022-02-16 12:39:18: No converter available for 'linkquality' (0)
error 2022-02-16 12:39:18: No converter available for 'battery' (0)
error 2022-02-16 12:39:18: No converter available for 'temperature' (0)
error 2022-02-16 12:39:18: No converter available for 'humidity' (0)
error 2022-02-16 12:39:18: No converter available for 'voltage' (0)
error 2022-02-16 12:39:18: No converter available for 'linkquality' (0)
info  2022-02-16 12:39:35: Changed config for device 0xa4c1385fc72de22e
info  2022-02-16 12:39:35: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"optimistic":true,"retain":true},"id":"0xa4c1385fc72de22e","to":{"optimistic":true,"retain":true,"temperature_precision":1}},"status":"ok","transaction":"2eyy2-4"}'
error 2022-02-16 12:39:35: No converter available for 'battery' (0)
error 2022-02-16 12:39:35: No converter available for 'temperature' (0)
error 2022-02-16 12:39:35: No converter available for 'humidity' (0)
error 2022-02-16 12:39:35: No converter available for 'voltage' (0)
error 2022-02-16 12:39:35: No converter available for 'linkquality' (0)
error 2022-02-16 12:39:35: No converter available for 'battery' (0)
error 2022-02-16 12:39:35: No converter available for 'temperature' (0)
error 2022-02-16 12:39:35: No converter available for 'humidity' (0)
error 2022-02-16 12:39:35: No converter available for 'voltage' (0)
error 2022-02-16 12:39:35: No converter available for 'linkquality' (0)
info  2022-02-16 12:43:18: MQTT publish: topic 'zigbee2mqtt/Home/Bed3/Temp', payload '{"humidity":56.84,"linkquality":110,"temperature":19.1}'
info  2022-02-16 13:02:03: MQTT publish: topic 'zigbee2mqtt/Home/Outside/Temp', payload '{"humidity":82.22,"linkquality":0,"temperature":13.4}'

Koenkk, 2022-02-16:
Please provide the debug log of this.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

closed 2022-03-26 by github-actions[bot]

surrounds, 2022-04-15:
I have same issue, do you have some advice?

BlackLight, 2022-10-30:
Same issue here, and it's actually been long-lasting. It regularly applies to properties such as `battery`, `linkquality` and `voltage`, regardless of the device.

[The API documentation](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-friendly-name-get) provides an example on how to retrieve the `state` of a property (push to `zigbee2mqtt/FRIENDLY_NAME/get` with `{"state": ""}` as a payload). But it doesn't tell much about how to retrieve properties that are _not_ strings.

I have tried to pass both `{"battery": ""}` and `{"battery": 0}` to the payload (and, in general, pass a zero if the property is numeric), as well as `{"battery": null}`, but none of them works. The `No converter available` error is always there.