Koenkk/zigbee2mqtt #11460: Zigbee 3.0 dual USB wireless socket plug incorrectly identified

NeuerUser, 2022-02-17:
### What happened?

Added the following device to z2m:
https://www.zigbee2mqtt.io/devices/ZP-LZ-FR2U.html
The interview works wirthout problems, but the device gets recognized as a different plug:
https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3

Zigbee model: TS011F  ( _TZ3000_2xlvlnez)

The effect is that the endpoint for the USB outlets is missing and instead it tries to show power metering values that the device is not reporting.

### What did you expect to happen?

To get it recognized as the correct device.

### Zigbee2MQTT version

1.23.0-dev commit: f284a5d

### Adapter firmware version

20211219

### Adapter

zStack3x0 zigStarGW

NeuerUser, 2022-02-17:
Could be the same problem as here: https://github.com/Koenkk/zigbee2mqtt/issues/10812
Except mine has a CH plug type and a different manufacturerName.

NeuerUser, 2022-02-18:
Yes, it is this device (except with a Swiss plug). I got it working with an external converter, using the one listed in #10812 with an adapted manufacturerName.
I commented in this issue.
Interesting side node is that the device also supports "child lock/unlock", which worked when it was incorrectly identified as the Tuya plug, but it's not included in the external converter for the Moe's plug.

Koenkk, 2022-02-19:
@jzmudzinski can you confirm?

jzmudzinski, 2022-02-19:
Yes, I confirm. That is the device. Sorry for any inconvenience my commit caused.

Koenkk, 2022-02-19:
@jzmudzinski thanks,

Fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-19 by Koenkk

NeuerUser, 2022-02-23:
Confirmed. Working correctly now. Thanks!

sliwma, 2022-02-24:
Affirmative, work now. And, besides "child lock", LED state select is missing - it also worked.

Barrysv, 2022-10-18:
@sliwma Hi there - are you saying there is a way to turn off the power LED via MQTT? Is that what you mean by "LED state select"? If so that would be great - what is the topic/command?  Cheers