Koenkk/zigbee2mqtt #11463: Develco energy meter, some readings changed from float to int

mikke49, 2022-02-17:
### What happened?

After update to verison 1.23.0 the volt and current readings on phases B and C lost fractional digits, i.e. is reported as int instead of float. Volt and current on phase A remain as expected with fractional digits.

I'm using the Develco HAN dongle for Norwegian energy meters.

Cluster as viewed in deConz application shows correct values:
[image]

Z2M trucates/rounds values for phases B and C:
[image]

### What did you expect to happen?

Volt and current readings for all three phases reported as floats.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

Koenkk, 2022-02-18:
Should be fixed now, please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

mikke49, 2022-02-21:
On vacation, will check as soon as back home later this week. Thanks.

mikke49, 2022-02-25:
Tested now, everything seems ok. The voltage and current readings on phases B and C are reverted to floats.

closed 2022-02-25 by mikke49