Koenkk/zigbee2mqtt #11464: MiBoxer FUT035Z declared unsupported after pairing

elle2, 2022-02-17:
### What happened?

The LED controller MiBoxer FUT035Z is in supported devices list at zigbee2mqtt.io, but after pairing it's defined as unsupported device in log file where it is declared as **Zigbee model 'TS0502B' and manufacturer name '_TZ3210_jtifm80b'**

### What did you expect to happen?

I expected it to be recognized correctly

### Zigbee2MQTT version

1.23.0 commit: 6ea61ba

### Adapter firmware version

 20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

info  2022-02-17 15:18:25: Accepting joining not in blocklist device '0xa4c138d719b4598e'
info  2022-02-17 15:18:25: Device '0xa4c138d719b4598e' joined
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d719b4598e","ieee_address":"0xa4c138d719b4598e"},"type":"device_joined"}'
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138d719b4598e"},"type":"device_connected"}'
info  2022-02-17 15:18:25: Starting interview of '0xa4c138d719b4598e'
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d719b4598e","ieee_address":"0xa4c138d719b4598e","status":"started"},"type":"device_interview"}'
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138d719b4598e"},"type":"pairing"}'
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138d719b4598e","ieee_address":"0xa4c138d719b4598e"},"type":"device_announce"}'
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138d719b4598e"},"type":"device_announced"}'
info  2022-02-17 15:18:25: Successfully interviewed '0xa4c138d719b4598e', device has successfully been paired
warn  2022-02-17 15:18:25: Device '0xa4c138d719b4598e' with Zigbee model 'TS0502B' and manufacturer name '_TZ3210_jtifm80b' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138d719b4598e","ieee_address":"0xa4c138d719b4598e","status":"successful","supported":false},"type":"device_interview"}'
info  2022-02-17 15:18:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xa4c138d719b4598e","supported":false},"type":"pairing"}'

Koenkk, 2022-02-18:
Just added this (https://github.com/Koenkk/zigbee-herdsman-converters/commit/0fd00b928e49c6283a83113a6f7cedf6a0dc46f6)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-02-18 by Koenkk

MotoMoto42, 2023-02-21:
Good day!

I have this problem with my FUT035Z - then connected to z2m, detect as :

Zigbee model: TS0502B
Manufacturer name: _TZB210_lmqquxus

tklnew, 2023-02-25:
The manufacturer ID isn't implemented yet.

Zigbee Model: TS0502B
Zigbee Manufacturer: _TZB210_lmqquxus

this device should be detected as:
[https://www.zigbee2mqtt.io/devices/FUT035Z.html#miboxer-fut035z](https://github.com/Koenkk/zigbee2mqtt/issues/url)

Link of this device (product page)
https://aliexpress.ru/item/1005004114049995.html?sku_id=12000028077770800

Please add it to the latest dev commit.

bzumik1, 2023-10-28:
Hi, I have **FUT035Z plus** and it was incorrectly recognized as [ts0502b](https://www.zigbee2mqtt.io/devices/TS0502B.html#tuya-ts0502b). It partially works but when I turn the controller using the switch (plus version supports wired switch) the state of the light is not reported and it is still showing as off even though it is turned on.