Koenkk/zigbee2mqtt #11466: Xiaomi QBKG03LM / QBKG04LM switches messed up 'action' when legacy mode is disabled

uncle-fed, 2022-02-17:
### What happened?

This issue picks up on the abandoned problem report [#3002](https://github.com/Koenkk/zigbee-herdsman-converters/issues/3002), with some additional information about incorrect behaviour.

There are several issues with the QBKG03LM / QBKG04LM  single and 2-button switches when you press (or hold) their button(s).

### What did you expect to happen?

The generated payload does not correspond to the documented (and the expected) one when z2m is running with disabled legacy mode.

### How to reproduce it (minimal and precise)

z2m has the following config options set:
```yaml
advanced:
  legacy_availability_payload: false
  legacy_api: false
device_options:
  legacy: false
```

* 1-button switch emits `action: release`, instead of the `action: single` when the button is clicked ordinarily. According to the documentation, the `release` action should only be seen when the switch button is held and then released.

* Similarly, the 2-button switch emits `action: release_left` or `action: release_right` or `release_both`, instead of the `single_left` or `single_right` or `single_both` when the buttons are pressed shortly.

* In addition to `action`, the 2-button switch has in its payload `button_left: ...` or `button_right: ...` property, which is not mentioned anywhere in the documentation. Even in the legacy mode it only mentions the deprecated `click_*` poperty, but nothing about `button_` . Single button switch behaves correctly (does not transmit the `button` property in this case).

* Even though it is not documented, it seems that the switch might be capable of distinguishing a double-click event. If you try it on a 2-button switch, for example, you get `action: undefined_left`, which is different from `release_left` or `hold_left`. This may suggest that the switch could potentially offer more actions.

* Getting the `operation_mode` property does not work. Every time I try, I get the timeout `Publish 'get' 'operation_mode' to '....' failed: 'Error: Read 0x00xxxxxxxxxxxxxxxx/2 genBasic([65314], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 36884 - 2 - 19 - 0 - 1 after 10000ms)'`.  However, setting the `operation_mode` with `/set` does work properly and is instant.

### Zigbee2MQTT version

1.23.0-dev commit: aace02cc

### Adapter firmware version

20190523

### Adapter

СС2538-RPi-Hat

uncle-fed, 2022-02-18:
> About the first 3 points, we should put the non legacy actions in the exposes, could you provide a list of all options?

I think one of us is confused here (currently me, with your reply :-)) 

I was trying to say that the documentation is fine (i.e. the documented behaviour = expected behaviour = how it _should_ be), but the current implementation does not conform to that. The names of the "single click" events that z2m emits are not correct at present (as described in the first 2 points). And the 3rd point is about the fact that the currently present property should NOT be present in the payload, because it is not expected and it is not documented, i.e. it should be removed from the code, not put into the documentation :-) 

In other words, this is not about documentation, it is about the code changes that should make z2m work as currently documented.

> This looks like your device does not support this.

In this case, shouldn't the web UI / Zigbee converter NOT allow this GET operation and reject it immediately?  Both models of the switches discussed here do not support "get" of the `operation_mode` but the operation is permitted with the timeout error that is produced after 10 seconds.  So you are basically allowed to send a request that you know is not supported, which results that for the next 10 seconds after such request the device is "paralyzed" (all the following requests to the same device are stuck until the timeout is reached).

Koenkk, 2022-02-18:
The value of `action` depends wether you set legacy: false or true, what should be shown in the docs is what is send when legacy: false.

> In this case, shouldn't the web UI / Zigbee converter NOT allow this GET operation and reject it immediately?

Done (shouldn't show the get anymore).

Changes will be available in the dev branch in tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

uncle-fed, 2022-02-18:
As descibed in my inital post, points 1 and 2:

* QBKG04LM emits `action: release`, instead of the documented `action: single` when the button is clicked ordinarily. 
* QBKG03LM emits `action: release_left` or action: `release_right` or `release_both`, instead of the `single_left` or `single_right` or `single_both`

uncle-fed, 2022-02-18:
I will provide the `undefined` on the weekend, as I am not able to test today.

Koenkk, 2022-02-19:
- Fixed the actions now (docs will be regenerated with next release on 1 March)
- `button_*` will not be send anymore for QBKG03LM when `legacy: false`
- action `double_right` should be send now, but are you able to trigger this for the left button and QBKG04LM? 
    - The value of `onOff` is used to determine this, messages with `61440` are ignored. 
    - 2 = double; 1 = release; 0 = hold;
    - For double click right you got `{"onOff":2}`
    - For double click left you got `{"onOff":1}`
    - **Note to self;** add these actions to the exposes once confirmed.

uncle-fed, 2022-02-19:
@Koenkk Thanks a lot, I've verified the latest code changes.

Double presses are now recognized (even though they are a bit awkward to actually generate consistently).
I was now able to trigger the `double left` (that was wrong in my last message).

So, everything you've said in the last comment is true and correct. Unfortunately.

Unfortunately, because both "single click" and a "release after hold" generate the same `release` event and this was precisely the reason why I opened this issue in the first place.
I was hoping that the documentation was right and that the implementation was wrong, but the reverse is true: the switch seems to only realize 3 actions: release, hold and double, as you wrote.

To me this essentially means that the switch cannot be used to create scenarios where the same device could launch different actions on single press and on hold.
If you configure action for the "release" event, it will be launched either on single press or at the end of hold - no distinction.
Which then means that if you have a single press action configured, then the hold cannot be used at all, as there is no way to distinguish between the end of hold or a single click.

I have a proposal here that maybe is able to resolve this "incovenience". 
Could there be an (internal, per device) flag that gets raised every time the `hold` event comes in.
When the `release` event comes, z2m will first check the flag.
If the flag is not set, the `release` should get converted into `single`.
If the flag was set, the `release` should remain without changes and the flag should get reset.

What do you think?

Koenkk, 2022-02-20:
What if the release actions will include the last action? So you will have a `hold_release` and `single_release`?

uncle-fed, 2022-02-20:
We could also do `single_release`... but that's just basically exposing the internal flag suggested by me to the end user.

This of course would be already much better than the current implementation, but why not just actually implementing what has been documented from the start? I.e., the documentation pages for these switches stated `action: single` (not `action: release`)  from the very first day when the `legacy: false` was introduced. So to me, the topic is not about "breaking behaviour" but actually making it work how it was always documented, and how it actually makes sense for being able to use automations properly.

uncle-fed, 2022-02-20:
See another user complaining about the documented behaviour not corresponding to reality here:
https://github.com/Koenkk/zigbee-herdsman-converters/issues/3002

Koenkk, 2022-02-20:
i understand your point, but changing the behaviour is a breaking change, users depend on the current behaviour. So this new behaviour (with the flag stuff) should definitely be an option imo.

uncle-fed, 2022-02-20:
> changing the behaviour is a breaking change

It is going to be breaking for those who never read the docs and just relied on the implemented behavior. Can we call something that _fixes_ the behavior to be compliant with the documentation to be breaking? I'll leave you to be the judge of that.  :)

I saw the problem because I read the docs and it didn't match the reality, so I immediately opened the issue here. Another person (quoted above) did the same. If others implemented something without following the API/docs and now it's going to break for them, whose problem is that? :)

Koenkk, 2022-02-20:
In my opinion the docs were wrong (since the `single` action was there while it was never intended to be exposed. But that doesn't bring us any further.

I was wondering, is the `hold` always send? Or just when you hold if for a few secs

uncle-fed, 2022-02-20:
`hold` is sent only when you hold the button for longer than a second or so, and the subsequent `release` is sent once you let go.
`hold` does not get sent at all if you single-press or double-press.

Koenkk, 2022-02-20:
I see, so another non-breaking possibility here would be to publish two actions for the single case. Although for the release case you don't know if it was from a hold or single. What do you think?

So:
- single press: `release`, `single` (new)
- hold: `hold`, `release`

uncle-fed, 2022-02-20:
That sounds like a compromise solution which aims at being non-breaking.

You're right, it leaves one use case incomplete / broken (depending on what you intend to do with the hold/release actions). Not ideal... But we could extend this idea and go all the way: we should also then add another (second) action for the end of the "hold/release" sequence and call it `hold_release` and thus would allow both old and new logic without any breaking changes. The old actions remain as the `release` and the new ones would allow to distinguish between all of the events.

However, I still would like to point out that this discussion changes the accents about what's more important (in the long run) if we look in the context of `legacy: false`. I understood that the initiative of doing away with the legacy mode was put together as an attempt to "harmonize and make more/less uniform" all the various button events' names and related properties that were previously named a bit wild and inconsistent. In this light, introducing a "somewhat weird event concept with the same event called differently" feels rather strange and inconsistent again.

If there were no "legacy: false" general approach, I would probably agree that it is better to add a set of "second" actions that would serve 2 purposes. But as the intention was to do away with the ambiguous stuff in the non-legacy mode, I think (for me personally) it is better to announce "a possible breaking change" and have "single", "double", "hold" and "release" events and that's it.

I know that a lot of people on the forums and Telegram channels dedicated to the theme are always discussing the changelogs for every single z2m release and even minor update. Surely people would notice a text "POTENTIALLY BREAKING" ;-)

uncle-fed, 2022-02-20:
Maybe, as yet another compromise, for now the set of secondary actions (`single` and `hold_release`) should be simply introduced. But the day you decide to make `legacy: false` the default behaviour of zigbee2mqtt, the `release` events should also be gone completely. What do you think?

z-x-f, 2022-02-21:
The following three pictures are the status of the switch report when the single hold release . I think this is more helpful to understand why the action is release instead of single when the switch is clicked.
[image]
[image]
[image]

uncle-fed, 2022-02-21:
> The following three pictures are the status of the switch report when the single hold release

Do you mean that the switch actually provides different messages for the release after single-click and release after hold but the current converter does not make this distinction? Or I don't fully understand ?...

z-x-f, 2022-02-21:
I mean, this switch, single-click and release report the same property, so it's cumbersome to distinguish them.

Koenkk, 2022-02-21:
> Maybe, as yet another compromise, for now the set of secondary actions (single and hold_release) should be simply introduced. 

This is fine by my

> But the day you decide to make legacy: false the default behaviour of zigbee2mqtt, the release after single-press and the hold_release events should also be gone completely. What do you think?

Legacy false is already the default behavior for many users (it will be set once a new network is started)

@uncle-fed for the sake of completeness, can you provide me the debug log of the following sequences:
- Press button and release immediately
- Press button, hold for a few seconds, release

z-x-f, 2022-02-21:
> Press button and release immediately
> Press button, hold for a few seconds, release

@Koenkk The above three images are all you need

uncle-fed, 2022-02-21:
Single button switch, single press
```
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"61440":59773986,"onOff":0}' from endpoint 2 with groupID 0
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
MQTT publish: payload '{"action":"release","linkquality":86,"state":"OFF"}'
```

Single button switch, hold... wait... release
```
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"61440":59773986,"onOff":1}' from endpoint 2 with groupID 0
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 4 with groupID 0
MQTT publish: payload '{"action":"hold","linkquality":92,"state":"ON"}'
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
MQTT publish: payload '{"action":"release","linkquality":94}'
```

2-button switch, left button, single press
```
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"61440":50343185,"onOff":0}' from endpoint 2 with groupID 0
MQTT publish: topic 'z2m/home/switch/livingroom/ceiling', payload '{"linkquality":99}'
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
MQTT publish: payload '{"action":"release_left","linkquality":81,"state_left":"OFF"}'
```

2-button switch, left button, hold... wait... release
```
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"61440":50343185,"onOff":1}' from endpoint 2 with groupID 0
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 4 with groupID 0
MQTT publish: payload '{"action":"hold_left","linkquality":84,"state_left":"ON"}'
Zigbee message, type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
MQTT publish:payload '{"action":"release_left","linkquality":84}'
```

Koenkk, 2022-02-21:
Thanks, added it now, can you test it?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

uncle-fed, 2022-02-22:
Hi, I've updated dev to get the 'converters 14.0.429' version but I don't get any changed actions, i.e., no "single" actions seen, everything looks like before.

z-x-f, 2022-02-22:
After update, there are two actions after single press, one is single and one is release, long press and then release, one is hold, one is hold_release and one is release

uncle-fed, 2022-02-22:
I've completely reinstalled z2m and switched to the latest dev branch. I am still not seeing the "two actions".
It must be some other parameter in my `configuration.yaml` that affects the behaviour if somebody can see it and I can't.
I don't have any other ideas.

```yaml
homeassistant: false
permit_join: false
frontend:
  port: 8883
mqtt:
  base_topic: z2m
  server: mqtt://127.0.0.1
serial:
  port: /dev/serial0
advanced:
  cache_state: false
  cache_state_persistent: false
  cache_state_send_on_startup: false
  channel: 15
  last_seen: ISO_8601
  legacy_availability_payload: false
  legacy_api: false
  rtscts: false
availability:
  active:
    timeout: 5
  passive:
    timeout: 60
device_options:
  retain: false
  legacy: false
  debounce: 0.5
devices:
  '0x00xxxxxxxxxxxxxxx':
    friendly_name: switch/livingroom/ceiling
  '0x00xxxxxxxxxxxxxxx':
    friendly_name: switch/kitchen/wall
```

Koenkk, 2022-02-23:
@uncle-fed checked the code but I don't see what is wrong yet, is the `release` event still being send?

uncle-fed, 2022-02-24:
Yes, the `release` is being sent as before. Basically I don't observe any changed behaviour as if there was no code changes.

Koenkk, 2022-02-24:
@uncle-fed can you provide the debug log starting from the point where you start z2m until a release action is send?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

uncle-fed, 2022-02-24:
```
Feb 24 19:07:36 rpi-don systemd[1]: Started zigbee2mqtt.
Feb 24 19:07:38 rpi-don npm[5352]: > zigbee2mqtt@1.23.0-dev start
Feb 24 19:07:38 rpi-don npm[5352]: > node index.js
Feb 24 19:07:45 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:07:45: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Feb 24 19:07:45 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:45: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-02-24.19-07-42' filename: log.txt
Feb 24 19:07:45 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:07:45: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-02-22.01-40-39'
Feb 24 19:07:45 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:45: Starting Zigbee2MQTT version 1.23.0-dev (commit #caafc64f)
Feb 24 19:07:45 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:45: Starting zigbee-herdsman (0.14.18)
Feb 24 19:07:45 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:07:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial0","rtscts":false}}'
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: zigbee-herdsman started (resumed)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190523,"transportrev":2},"type":"zStack30x"}'
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:07:47: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b00193650cc","panID":6754}
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: Currently 7 devices are joined:
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/sensor/kitchen/env (0x00158d000322a2f9): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/sensor/outside/env (0x00158d0002720d78): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/sensor/closet/fridge (0x00158d0004855456): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/sensor/bedroom/env (0x00158d00034cec62): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/sensor/livingroom/thermostat (0x000d6f000c2cddc2): AV2010/32 - SMaBiT (Bitron Video) Wireless wall thermostat with relay (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/switch/livingroom/ceiling (0x00158d0002ab622e): QBKG04LM - Xiaomi Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: home/switch/kitchen/wall (0x00158d0002a718ff): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (EndDevice)
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: Zigbee: disabling joining new devices.
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: Connecting to MQTT server at mqtt://127.0.0.1
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:07:47: Using MQTT anonymous login
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: Connected to MQTT server
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:07:47: MQTT publish: topic 'z2m/bridge/state', payload '{"state":"online"}'
Feb 24 19:07:47 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:07:47: Received MQTT message on 'z2m/bridge/info' with data '{"commit":"caafc64f","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":false,"cache_state_persistent":false,"cache_state_send_on_startup":false,"channel":15,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"last_seen":"ISO_8601","legacy_api":false,"legacy_availability_payload":false,"log_directory":"/opt/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"rtscts":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"availability":{"active":{"timeout":5},"passive":{"timeout":60}},"blocklist":[],"device_options":{"debounce":0.5,"legacy":false,"retain":false},"devices":{"0x000d6f000c2cddc2":{"friendly_name":"home/sensor/livingroom/thermostat"},"0x00158d0002720d78":{"friendly_name":"home/sensor/outside/env"},"0x00158d0002a718ff":{"friendly_name":"home/switch/kitchen/wall"},"0x00158d0002ab622e":{"friendly_name":"home/switch/livingroom/ceiling"},"0x00158d000322a2f9":{"friendly_nam [... cut ...]
Feb 24 19:08:43 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:08:43: MQTT publish: topic 'z2m/home/switch/livingroom/ceiling', payload '{"action":"hold","last_seen":"2022-02-24T16:08:43.067Z","linkquality":99,"state":"ON"}'
Feb 24 19:08:44 rpi-don npm[5352]: Zigbee2MQTT:debug 2022-02-24 19:08:44: Received Zigbee message from 'home/switch/livingroom/ceiling', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
Feb 24 19:08:44 rpi-don npm[5352]: Zigbee2MQTT:info  2022-02-24 19:08:44: MQTT publish: topic 'z2m/home/switch/livingroom/ceiling', payload '{"action":"release","last_seen":"2022-02-24T16:08:44.272Z","linkquality":99}'
```

uncle-fed, 2022-02-24:
First is a single press, then it is hold and release sequence.

Koenkk, 2022-02-24:
From the z2m dir, can you do  `nano node_modules/zigbee-herdsman-converters/converters/fromZigbee.js` and above: https://github.com/Koenkk/zigbee-herdsman-converters/blob/e5f65e00abd7d7c6a8bfb27cd54fa4fee5febda9/converters/fromZigbee.js#L5533 add `console.log("ACTION", action);` and provide the STDOUT again?

uncle-fed, 2022-02-24:
```
> zigbee2mqtt@1.23.0-dev start
> node index.js

Zigbee2MQTT:debug 2022-02-24 19:22:46: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
Zigbee2MQTT:info  2022-02-24 19:22:46: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-02-24.19-22-43' filename: log.txt
Zigbee2MQTT:debug 2022-02-24 19:22:46: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-02-22.15-57-19'
Zigbee2MQTT:info  2022-02-24 19:22:46: Starting Zigbee2MQTT version 1.23.0-dev (commit #caafc64f)
Zigbee2MQTT:info  2022-02-24 19:22:46: Starting zigbee-herdsman (0.14.18)
Zigbee2MQTT:debug 2022-02-24 19:22:46: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial0","rtscts":false}}'
Zigbee2MQTT:info  2022-02-24 19:22:48: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-02-24 19:22:48: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20190523,"transportrev":2},"type":"zStack30x"}'
Zigbee2MQTT:debug 2022-02-24 19:22:48: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b00193650cc","panID":6754}
Zigbee2MQTT:info  2022-02-24 19:22:48: Currently 7 devices are joined:
Zigbee2MQTT:info  2022-02-24 19:22:48: home/sensor/kitchen/env (0x00158d000322a2f9): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: home/sensor/outside/env (0x00158d0002720d78): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: home/sensor/closet/fridge (0x00158d0004855456): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: home/sensor/bedroom/env (0x00158d00034cec62): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: home/sensor/livingroom/thermostat (0x000d6f000c2cddc2): AV2010/32 - SMaBiT (Bitron Video) Wireless wall thermostat with relay (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: home/switch/livingroom/ceiling (0x00158d0002ab622e): QBKG04LM - Xiaomi Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: home/switch/kitchen/wall (0x00158d0002a718ff): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (EndDevice)
Zigbee2MQTT:info  2022-02-24 19:22:48: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-02-24 19:22:48: Connecting to MQTT server at mqtt://127.0.0.1
Zigbee2MQTT:debug 2022-02-24 19:22:48: Using MQTT anonymous login
Zigbee2MQTT:info  2022-02-24 19:22:49: Connected to MQTT server
Zigbee2MQTT:info  2022-02-24 19:22:49: MQTT publish: topic 'z2m/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:debug 2022-02-24 19:22:49: Received MQTT message on 'z2m/bridge/info' with data '{"commit":"caafc64f","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":false,"cache_state_persistent":false,"cache_state_send_on_startup":false,"channel":15,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"last_seen":"ISO_8601","legacy_api":false,"legacy_availability_payload":false,"log_directory":"/opt/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"info","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"output":"json","pan_id":6754,"report":false,"rtscts":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"availability":{"active":{"timeout":5},"passive":{"timeout":60}},"blocklist":[],"device_options":{"debounce":0.5,"legacy":false,"retain":false},"devices":{"0x000d6f000c2cddc2":{"friendly_name":"home/sensor/livingroom/thermostat"},"0x00158d0002720d78":{"friendly_name":"home/sensor/outside/env"},"0x00158d0002a718ff":{"friendly_name":"home/switch/kitchen/wall"},"0x00158d0002ab622e":{"friendly_name":"home/switch/livingroom/ceiling"},"0x00158d000322a2f9":{"friendly_name":"home/sensor/kitchen/env"},"0x00158d00034cec62":{"friendly_name":"home/sensor/bedroom/env"},"0x00158d0004855456":{"friendly_name":"home/sensor/closet/fridge"}},"external_converters":[],"frontend":{"host":"0.0.0.0","port":8883},"groups":{},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"z2m","force_disable_retain":false,"include_device_information":false,"server":"mqtt://127.0.0.1"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/serial0","rtscts":false}},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debou [... cut ...]
Zigbee2MQTT:debug 2022-02-24 19:23:32: Received Zigbee message from 'home/switch/livingroom/ceiling', type 'attributeReport', cluster 'genOnOff', data '{"61440":59773975,"onOff":1}' from endpoint 2 with groupID 0
Zigbee2MQTT:debug 2022-02-24 19:23:32: Received Zigbee message from 'home/switch/livingroom/ceiling', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 4 with groupID 0
ACTION release
Zigbee2MQTT:info  2022-02-24 19:23:33: MQTT publish: topic 'z2m/home/switch/livingroom/ceiling', payload '{"action":"release","last_seen":"2022-02-24T16:23:32.798Z","linkquality":105,"state":"ON"}'
```

Koenkk, 2022-02-25:
Ah I think this is because the `debounce` in your `configuration.yaml`  under `device_options`

uncle-fed, 2022-02-25:
Indeed that was the reason! Thanks!
I guess we kind of forgot that multiple `action` properties will collapse this way too.
I've added `debounce_ignore: [action]` to the config and was able to see the `single` and the `hold_release` events now.
Perhaps a warning about this should be given in the docs, 'cause this is something quite easy to overlook.

Koenkk, 2022-02-25:
done, can this be closed?

uncle-fed, 2022-02-25:
Yes, thanks a lot for your efforts and persistence!

closed 2022-02-25 by uncle-fed