Koenkk/zigbee2mqtt #11468: TRADFRI on/off switch E1743 doesn't pair correctly anymore with 1.23.0 but worked earlier

KaHeMu, 2022-02-17:
### What happened?

I have a number of [E1743 ](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) in my network that worked well for quite a while. Since recently (I guess it happened with 1.22.0 or later) it happens that a [E1743 ](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) loses its connection and I cannot re-pair it properly anymore. The device is recognized as "supported" after pairing and gets configured with no errors logged but nothing is reported by the device anymore. In the frontend under Devices -> State it shows 
{
    "linkquality": 157
}
instead of 
{
    "battery": 100,
    "linkquality": 128,
    "update_available": false,
    "action": "off",
    "action_rate": 83
}
I tried on two different networks one with the Slaesh's CC2652RB adapter and the other with a CC2531. Same error in both.

### What did you expect to happen?

The device reporting 
{
"battery": 100,
"linkquality": 128,
"update_available": false,
"action": "off",
"action_rate": 83
}

### How to reproduce it (minimal and precise)

remove device from network and try to re-pair

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

tried both zStack3x0  20211217 and zStack12 20201128 as well

### Adapter

tried both Slaesh's CC2652RB and CC2531

### Debug log

Device '0x5c0272fffeaceb40' joined
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffeaceb40","ieee_address":"0x5c0272fffeaceb40"},"type":"device_joined"}'
Starting interview of '0x5c0272fffeaceb40'
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffeaceb40","ieee_address":"0x5c0272fffeaceb40","status":"started"},"type":"device_interview"}'
Debug Device '0x5c0272fffeaceb40' announced itself
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffeaceb40","ieee_address":"0x5c0272fffeaceb40"},"type":"device_announce"}'
Log from pairing event: 

Retrieving state of '0x5c0272fffeaxxxxx' after reconnect
Received Zigbee message from '0x5c0272fffeaxxxxx', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":172800}' from endpoint 1 with groupID 0
Successfully interviewed '0x5c0272fffeaxxxxx', device has successfully been paired
Device '0x5c0272fffeaxxxxx' is supported, identified as: IKEA TRADFRI ON/OFF switch (E1743)
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"TRADFRI ON/OFF switch","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["on","off","brightness_move_down","brightness_move_up","brightness_stop"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E1743","options":[{"access":2,"description":"Set to false to disable the legacy integration (highly recommended), will change structure of the published payload (default true).","name":"legacy","property":"legacy","type":"binary","value_off":false,"value_on":true},{"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta.","features":[{"access":2,"description":"Delta per interval, 20 by default","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x5c0272fffxxxxx","ieee_address":"0x5c0272fffeaxxxxx","status":"successful","supported":true},"type":"device_interview"}'

mathieuf20, 2022-03-13:
I'm having the same issue following an OTA update to the button. Kept working after OTA but when I had to re-pair it to another setup, it wanted to.

mathieuf20, 2022-03-14:
So I got able to pair it from ZHA in Home Assistant. Seems to be an issue only related to z2m (or dependencies).

KaHeMu, 2022-03-22:
The issue keeps re-occuring repeatedly. After around 1 week my two [E1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743)  silently disappeared again. When trying to re-connect (press the button 4 times like for re-pairing) the log says: 
`Device 'TF_WIPPE_Test' announced itself
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"TF_WIPPE_Test","ieee_address":"0x680ae2fffefexxxx"},"type":"device_announce"}'
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"TF_WIPPE_Test"},"type":"device_announced"}'
Retrieving state of 'TF_WIPPE_Test' after reconnect
`
But still Z2M does not report anything from the device anymore. The same happened with my 2nd [E1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) simultaneously. After "forced-removing the device it does not re-pair again. What am I doing wrong?

KaHeMu, 2022-03-22:
(cont.) Only after manually removing the corresponding sections from the coordinator_backup.json file Z2M finally re-paired with my two E1743. Still the successful configuration took a while: 

`
Failed to configure '0x5c0272fffeaxxx', attempt 1 (Error: Bind 0x5c0272fffeacxxx/1 genOnOff from '0x00124b0014bxxx/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))
Error Failed to interview '0x5c0272fffeacxxx', device has not successfully been paired
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffeacxxx","ieee_address":"0x5c0272fffeacxxx","status":"failed"},"type":"device_interview"}'
Info Configuring '0x5c0272fffeacexxx'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x5c0272fffeaxxx"},"type":"pairing"}'
Info Starting interview of '0x5c0272fffeacxxx'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffeacexxx","ieee_address":"0x5c0272fffeacexxx","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x5c0272fffeacexxx"},"type":"pairing"}'
Debug Saving state to file /app/data/state.json
Debug Device '0x5c0272fffeacxxxx' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5c0272fffeacxxx","ieee_address":"0x5c0272fffeacxxx"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x5c0272fffeacxxx"},"type":"device_announced"}'
Error Failed to configure '0x5c0272fffeacxxx', attempt 2 (Error: Bind 0x5c0272fffeacexxx/1 genOnOff from '0x00124b0014b9xxxx/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7))
Debug Received Zigbee message from '0x5c0272fffeacxxx', type 'readResponse', cluster 'genPollCtrl', data '{"checkinInterval":172800}' from endpoint 1 with groupID 0
Info Successfully interviewed '0x5c0272fffeacxxx', device has successfully been paired
Info Device '0x5c0272fffeacxxx' is supported, identified as: IKEA TRADFRI ON/OFF switch (E1743)
[image]
`

I wonder why it took multiple attempts to reconfigure (my 2nd E1743 switch took even longer) and why it did not re-pair without manually editing the coordinator_backup.json file.

KaHeMu, 2022-04-07:
Does anybody know why for re-pairing any device (i.e. it was properly paired and removed intentionally before) one has to manually remove its section from the coordinator_backup.json? It happens with any device in my system (Z"M version 1.23.0). After I deleted it via the Frontend, I cannot pair it again. Once I stop Z2M, remove the device leftover section from the coordinator_backup.json and restart Z2M it pairs properly. 

Thanks in advance for any help.

closed 2022-05-16 by github-actions[bot]

ti-mo, 2024-01-29:
Same happened here:

- OTA updated my switch to 20230308
- Battery ran out a few months later, so I swapped it
- Due to a misconfiguration in my HA automation, it seemed like the button no longer worked, so I decided to re-pair it
- Failed interview on a fresh battery ~10 times

In the end, I force-removed it through the UI (this didn't leave any entries in `coordinator_backup.json`), restarted z2m and only then did it successfully pair, first-try.