Koenkk/zigbee2mqtt #11473: SONOFF Zigbee 3.0 USB Dongle Plus fails to OTA update IKEA Buttons without a repeater

nvx, 2022-02-18:
### What happened?

I have two SONOFF Zigbee 3.0 USB Dongle Plus sticks, both updated using `CC1352P2_CC2652P_launchpad_coordinator_20211217.zip` with Zigbee2MQTT at different locations. Both locations have a number of IKEA buttons (mostly E1812, but also the odd E1743). One location has a couple of IKEA E1746 repeaters, while the other location just has the coordinator.

The site with the repeaters can OTA update the buttons fine (the buttons are connected to a repeater according to the map), while the site without any repeaters manages to (after a few attempts) check for firmware updates and determine there's an update available, any attempts to actually update the firmware says failed.

### What did you expect to happen?

Successfully OTA update the buttons

### How to reproduce it (minimal and precise)

Pair a IKEA E1812 button to a SONOFF Zigbee 3.0 USB Dongle Plus coordiantor without any routers in the network
Attempt to firmware update the button

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

nvx, 2022-03-21:
Still relevant

nvx, 2022-03-21:
@Koenkk I notice the stale label is still on this, is the bot broken?

nvx, 2022-04-21:
Still relevant

nvx, 2022-05-23:
Still relevant

nvx, 2022-06-23:
Still legit. Can the bot be disabled for this issue @Koenkk?

nvx, 2022-07-24:
Still an issue on the latest version

nvx, 2022-08-24:
Still an issue

nvx, 2022-09-24:
Still relevant

nvx, 2022-10-25:
Still an issue

MattWestb, 2022-10-25:
We have founding that TI coordinator is not playing to` end device timeout request` then Zigbee 3 compatible end device is joining the coordinator.
Some device dont have any problems but oft is ZB3 end device leaving the network then the parent router (the router part of the coordinator) is not Zigbee R22 compatible.

Interesting if is its working with some router and getting the end device jumping to them (for controllers taking out the battery and putting it back then its near the new router).

nvx, 2022-11-28:
Monthly still a relevant issue post

nvx, 2022-12-30:
Pong

nvx, 2023-01-30:
Ping

nvx, 2023-03-03:
Pong

kevin-david, 2023-06-05:
Since stale will hit this, commenting to say that with the new firmware update that just came out this issue allowed me to figure out a (very annoying) workaround 😊

kevin-david, 2023-06-05:
Created a PR for people who take a while to remember to search GitHub issues: https://github.com/Koenkk/zigbee2mqtt.io/pull/2069

nvx, 2023-07-11:
ping

nvx, 2023-07-13:
> For TI coordinators you need updating the firmware then the old is having Zigbee 3 functionality disables so real Zigbee 3 end devices cant have it as parent but is OK with other good routers in the network if you is having some.

I'm not quire sure I follow. Are you saying newer firmware versions cause issues and using an older firmware fixes it? Or the other way around? (Or something else again?)

nvx, 2023-07-16:
> Pre 2023 was having end device functionality disables => Zigbee 3 direct children was having problems. But they have still some not standard settings that is doing problems but direct children looks working OK. (Direct children = end device is using the coordinator as parent and not one other router) [Koenkk/Z-Stack-firmware#439](https://github.com/Koenkk/Z-Stack-firmware/issues/439)

Oh hey, so updating to the latest z stack coordinator firmware seems to have fixed the problem! Cheers for the tip!

closed 2023-07-16 by nvx

cooperaj, 2024-01-08:
What coordinator version did you find worked? I've just upgraded to 20230507 to fix this and still have the issue.

nvx, 2024-01-09:
> What coordinator version did you find worked? I've just upgraded to 20230507 to fix this and still have the issue.

That's the version I'm on that fixed it for me

mrmaximas, 2024-02-07:
Hello!
Is the 20221114 - 1.0.47 latest version?
Thanks!

BurningTheSky, 2024-05-09:
I just updated to 20230507 and am still having this issue. The E2201 is on firmware 1.0.47. If I don't keep the device active I get "OTA: Device didn't respond to OTA request", if I do keep pressing a button to make sure it's awake I get "OTA: Timeout, device did not request any image blocks"

dustdancer, 2024-12-01:
still got the problem

cooperaj, 2025-02-07:
I've just updated to 20240710 and unfortunately am still getting this on one stubborn switch. Somehow I've managed to get the others to update after endless retries but this one just refuses.

ashimokawa, 2025-07-02:
@cooperaj 

I have the same problem, all switches ok, except one.  Firmware is displayed as 0.0.00-0.0.-1
Guess it will never update.