Koenkk/zigbee2mqtt #11476: 3x mislabeled expose "energy_consumed" in the database

createcandle, 2022-02-18:
### What happened?

I found another mislabeled expose name, for the same three devices as before.

Most devices use "energy" to denote how much Kwh they're used.

https://www.zigbee2mqtt.io/supported-devices/#e=energy

These three devices expose `energy_consumer` instead:

https://www.zigbee2mqtt.io/supported-devices/#e=energy_consumed

### What did you expect to happen?

I wanted to find all devices that measure energy use in the devices database

### How to reproduce it (minimal and precise)

Use the device database

### Zigbee2MQTT version

not relevent

### Adapter firmware version

not relevant

### Adapter

not relevant

Koenkk, 2022-02-18:
Also added the energy expose, decided to keep energy_consumed since not only the spelling is different (to prevent a breaking change).

Koenkk, 2022-02-19:
> That pull requests seems to also change some other variables in the Tuya file? Variables not related to this issue?

If you mean the exposes, I've cleaned that up a bit, the end result will be identical

createcandle, 2022-02-19:
Yes that's what I meant. Cool, just checking :-)

closed 2022-02-19 by createcandle